bigtreetech/Eddy #1: Option 'data_rate' is not valid in section 'probe_eddy_current btt_eddy'

krautech, 2024-04-25:
[image]
[image]

Following the guide. Have gotten it flashed with latest firmware etc but when adding the config to my printer.cfg this happens and halts the printer on restart.

krautech, 2024-04-25:
I removed the data_rate: 500 line and it started successfully with no immediate issues. I will continue with the setup process and see if any issues arise from removing this line or whether something else is wrong.

bigtreetech, 2024-04-25:
Hi, it seems that you have not use the klipper version of bigtreetech https://github.com/bigtreetech/klipper/tree/eddy

The upstream of Klipper has not yet merged the PR of [eddy fully functional](https://github.com/Klipper3d/klipper/pull/6558) (Kevin and Arksine are discussing a more elegant and robust way to implement this feature), so we need use the bigtreetech version of klipper to use eddy for the time being.

Please execute the following command in ssh to switch klipper to bigtreetech version
```
cd ~/klipper/
git remote add eddy https://github.com/bigtreetech/klipper
git fetch eddy
git checkout eddy/eddy
```
Restart klipper in the Mainsail, or execute the following command in the ssh command to restart klipper
```
sudo systemctl restart klipper
```

krautech, 2024-04-25:
Ill try again but I followed word for word out of the docs you supplied and those are the commands I used initially where I pulled BTTs version of klipper and still had the issue. Although, in the docs there isnt anything about restarting klipper via ssh (This may need to be added to the documentation.

krautech, 2024-04-25:
A simple restart in mainsail didnt work nor did issuing the command. However after running `sudo reboot` on the Pi and allowing it to reboot, this now is fixed.

Could a reboot command be added between step 1 and 2 here please for anyone who may encounter this issue as well. Thanks guys. Will continue with the setup process.
[image]

bigtreetech, 2024-04-25:
> Could a reboot command be added between step 1 and 2 here please for anyone who may encounter this issue as well.

Got it, Will add this step.

Thank you very much for this feedback.

krautech, 2024-04-25:
No worries happy to help 👍

closed 2024-04-26 by krautech

Hoot0118, 2024-05-14:
I keep trying this command outlined to git eddy but it keeps requesting username and password.
cd ~/klipper/
git remote add eddy https://github.com/bigtreetech/klipper
git fetch eddy
git checkout eddy/eddy

Hoot0118, 2024-05-17:
anybody?

krautech, 2024-05-17:
Hmm it definitely shouldnt be requesting a username and password. Have a screenshot?

Hoot0118, 2024-05-17:
> Hmm it definitely shouldnt be requesting a username and password. Have a screenshot?

[image]

krautech, 2024-05-17:
Check your spelling 👍

Hoot0118, 2024-05-17:
> cd ~/klipper/
> git remote add eddy https://github.com/bigtreetech/klipper
> git fetch eddy
> git checkout eddy/eddy

> Check your spelling 👍
> […](#)

when i copy and past these in i get same thing?? iseen the error in screen shot but thats what it replied back with 

 cd ~/klipper/
git remote add eddy https://github.com/bigtreetech/klipper
git fetch eddy
git checkout eddy/eddy

Hoot0118, 2024-05-22:
I'm not seeing where i miss spelled anything as i double checked as copy and pasting from guide but i keep having this issue where  one step that's hanging me up asking for password it did the first command fine the first time which is why i get already exist. but when i follow up with the git fetch eddy command the username and password keeps showing up.

 biqu@BTT-CB1:~/klipper$ make flash FLASH_DEVICE=2e8a:0003
  Flashing out/klipper.uf2 to 2e8a:0003
sudo lib/rp2040_flash/rp2040_flash out/klipper.uf2

Loaded UF2 image with 109 pages
Found rp2040 device on USB bus 1 address 43
Flashing...
Resetting interface
Locking
Exiting XIP mode
Erasing
Flashing
Rebooting device
biqu@BTT-CB1:~/klipper$ ls /dev/serial/by-id/*
/dev/serial/by-id/usb-Klipper_rp2040_4547415053030ABA-if00
/dev/serial/by-id/usb-Klipper_stm32g0b1xx_1F004D001450415833323520-if00
biqu@BTT-CB1:~/klipper$ git remote add eddy https://github.com/bigtreetech/klipper
error: remote eddy already exists.
biqu@BTT-CB1:~/klipper$ git fetch eddy
Username for 'https://github.com': git checkout eddy/eddy
Password for 'https://git checkout eddy/eddy@github.com':

eddy is set up but anytime i try to home calibrate eddy current it says must home axis first and rinse and repeat this is the only step i cannot complete as it ask for password     
These are the commands 

git remote add eddy https://github.com/bigtreetech/klipper
git fetch eddy
git checkout eddy/eddy

copied directly from https://github.com/krautech/btt-eddy-guide   i am using a cb1 on a manta m4p flashed with btt klipper but for life of me cannot get it to function any help be greatly appreciated.

mateusdemboski, 2024-06-06:
Hey @Hoot0118! Hope you are doing good!

Try these steps:

```bash
git remote remove eddy
git remote add eddy https://github.com/bigtreetech/klipper
git fetch eddy
git checkout eddy/eddy
```