bigtreetech/Eddy #3: General Feedback (Had variant of this originally on Voron Discord)

flyespresso, 2024-04-27:
So I got the BTT Eddy today from Amazon Prime, but there is zero additional docs in the kit and the manual on the github has you replacing your klipper with a branch from BTT–which is severely limiting. No mention of how Klipper now supports Eddy current sensors and some but not all the changes are implemented on mainline klipper branch from this github repo. 

Can the manual / somewhere get more info on what you get with what method or if one method (like using BTT's fork) is required? Loosely seems there isn't any z-endstop support on Klipper Mainline? Possibly BTT fork is required for Eddy?

**Manual Feedback:**

- Entire Section 5.5 on page 18-19 is confusing, it's labeled as temp compensation, has steps 1-3, but then 3 is only for coil / I2C (which is marketed as not having temp compensation) but that should be separate entire section as its for a different component. Not to mention section 1-2 do not mention temperature compensation. So as someone with USB Eddy, is this just additional steps to setup (1+2) and not actually temp compensation? (Step 3 if only for coil needs to be more clear, even ModBot on his youtube live stream covering Eddy is doing Step 3 over and over even though he's using the USB variant...!)
- 5.4.1 and 5.4.2 need to be distinctly different, you'll be getting people adding both to their configs erroneously
- 5.4.3 should be called Bed Mesh Details and have a summary that this is more info on the commands included in 5.4.1 or 5.4.2 and not something additional to add (some people will end up duplicating)

**Completely Missing:**

- No explanation of what you need to change/add/remove in any startup script you have for your machine
- Github has no sample config or any of the config snippets shared in the manual meaning you gotta copy paste with bad PDF formatting

Think this is a case of hardware ready and already at vendors but not the software or instructions–for toolhead boards or mainboards thats sort of OK as there isn't much that's novel/schematic with pinout is all thats needed for a Klipper vetran. For something like this? Kind of a showstopper. Not sure how anyone is going to use it without it being a frustrating journey

(I'm well versed in Klipper, linux, printers, DoomCubes, etc–this is first BTT product where I feel I am being transported back to 2017 in terms of what is available for a 3D Printing Product)
﻿

flyespresso, 2024-04-27:
Adding on, not all Voron 2.4 mounts support it either, as it'll be touching the x-carriage (actually the hole spacing is slightly too tall) and you can't then plug in the USB connector on the top. Bit unfortunate. Looking at the PCB it's not clear why it's so tall, and I'd think it's better with a side USB port (if cost isn't too much higher for that connector, I'd add one on top and one to one side, allowing user to pick what works for their setup. Dimensions don't match the Omron so should be a warning–

flyespresso, 2024-04-27:
CAD file for the Eddy doesn't have the PCB, just the housing and all the components that _would_ be on the PCB. Can this be fixed? It's hard for folks to design custom mounts to possibly rectify the issue if there isn't a PCB in the CAD :P 

[image]

flyespresso, 2024-04-27:
Yeah playing around in CAD and comparing to the Omron  TL-Q5MC2 that is the standard probe on a Voron, this is not a drop in replacement. The enclosure of the sensor is simply way to tall and not inline with what the Omron is, so things that are made for it will not necessarily work with the Eddy. Worse yet, the PCB is also the same size inside the enclosure so it's not even a quick fix if you could print something new as the mounting holes + USB header still are the limiting factor. 

This needs to be overall ~8mm shorter in length, from top towards the holes (I.e, mimic the spacing of the TL-Q5MC2. 

I feel this might have been a waste of a purchase on my side, not a single one of my printers can use it with this packaging / PCB layout. _Most of the popular non-stealth burner toolheads probably have the same issue as well._

flyespresso, 2024-04-27:
Also issue once I do get something rigged up to mount it, it's not stable to do QGL once the heatbed is at temp. It can do it once it's at cold or as it's first heating up.

[image]

flyespresso, 2024-04-27:
So you see also in that screenshot it's right below the 60c limit for the device but this is totally expected use case to... be near the hotbed.

Also not sure why it's that high—rp2040 powered tool head board is a lot less and both my ambient sensors are 47c and 27c (tool head and gantry respectively)

I've tried the auto cal both at ambient room (not chamber) temp and after a heat soak with bed at 100c for 15 min and get the same result after it's at temp for anything more than 5 mins..!

flyespresso, 2024-04-27:
Finally got somewhat stable readings with the Eddy's temps below 45, but... once printing after a mesh it's way off what I calibrated and saved as the probe offset (and verified after restart..!). A solid -.670 to get it to actually print on the bed 🤣

This thing is so inconsistent! Was this only bench tested or on open air printers? Like I can't see this at all for any printer that's enclosed (which means all Voronz, Ratrigs, VZ Bots, etc...).

krautech, 2024-04-27:
I'm just gonna point out that there is indeed an example config supplied by BTT

https://github.com/bigtreetech/Eddy/blob/master/sample-bigtreetech-eddy.cfg

krautech, 2024-04-27:
The replacement of klipper with the BTT version is temporary while the guys at klipper implement it properly. It will be merged into the main fork soon 👍

Annoying, but atleast Kevin etc are working on implementing it actively.

krautech, 2024-04-27:
I agree with the bottom of the guide comment. However the entirety is for eddy USB. The final steps are NOT for eddy coil. 

This should definitely be cleared up in the docs.

https://GitHub.com/KrauTech/vyper-klipper and check out the eddy installation guide I made for clarification

flyespresso, 2024-04-27:
> I'm just gonna point out that there is indeed an example config supplied by BTT
> 
> https://github.com/bigtreetech/Eddy/blob/master/sample-bigtreetech-eddy.cfg

It's missing a lot of what is in the Manual though (Bed Mesh, Safe Z), nor does it switch between USB/Coil. Is that what you used on yours? I never was able to get it do a 'smooth' pass without z-hops. 

Can you post yours by chance?

krautech, 2024-04-27:
Section 3 line should read "Temperature Compensation - USB Only & Not Applicable to Eddy Coil"

Not

"Temperature Compensation (eddy Coil without temperature compensation)

flyespresso, 2024-04-27:
> I agree with the bottom of the guide comment. However the entirety is for eddy USB. The final steps are NOT for eddy coil.
> 
> This should definitely be cleared up in the docs.
> 
> https://GitHub.com/KrauTech/vyper-klipper and check out the eddy installation guide I made for clarification

Oh what, that's soooooo not clear in the manual... Ugh! Let me look at your guide–seems we're replying in realtime 🤣 

I never did those as I thought that was just for Coil–ended up trying to sort it out till 4:30 AM as it's always 'almost there' but not really working.

krautech, 2024-04-27:
`5.4.1 and 5.4.2 need to be distinctly different, you'll be getting people adding both to their configs erroneously

5.4.3 should be called Bed Mesh Details and have a summary that this is more info on the commands included in 5.4.1 or 5.4.2 and not something additional to add (some people will end up duplicating)`

I agree. Whilst they are under different labelled sections, it's easy to confuse the 2 as they are VERY similar.

I suggest moving USB and COIL to seperate guides to help alleviate any possible issues CLEARLY defining in each guide which one is suitable for which and also linking to each other's guide if someone was to get the wrong one.

5.4.3 should definitely not be its own (step) but a sort of glossary

krautech, 2024-04-27:
Haha yeah it's very confusing 🤣 I spent a few hours working on it to get it working properly where I'm laying successful perfect prints

krautech, 2024-04-27:
> - 5.4.1 and 5.4.2 need to be distinctly different, you'll be getting people adding both to their configs erroneously
> - 5.4.3 should be called Bed Mesh Details and have a summary that this is more [...]

I agree. Whilst they are under different labelled sections, it's easy to confuse the 2 as they are VERY similar.

I suggest moving USB and COIL to seperate guides to help alleviate any possible issues CLEARLY defining in each guide which one is suitable for which and also linking to each other's guide if someone was to get the wrong one.

5.4.3 should definitely not be its own (step) but a sort of glossary

flyespresso, 2024-04-27:
Yeah I'm stepping through your writeup, we have the same config for the most part (just obviously different mechanics). @krautech How did you get the z-hop to stop on your bedmesh passes? (meaning a more Beacon/Cartographer/IDM style pass?)

I don't think there is anything in this branch or what BTT is trying to get into Klipper that has the higher fidelity mesh either. BTT seems to be sticking with a more standard mesh with Eddy.

krautech, 2024-04-27:
> It's missing a lot of what is in the Manual though (Bed Mesh, Safe Z), nor does it switch between USB/Coil. Is that what you used on yours? I never was able to get it do a 'smooth' pass without [...]

Mines all on my GitHub.

As I wrote in my notes, z hop was caused in my case, by KAMP being apart of my configuration, I had to remove it and use native adaptive mesh since KAMP was merged into klipper and isn't required as a standalone configuration anymore 👍 fixed it completely and I added the line to my print start macro instead (found at the bottom of my guide)

flyespresso, 2024-04-27:
> I agree. Whilst they are under different labelled sections, it's easy to confuse the 2 as they are VERY similar.
>
> I suggest moving USB and COIL to seperate guides to help alleviate any possible [...]

Agreed 100%, this will quickly become a nightmare for BTT in tech support with people not clearly understanding what is relevant to them, worse once CAN is also available.

krautech, 2024-04-27:
Contact me on discord? We can figure this out and update a better guide based on recommendations that'll get done sooner than BTT lol

flyespresso, 2024-04-27:
> Contact me on discord? We can figure this out and update a better guide based on recommendations that'll get done sooner than BTT lol

I know right lol

flyespresso, 2024-04-28:
After spending another day and night trying to sort things out and comparing notes along the way with @krautech. I'm not sure what is up with mine–it's comparatively useless.

You can sum it up pretty easily by the results from Probe Accuracy, the best I got:
`// probe accuracy results: maximum 0.907119, minimum 0.846850, range 0.060268, average 0.872703, median 0.868160, standard deviation 0.017624
`
Comparing to @krautech's setup:
`probe accuracy results: maximum 1.335915, minimum 1.328258, range 0.007657, average 1.332175, median 1.332410, standard deviation 0.001434
`
This pretty much kills the ability to use it for QGL on a Voron (Common tolerance in cfg is 0.0075). The other issue is that this level of variation means the mesh's are... Not great and useless for printing–the spikes end up meaning your bed mesh isn't going to help, as the toolhead will be so varied (and spiking) as it traverses the bed. See here for 'Best' keep in mind this is a Voron 2.4 with a mic6 Mandala Roseworks bed (not magbed, I have a magnetic sheet, my surface is CS Hyde PEI sheet–not a textured sheet so it's got less variation from coating); scan mode:
[image]

Expanding the defualt config to 25x25 mesh:
[image]

Better yet, 50x50 mesh:
[image]

After hours, multiple temp calibrations, and identifying so many gotcha's for @krautech's Eddy Setup guide, we're out of ideas. 

Summary:
- Too much variance in readings, no matter the temperture, even at ambient, even after complete config wipe– 😕
- Can't use it for QGL with the variance 🫠
- Klipper Build Bug: QGL with this separate Klipper build only measures once for each corner, regardless of settings in printer.cfg to do otherwise
- Mesh's are useless for printing as the variance makes it so you can't actually get a first layer completed 😅
- Mechanically awesome printer, re-verified everything, had no issue with Chaotic Labs CNC Tap V2 prior to Eddy nor Omron inductive probe
- Packaging of the Eddy doesn't work with many Voron setups unless you're using exactly stock printed X Carriage – because it's taller than the other probes that Eddy is supposed to be a drop in replacement for
- CAD file is incorrect, PCB missing, component layout is incorrect too

@bigtreetech Any follow on ideas here? Something is up, as frankly seems either I have a faulty unit or missing something completely or this shouldn't be advertised for Voron's with bed compensation (i.e, QGL).

looxonline, 2024-04-29:
Thanks for posting. We have some guys looking into it and will offer feedback as soon as we chew through it.

flyespresso, 2024-04-29:
Thanks @looxonline ! I did swap the carriage out and back to TAP which I had before and it's running 100%. I'll reply here with the images / results but it's bed mesh is what I'd expect and same with probe accuracy (similar to @krautech with Eddy). So can rule out mechanical performance—

flyespresso, 2024-04-30:
As promised, this is what it is with TAP, same machine, same bed.
[image]

Probe Accuracy Results:
`00:56:25 
// probe accuracy results: maximum -0.369986, minimum -0.372486, range 0.002500, average -0.370236, median -0.369986, standard deviation 0.000750
00:56:25 
// probe at 25.000,292.000 is z=-0.369986
00:56:24 
// probe at 25.000,292.000 is z=-0.369986
00:56:23 
// probe at 25.000,292.000 is z=-0.369986
00:56:22 
// probe at 25.000,292.000 is z=-0.369986
00:56:21 
// probe at 25.000,292.000 is z=-0.369986
00:56:20 
// probe at 25.000,292.000 is z=-0.369986
00:56:19 
// probe at 25.000,292.000 is z=-0.369986
00:56:17 
// probe at 25.000,292.000 is z=-0.369986
00:56:16 
// probe at 25.000,292.000 is z=-0.369986
00:56:16 
// probe at 25.000,292.000 is z=-0.372486`

So yeah, not sure what is up with Eddy being clearly, not right; plus ruled out any mechanical/machine issue. Let me know what could be it @looxonline & @bigtreetech

BTTJUSTIN, 2024-05-24:
Hi， apologize for the inconvenience, can i have your facebook account or email address? We will offer you the solution as soon as possible.

BTTJUSTIN, 2024-05-27:
Hi， Thanks for your reply ,my wrokmate said that he has send you an email ,please pay attention to your mail box.@flyespresso

BTTJUSTIN, 2024-05-28:
Hi flyespresso,did you check on your mail box?My workmate said that he did not receive a reply from you.
Please send email to my mail box ligw@biqu3d.com, i will forward your email to my sale workmate @flyespresso

krautech, 2024-05-28:
> Hi flyespresso,did you check on your mail box?My workmate said that he did not receive a reply from you.
> Please send email to my mail box ligw@biqu3d.com, i will forward your email to my sale workmate @flyespresso

Hey @BTTJUSTIN I replied to Luke Harrison's email the other day but haven't heard anything since. Could you get him to check his emails please :)

krautech, 2024-05-28:
Hey Luke all good!

Yeah I am AU but I'm awake all hours of the night and day lately :)

flyespresso, 2024-05-28:
> Hi flyespresso,did you check on your mail box?My workmate said that he did not receive a reply from you.
> 
> Please send email to my mail box ligw@biqu3d.com, i will forward your email to my sale workmate @flyespresso

Replied this morning, I FW'ed it to you as well as including my discord. 👍

krautech, 2024-05-29:
@bigtreetech @BTTJUSTIN I don't suppose we have an ETA on eddy can? Wanting to try it out with my EBB 36,42 and SB2209 boards

BTTJUSTIN, 2024-05-29:
Hi, We are already working on this product. It is already far into development.

BTTJUSTIN, 2024-06-04:
Hi flyespresso,did you check on your mail box?My workmate said that he did not receive a reply from you.
Please send a reply via email to my sale workmate @flyespresso

BTTJUSTIN, 2024-06-06:
Hi flyespresso,did you check on your mail box?My workmate said that he did not receive a reply from you.
He will offer you the solution.Please send a reply via email to my sale workmate @flyespresso

flyespresso, 2024-06-06:
Yes I have @BTTJUSTIN

Kooramah, 2024-07-01:
> After spending another day and night trying to sort things out and comparing notes along the way with @krautech. I'm not sure what is up with mine–it's comparatively useless.
>
> **You can sum it up [...]

I seem to have this same issue on an Ender 3. I'm wondering iwhat the fix was about

krautech, 2024-07-01:
> I seem to have this same issue on an Ender 3. I'm wondering iwhat the fix was about

There wasn't a fix. Eddy was removed and replaced with a different probe.

flyespresso, 2024-07-01:
> This pretty much kills the ability to use it for QGL on a Voron (Common tolerance in cfg is 0.0075). The other issue is that this level of variation means the mesh's are... Not great and useless for printing–the spikes end up meaning your bed mesh isn't going to help, as the toolhead will be so varied (and spiking) as it traverses the bed. See here for 'Best' keep in mind this is a Voron 2.4 with a mic6 Mandala Roseworks bed (not magbed, I have a magnetic sheet, my surface is CS Hyde PEI sheet–not a textured sheet so it's got less variation from coating); scan mode: image <https://private-user-images.githubusercontent.com/1707612/326280729-9d4cc714-4208-442a-a8d5-57a7691d0134.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTk3OTI0ODAsIm5iZiI6MTcxOTc5MjE4MCwicGF0aCI6Ii8xNzA3NjEyLzMyNjI4MDcyOS05ZDRjYzcxNC00MjA4LTQ0MmEtYThkNS01N2E3NjkxZDAxMzQucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDcwMSUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA3MDFUMDAwMzAwWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9YTE2NmZmNGNhYzQ2MTE3NjcwYjkxZDdmM2ZjMTNhMWY2Yjk1ZjliNjlhZmU3MDFlZGYwY2I2MDY0MjJiMDgwMSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.2AqQnOCynQ7iTVvYvGgkL0Wuqj4zPFQlPBk2iC2n5wI>
> Expanding the defualt config to 25x25 mesh: image <https://private-user-images.githubusercontent.com/1707612/326281310-01103746-e639-4d43-ac2b-a55fb05bd938.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3M [... cut ...] yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA3MDFUMDAwMzAwWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9YzRhMWYwZmI4MmUwYmIxZTA2OGRlNTQ1YmI5NjkyMmNlN2I3YzBkY2Y4N2UxNDU3YWZkYWIxZDg5ODE2MmNjYiZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.ranPZ3dpTiyieVMa87dOuLYkf-QyjCgKEpModM5BGqs>

Yeah there wasn't a fix, BTT has also not reached out when they resolved it (I was told they would–) so I can re-evaluate. As much as I pushed to understand if it was just my unit (i.e, hardware issue), they never asked for more data or information. Just that their engineers where looking into it and get back to me. I did get a refund eventually–so there's that at least. 

It's a bit disappointing, I'm not one to be super negative on any particular product. Just how they've handled this has been really poor, even trying to get me to remove the details here in this thread and *to change my Amazon review...* 

In other news, support for Eddy got into mainline Klipper; yet the same issue's I outlined persist, there's a thread on Klipper github of the exact same issues @krautech and I have had.

looxonline, 2024-07-01:
> Yeah there wasn't a fix, BTT has also not reached out when they resolved it (I was told they would–) so I can re-evaluate. As much as I pushed to understand if it was just my unit (i.e, hardware [...]

It's completely unacceptable that they asked you to change anything. I have actually gone straight to the CEO to speak about this and him and I will follow up and find out who it was. They probably thought that they were doing a good thing. We will make sure that they know that this is not acceptable and certainly not company policy.

Regarding the issues you experienced, I have run through the whole scenario that you described and I cannot replicate anything like you were experiencing. See the video at this link:  https://drive.google.com/file/d/1b6dHDi72AZDF25EGkQ06DW6ZlVIkE8nZ/view?usp=sharing

I can only assume that you had a bad piece of hardware. There are a few things that I would like to mention though:

1.) When you do thermal calibration you likely have the hotend and the heatbed on. The same should be true when testing accuracy afterwards. The hotend causes a lot of thermal expansion that the thermal calibration has accounted for and without this expansion the result will be marginally off (probably around +/- 0.1mm).
2.) The homing procedure is currently inaccurate if you are using the eddy to home and only have a standard G28 as part of the homing process. It triggers on the first sample that matches the needed value when it should take tens more and average them. This can result in the first layer varying by more than a layer height. I will update the readme tomorrow to reflect the correct homing macro that takes many, many samples and averages them.
3.) The assumption that QGL is only taking one sample is wrong. The Eddy is taking thousands of samples per second and averaging them. There is therefore no need to make multiple dives. This is the same reason why there is no need to make multiple passes over a point when running a rapid scan.

I really believe that you had bad hardware but I am interested to investigate other reports so I am going to find that discord thread you mentioned and chase up. Thanks for letting us know about it.

looxonline, 2024-07-01:
> I seem to have this same issue on an Ender 3. I'm wondering iwhat the fix was about

I am going to reopen this issue. Please could you kindly add more detail about what you are experiencing so that I can understand whether or not it is the same as @flyespresso

krautech, 2024-07-01:
@looxonline thanks for reopening this. I have a list somewhere of about 6 or 7 people who had this same issue as @flyespresso 
Although I'll admit it was some time ago and I haven't had contact with most, they've gone back to their original probes.

Ill try dig up the list.
I too ended up removing Eddy, even doing everything cold still produced inconsistencies and the eventual MASS eddy disconnects caused me to give up.

I tried multiple different USB cables, including buying new ones, modifying some etc to try remedy the situation  but alas, same result.

I haven't touched eddy in a while because of this unfortunately.

bigtreetech, 2024-08-19:
The issues in this thread have all been resolved via one or more of the points below:

a. Hardware issue identified with a user's eddy.
b. Updates to macros and firmware.
c. Cable update for disconnect issues.

closed 2024-08-19 by bigtreetech