bigtreetech/Eddy #5: Z Offset Changing

krautech, 2024-05-02:
Unfortunately everytime I have to printer restart (I think) eddy seems to ignore the z offset that's been set using live printing and saving from the console in mainsail. I can see it adding to the printer.cfg that the z_offset is changing when saving it, yet whenever I have to restart my printer, I need to adjust my z_offset by 0.2mm everytime roughly. I started with a z_offset of 1 and now I'm at near 2mm!! It continues to happen again and again. Nothing in my setup changes other than a restart. [image]

krautech, 2024-05-02:
I redid it fresh, removed all of Eddys config and redid it.

Same problem.

I started with Z_Offset = 1 did the paper test etc all went well, saved it.

Fired up a test print, watched the first layer, too high obviously, so adjusted it -0.125 for a perfect squish, applied and SAVE_CONFIG

z_offset: 1.125 is now in printer.cfg under the do not edit portion.

the SAVE_CONFIG caused a printer restart, so started a new print. 
THE Z OFFSET IS THE SAME as before i adjusted it even though it has z_offset set correctly in printer.cfg

So i adjust it. Again its a -0.125 for a perfect squish, SAVE_CONFIG again.

Go look at printer.cfg and its now Z_Offset: 1.250

Something isnt right..

krautech, 2024-05-02:
I manually set z_offset this way
[image]

Same issue, it doesnt acknowledge the z_offset and i still have to manually adjust -0.125

flyespresso, 2024-05-02:
MidBot's stream of the I think switch wire I think had similar issue. FWIW

allanwrench28, 2024-05-02:
Yep I have the Eddy "Coil" and I have the same issue, I just placed the z offset I needed as a G code line in my start print macro right after G28

krautech, 2024-05-02:
> MidBot's stream of the I think switch wire I think had similar issue. FWIW

I noticed this last night after watching his stream playback as well. Hmm

krautech, 2024-05-02:
> Yep I have the Eddy "Coil" and I have the same issue, I just placed the z offset I needed as a G code line in my start print macro right after G28, and in my end print macro I use G code to clear the offset

Yeah I've done the same until I figure this out. It's annoying but it's not the end of the world.

But I do believe it needs a fix please @bigtreetech

madsi1m, 2024-05-06:
I am also seeing this, i am 1mm off each time. @bigtreetech does need to fix this!
@allanwrench28 what magics did you do to manually set and unset it in the gcode?

sandmmakers, 2024-05-06:
I'm also experiencing similar issues with the USB version installed in a Neptune 3 Max. If I follow all of the "cold directions", the nozzle is too far above the bed to print. I can hand modify the `z_offset` to get a successful print, but I can't figure out the exact relationship between the calibration results and the amount of change I need to apply to the `z-offset`. I've also noticed that when I perform a `PROBE`, the values don't seem correct, since I'm getting values around -4mm. When I perform a cold bed mesh, all of the values are also larger negative values. When I perform the temperature calibration (I've tried at least six+ times), the values from the probe seem more erratic and less consistent. Most of my temperature calibration attempts have had a target value around 64C (Nozzle @260 and bed @100 with an unenclosed 430x430 bedslinger) but I've also tried with a target value closer to 50C.

Creating a cold mesh always completes (I can't speak to the mesh accuracy), but at temp, the mesh will frequently error out with move out of range errors. At one point, I got a `PROBE` result of -99mm, which is obviously incorrect.

With my current bracket, the bottom of the probe is 1mm above the tip of the nozzle.

krautech, 2024-05-07:
I'm over it.

I think I'll be removing eddy and putting my BL touch back on

I just can't get it to be consistent. 
Everytime I print I have to manually adjust the z offset. I've calibrated for thermal drift in range 30c to 85c and nothing works.

I might get one good print if I'm lucky and then when I go to do another I'm having to manually change the z offset, which I can't save.

I'm honestly disappointed.

krautech, 2024-05-08:
@madsi1m @flyespresso @sandmmakers @allanwrench28 

I can neither confirm or deny at this stage that swapping back to the mainline of klipper instead of btt's klipper will solve or atleast help with some issues.

Eddy has been merged into the main branch of klipper from what I've been told.

madsi1m, 2024-05-08:
Looking at https://github.com/Klipper3d/klipper/pulls?q=eddy I can see that @Arksine did some PRs but doesn't seem to include everything (yet?)
https://github.com/bigtreetech/klipper/commits/eddy/

madsi1m, 2024-05-08:
For me the Eddy USB works fine other than the z offset. my only thing is manual `SET_GCODE_OFFSET Z=<number>` is an absolute number so start with a high number and work your way down

krautech, 2024-05-08:
Yeah I've just found that. Working with someone ATM to figure it out. I'll report back

krautech, 2024-05-08:
> For me the Eddy USB works fine other than the z offset. my only thing is manual `SET_GCODE_OFFSET Z=<number>` is an absolute number so start with a high number and work your way down

What temps are you running everything? I found if I don't heat my enclosure it's fine (minus the z offset issue) but when I heat the enclosure it goes whack and throws the z offset out

madsi1m, 2024-05-08:
no enclosure, but from room temp (cold start) and reprints it seems consistent (bed at 66 degrees)

some random noise no one understands:
```
#*# [probe_eddy_current btt_eddy]
#*# reg_drive_current = 15
#*# calibrate =
#*# 	0.050000:3269931.544,0.100000:3269058.324,0.150000:3268119.785,
#*# 	0.200000:3267239.573,0.250000:3266327.833,0.300000:3265354.825,
#*# 	0.350000:3264522.581,0.400000:3263665.720,0.450000:3262825.103,
#*# 	0.500000:3261878.305,0.550000:3261060.561,0.600000:3260292.457,
#*# 	0.650000:3259547.399,0.700000:3258723.561,0.750000:3257903.601,
#*# 	0.800000:3257106.981,0.850000:3256241.135,0.900000:3255457.651,
#*# 	0.950000:3254623.690,1.000000:3253837.685,1.050000:3253050.210,
#*# 	1.100000:3252328.509,1.150000:3251598.041,1.200000:3250832.687,
#*# 	1.250000:3250120.583,1.300000:3249440.238,1.350000:3248697.232,
#*# 	1.400000:3248002.954,1.450000:3247358.677,1.500000:3246697.628,
#*# 	1.550000:3246020.265,1.600000:3245395.472,1.650000:3244729.586,
#*# 	1.700000:3244134.493,1.750000:3243500.088,1.800000:3242887.414,
#*# 	1.850000:3242285.741,1.900000:3241687.564,1.950000:3241118.487,
#*# 	2.000000:3240598.871,2.050000:3240055.509,2.100000:3239561.220,
#*# 	2.150000:3239038.240,2.200000:3238498.487,2.250000:3238003.609,
#*# 	2.300000:3237544.397,2.350000:3237037.158,2.400000:3236556.467,
#*# 	2.450000:3236077.954,2.500000:3235662.605,2.550000:3235210.751,
#*# 	2.600000:3234781.504,2.650000:3234314.051,2.700000:3233942.738,
#*# 	2.750000:3233548.676,2.800000:3233134.001,2.850000:3232751.718,
#*# 	2.900000:3232337.861,2.950000:3231972.273,3.000000:3231600.079,
#*# 	3.050000:3231226.917,3.100000:3230881.033,3.150000:3230540.783,
#*# 	3.200000:3230210.851,3.250000:3229886.582,3.300000:3229580.396,
#*# 	3.350000:3229236.875,3.400000:3228923.474,3.450000:3228612.058,
#*# 	3.500000:3228326.307,3.550000:3227998.205,3.600000:3227721.665,
#*# 	3.650000:3227423.848,3.700000:3227135.987,3.750000:3226920.543,
#*# 	3.800000:3226633.863,3.850000:3226348.977,3.900000:3226073.794,
#*# 	3.950000:3225818.643,4.000000:3225594.644,4.050000:3225298.757
#*# calibration_temp = 29.848553
#*# drift_calibration =
#*# 	3310127.841833, -554.728091, 2.752797
#*# 	3295137.904060, -467.993916, 2.074839
#*# 	3282545.182362, -393.277901, 1.509666
#*# 	3270547.524380, -274.499385, 0.561740
#*# 	3259506.737881, -155.874581, -0.318383
#*# 	3250420.774441, -73.903284, -0.841705
#*# 	3243146.392618, -31.077439, -1.015140
#*# 	3236971.173436, 13.473592, -1.289882
#*# 	3232475.967612, 28.823786, -1.318070
#*# drift_calibration_min_temp = 27.68500865045914
```

madsi1m, 2024-05-08:
I seem to be off by 1mm after a G28 regardless of bed temp or how heatsoaked the eddy is

krautech, 2024-05-08:
Can you upload your klippy?

sandmmakers, 2024-05-08:
I can provide my data point tomorrow. Will redo all steps and take detailed notes. I just want to gain enough confidence about what the probe is doing so I can finish my video.

madsi1m, 2024-05-08:
here is my current config https://github.com/madsi1m/BeltEnder/blob/main/printer.cfg

Arksine, 2024-05-08:
Hopefully one of BTT's reps will be able to respond to this issue and provide more detailed support.  That said, I'm willing to provide what information I can with regard to work I have done and my experience with the probe.

To begin, my testing/development has been done with an Eddy on a Voron 2...however I do not use it to home the machine, I use the nozzle switch.  I agreed to help with two software additions to support eddy current probes, these include a "scanning" mode for sampling a mesh and implementing thermal drift compensation.  The documentation for using both of these should hopefully be in the repo.

With regard to homing, I can state the the `z_offset` for this probe works differently.  It is used to calculate the trigger frequency, so changing the z_offset will not correct a probe triggering too high and I would recommend keeping at it a `.5`.   I suspect the issue some of you are running into is due to *thermal expansion*.  That is, if you ran `PROBE_EDDY_CURRENT_CALIBRATE` when the printer is cold, its possible that thermal expansion has changed the relationship between the nozzle and the probe, which will throw off calibration.   One potential solution is to calibrate with the printer warm, another would be to apply gcode offset in a macro.

I don't know what BTT recommends with regard to the mounting position, however I have run mine at around 2.5mm above the nozzle.  I would think that recommending a consistent height would be advisable.  Also, while I have not experienced it myself, I have heard that some testers experienced a change in the required drive current.  I don't know enough about the ldc1612 to say what would cause this, but it would result in a probe that is not consistent.  This is something BTT should likely research.

Finally, with regard to thermal compensation, it cannot correct issues introduced by thermal expansion.  A proper calibration will correct drift introduced by changes in coil temperature.

krautech, 2024-05-08:
> Hopefully one of BTT's reps will be able to respond to this issue and provide more detailed support.  That said, I'm willing to provide what information I can with regard to work I have done and my [...]

Thanks for the detailed response.
Alot of that actually makes sense and coincides with some testing I did today.

Calibrating the probe at temperature right before I printed is what worked, so heatsoaked and at temp, I've been able to successfully run multiple prints back to back.

I did the paper test and did it a little lower than I normally would and that seems to have worked, but adding a macro to lower it slightly more before printing is definitely what I think I'll do.

I think alot of confusion I had was because I was using it like it was a BL Touch which obviously it isn't.

I've got it to work in the mean time even if that means doing the quick calibration just before a print (if temperatures are different etc)

Be curious to see what others come up with and follow along as things progress especially while I'm building my voron 🤣

Arksine, 2024-05-08:
> Creating a cold mesh always completes (I can't speak to the mesh accuracy), but at temp, the mesh will frequently error out with move out of range errors. At one point, I got a PROBE result of -99mm, which is obviously incorrect.

FWIW, if you are using the scanning mode, this indicates that `HORIZONTAL_MOVE_Z` is too high.  I tend to run mine at 1 or 2 mm.

krautech, 2024-05-08:
> I have my `HORIZONTAL_MOVE_Z` set to 2mm, as described in the documentation, and the bottom of the probe is 1mm above the bottom of the nozzle. I originally tried having my probe 2mm above the nozzle, but the results seemed less stable.
> 
> I will need to go back through the calibration steps to provide a consistent and accurate report, since I was previously trying lots of different things and they are starting to get blurred together.
> 
> Also, I have the probe installed in a Neptune 3 Max (with Klipper) which is a 430x430 bed slinger with a magnetic PEI plate. The magnet is a full bed sheet with six holes for mounting/adjustment screws. 

Mines about 1.5mm  above the nozzle give or take.

We run the same bed with holes for the screws etc so I know that isn't an issue

When you do a probe_accuracy, what're your results?

sandmmakers, 2024-05-09:
Here are the results from my testing last night and this morning. My results don't really make a lot of sense. Using the probe while cold sort of works, but at temp, the values are inconsistent and sometimes very wrong. Even though I have the USB version, I didn't test temperature compensation last night. However, when I calibrated at temp and then tried probing cold, the results were actually more consistent than when I calibrate at temp and test at temp.
[S&M Makers - Eddy Testing v1.7z.txt](https://github.com/bigtreetech/Eddy/files/15262822/S.M.Makers.-.Eddy.Testing.v1.7z.txt)

madsi1m, 2024-05-10:
I had things going good, then i changed the build plate and it went way off.
@bigtreetech I reverted back to bltouch until this becomes a more complete product.

haynesdavid, 2024-05-11:
same issue, thinking of using a sex bolt and only using it to mesh, as its not good at offset

atroph, 2024-05-15:
Just chiming in for the same issue w/ the USB Eddy.

Every print *after* a "save_config/fw restart" I have to re-enter a z-offset of around -1.25mm. 

If I get it adjusted, hit save, and don't save_config/fw restart before the next print it'll be spot on every time.

krautech, 2024-05-15:
> Just chiming in for the same issue w/ the USB Eddy.
> 
> Every print *after* a "save_config/fw restart" I have to re-enter a z-offset of around -1.25mm. 
> 
> If I get it adjusted, hit save, and don't save_config/fw restart before the next print it'll be spot on every time. 

Yeah this is pretty much the going standard.

In your start print macro add some g code to adjust the z height by your chosen amount if it's always the same is generally what people are doing

madsi1m, 2024-05-15:
I was doing that until i changed build plate and the manual z-offset i needed to set via gcode changed...

krautech, 2024-05-15:
> I was doing that until i changed build plate and the manual z-offset i needed to set via gcode changed...

Yeah I um.. ooops.

splatula, 2024-05-19:
I just finished perfecting my current install after a week of screwing around.
I noticed a couple changes to the manual as well, notably the precise location of the "sensor" for your XY offset.  Make certain your XY offset is accurate, or the probe won't be.

I became REALLY frustrated with the Z offset, AND the calibration method for the temperature drift.
So, back to basics.
Using a feeler gauge and several test prints I found the perfect offset for my Eddy.
I made a drop-in BL-Touch to Eddy mount, so I knew more or less where the offset was mathematically as I'd designed the mount to put Eddy just under the recommended 2mm from nozzle height.
Then I  proceeded to use different gauges to calibrate Eddy around the proper one, test printing between to see how the progress was going.
Many .02mm layer tests later I had Eddy tuned.
To fine tune I used a caliper to see how thick my .02mm layer really was, then offset by using a feeler gauge of the exact amount off + the prior gauge to hone Eddy's accuracy.  NOTE - I DIDN'T use the Z-Offset buttons, or any manual z-height tweaks. This method is "measure, start printing, cancel print, measure, start printing, cancel print" until you find it.

Once I had my Z-Offset I then used eddy to calibrate my bed screws to .04 of a difference, redid the calibration with the gauges and then did my heightmap.

After several tried this week to get thermal drift working properly, I just don't think a human is good enough to properly feel that many nozzle-bed distances accurately in a row, and I said screw it.  Calibrate it hot and print happily, just as you would if you were calibrating a non-self-levelling bed.

If you've already done it once, REMOVE THE DRIFT SETTINGS FROM YOUR CONFIG THAT YOU DON'T WANT...  don't just leave those settings in there or it will screw your offsets.  Unless you somehow managed to do it accurately, then my hats off to you.

Arksine, 2024-05-20:
> After several tried this week to get thermal drift working properly, I just don't think a human is good enough to properly feel that many nozzle-bed distances accurately in a row, and I said screw [...]

Admittedly the drift calibration procedure is a big ask.  That said, it isn't all that different from manually probing a 5x5 or greater mesh.  It isn't necessary for each manual probe to be identical, however what is important is to get the widest range possible, with as many samples as possible.  With practice it can be done, as shown in the attached plot (admittedly I had A LOT of practice during development).

[image]

As things stand right now, there isn't a better way to perform the calibration for a mounted probe.  Without the manual probe thermal expansion of the frame and/or gantry will impact the calibration.  It is possible to do the calibration on an unmounted probe, and I could look into providing a script and instructions for doing so.  It would involve printing some parts to hold the probe, then collecting data while the probe is sitting on the bed.

If  "contact probing" can be implemented in the future then it will be possible to automate the entire drift calibration process.

splatula, 2024-05-20:
> Arksine

Yeah, it's not really THAT bad, as long as you make sure you're using a hard rule as to how much feel on your gauge is the proper feel.
I guess part of my frustration might have been caused by some inconsistency between my "feel" on the gauges and the heating process.  The way it is in the instructions you just set the bed and nozzle to where you're going to be printing at, and as the probe heats up you're supposed to test the offset at each interval... however the temperature has usually gone up 4-8 degrees or so between the beginning of my gauging and the end.

So the closest I've been able to make the drift work was using the following method:
PROBE_EDDY_CURRENT_CALIBRATE CHIP=btt_eddy
perform the z-offset routine with gauges
SAVE_CONFIG
then
SET_IDLE_TIMEOUT TIMEOUT=36000
PROBE_DRIFT_CALIBRATE PROBE=btt_eddy TARGET=55 STEP=5
do the first z-offset cold, then I set the temp of the nozzle to 220, and the bed to 6-8 degrees warmer than Eddy currently reports - when it hits the temp you do the offset calc, then crank it another 6-8 degrees.

Is there a way to couple the calibration of Eddy not just to a z-position and Eddy's temp, but also the bed/nozzle temp?  So when you call PROBE_DRIFT_CALIBRATE PROBE=btt_eddy TARGET=55 STEP=5 it would raise the bed/nozzle temps to log and consider the curve on those as part of the overall z-stop calculation?

ifreislich, 2024-05-22:
I spent a very long time on thermal calibration. My process was to creep up on each node from 35C to 70C in 5C steps.  I performed the calibration with my nozzle at my most common print temperature to account for its thermal expansion. Once the Z height paper test is triggered I used an SSH connection to my printer and a series of temperature adjustments from the CLI because both KlipperScreen and Mainsail prevent this once the height dialog starts. I continued to tweak the bed temperature as follows via CLI to ensure that the Eddy was heat soaked at or close to the node temperature before applying the final Z adjustment and accepting the value.
`echo "SET_HEATER_TEMPERATURE HEATER=heater_bed TARGET=83" >~/printer_data/comms/klippy.serial`

It was extremely time consuming but I think the end result was a good set of thermal calibration samples.

Regarding the Z-offset, as people have noticed, the offset is about 0.12mm which is roughly the thickness of the paper used in the test. My experimentation leads me to believe that all you need to do to correct the Z-offset is to run the manual Z-offset calibration using `PROBE_EDDY_CURRENT_CALIBRATE CHIP=btt_eddy` and apply your print determined offset to the paper test result before accepting. Note that it is really hard to be consistent with the paper test. You do not need to perform the thermal calibration again.

FWIW, I used the following on another computer to make patch which I've been applying to the official Klipper:

`git clone https://github.com/bigtreetech/klipper.git`
`cd klipper`
`git checkout eddy`
`git diff master >~/eddy.patch`

You can apply this patch with `patch -p1 <~eddy.patch` in your klipper folder up until the recent LDC1612 changes that check for an error condition in the eddy current sensor chip. The BTT Eddy appears to always be in an error state during homing so the print aborts.

BTTJUSTIN, 2024-05-29:
Thanks for reporting. We will get this to the software team and provide feedback once they analyze it.

BTTJUSTIN, 2024-06-14:
Hi, thanks for your reporting
I get news from our engineer, The "z offset save" issue will be fixed in the end of weekend (approximately date is 24th )

j-collom, 2024-06-27:
> Hi, thanks for your reporting I get news from our engineer, The "z offset save" issue will be fixed in the end of next weekend (approximately date is 24th )

@BTTJUSTIN  - is this fully fixed and out?

BTTJUSTIN, 2024-07-04:
The eddy has been merge to the klipper .
About the Z offset save issue ,our engineer already has a solution on it ,we need some time to take care of its details.
I will keep on following up .
Thanks for your information.

madsi1m, 2024-07-04:
great! happy to test if needed

looxonline, 2024-07-13:
> great! happy to test if needed

To use the new z-offset feature simply set your eddy up according to our latest guide in the readme document.

closed 2024-07-13 by looxonline

maulionc, 2024-12-29:
same issues in my side 
need to adjust manually arround zoffset -0.125 at each printing 
any solution?

yellowicz, 2025-02-07:
I use the latest configuration and calibrated according to the latest instructions. Every time after finishing printing I have a question whether to save z_offset and after restarting during printing I have to set z_offset again. But I found a solution that I used before and was the most satisfied with - Vitalii TAP. Or Cartographer. Help on Discord, great manual. BTT should learn from them.

maulionc, 2025-02-07:
Hello Marcin do you have more details please

bigtreetech, 2025-02-07:
> Hello Marcin do you have more details please
> 
> Le ven. 7 févr. 2025 à 10:49, Marcin R. ***@***.***> a écrit :
> […](#)

Please try the latest software here: https://github.com/vvuk/eddy-ng/wiki