bigtreetech/Eddy #7: Probe center location missing

V1EngineeringInc, 2024-05-08:
Hi, I looked at the step file, the instructions and the shop page and I see dimensions of the eddy but no indication of what we should use and the center point. Trying to get an accurate offset.

I took mine apart, and I see the center of the coil looks to be 4.8mm from the front of the PCB which means it is about 9.4cm from the back of the probe.

I assume left to right, it is in the center of the footprint?

ivindor, 2024-05-12:
[image]

V1EngineeringInc, 2024-05-12:
All the edges other than the backplate are tapered for mold release, the best place to have accurate dimensions is the back.

I am using the dimensions I listed, just trying to let BTT know they are missing a critical piece of info.

haynesdavid, 2024-05-14:
are you having good results? centre and 9.5 mm from the back. i started with my offset 4mm forward of that as stole someone elses config. think that worked slightly better although need to test more.

V1EngineeringInc, 2024-05-14:
I am not sure how to test it with a lot of precision but I actually just pulled one apart again and measured the assembly. I got 11.4mm from the backplate to the center of the coil.

I updated the first post with the corrected info.

haynesdavid, 2024-05-14:
hmmm, thats in the middle of my 4mm i guess. will also try.

V1EngineeringInc, 2024-05-14:
I tried to take a picture so you could verify but I just don't have enough hands!

V1EngineeringInc, 2024-05-20:
Fixed here, https://github.com/bigtreetech/Eddy/commit/516355262bbe5b122a30733da37a058067cdf916

closed 2024-05-20 by V1EngineeringInc