bigtreetech/Eddy #8: Power offset incorrect?

V1EngineeringInc, 2024-05-13:
"Place Eddy Approx. 20mm above the bed.
From Mainsail or Fluidd run command LDC_CALIBRATE_DRIVE_CURRENT CHIP=btt_eddy
Type SAVE_CONFIG to save the drive currant to your config"

This did not work for me, it resulted in an offset that was above the 4mm limit and would fail.

I think it should be "2mm", that works perfectly for me.

Place Eddy Approx. 2mm above the bed.
From Mainsail or Fluidd run command LDC_CALIBRATE_DRIVE_CURRENT CHIP=btt_eddy
Type SAVE_CONFIG to save the drive currant to your config

BTTJUSTIN, 2024-06-04:
Thanks for reaching out. Our support team does not operate using the GitHub platform but we would be happy to help you via email based ticketing system. Please visit https://biqu.equipment/pages/technical-support and submit a ticket and we will be in touch with you soon.

looxonline, 2024-06-04:
> "Place Eddy Approx. 20mm above the bed. From Mainsail or Fluidd run command LDC_CALIBRATE_DRIVE_CURRENT CHIP=btt_eddy Type SAVE_CONFIG to save the drive currant to your config"
>
> This did not work [...]

Hmmmm....it should indeed be 20mm. What error was thrown when you tried?

V1EngineeringInc, 2024-06-04:
This did not work for me, it resulted in an offset that was above the 4mm limit and would fail.

Changing the LDC_CALIBRATE_DRIVE_CURRENT CHIP=btt_eddy to 16 worked. At 20mm it would calibrate to 15.

I know understand why this is all wrong but the current calibration does not seem to work right, still need to manually edit the offset to work.

15 would actually work as well but the 4mm max offset is hard-coded in and for that reason it fails if you need more.

splatula, 2024-06-30:
LDC_CALIBRATE_DRIVE_CURRENT CHIP=btt_eddy isn't the calibration command for the Z offset. 

There are TWO (three if you count the temp one I don't bother with) calibrations you need to do. 

You need to run LDC_CALIBRATE_DRIVE_CURRENT CHIP=btt_eddy first to calibrate the sensor internally... then PROBE_EDDY_CURRENT_CALIBRATE CHIP=btt_eddy to do the actual Z offset.

*Physically your eddy should be less than 4mm offset-height of your nozzle (ideally 2mm), and if you have a magnetic bed that uses magnets instead of a magnetic sheet it can/will throw things off as well.

IF you did the temp offset, remove the settings from the printer.cfg until that feature is reliable.  I always calibrate z offsets at my printing temp (noz and bed) for the best accuracy anyhow, and Eddy is no different.

looxonline, 2024-07-13:
> This did not work for me, it resulted in an offset that was above the 4mm limit and would fail.
>
> Changing the LDC_CALIBRATE_DRIVE_CURRENT CHIP=btt_eddy to 16 worked. At 20mm it would calibrate to [...]

Are you saying that you mounted your probe above the 4mm limit? I really would not recommend going too much above that because it is operating outside of the intended design spec. You could probably push a mm or two extra but not much more.

V1EngineeringInc, 2024-07-13:
No I mounted it ~2mm above, or whatever the specs are. Not sure the exact number as I am not near the printer right now.
I have not tried this since any of the updates since I got it to work. I will wait until it is all in mainline klipper to try again.

looxonline, 2024-07-13:
> No I mounted it ~2mm above, or whatever the specs are. Not sure the exact number as I am not near the printer right now. I have not tried this since any of the updates since I got it to work. I will wait until it is all in mainline klipper to try again.

Cool but the updates will offer better repeatability between homing sessions and they are all implemented in macros so no need to do any klipper updates.