bigtreetech/Eddy #9: Z_offset does not save

V1EngineeringInc, 2024-05-13:
Z_Offset does not save.

If the printer.cfg file is updated manually it works as it should but SAVE_CONFIG does not save Z_Offset.

diegostf, 2024-06-14:
> Z_Offset does not save.
> 
> If the printer.cfg file is updated manually it works as it should but SAVE_CONFIG does not save Z_Offset.

Same problem here.

BTTJUSTIN, 2024-06-14:
Hi, thanks for your reporting 
I get news from our engineer, The "z offset save" issue  will be fixed  in the end of weekend (approximately date is 24th )

0xfde9, 2024-06-30:
hello, any update regarding this BIG issue?

krautech, 2024-06-30:
Negative

Subcode, 2024-07-01:
> Hi, thanks for your reporting I get news from our engineer, The "z offset save" issue will be fixed in the end of next weekend (approximately date is 24th )

Is it possible to get an update from the Engineer? The fix was supposed to be posted a week ago at this point.

Kooramah, 2024-07-02:
I need some clarification, I see that the Readme.md has been updated with the new macro SET_GCODE_OFFSET Z= regards to not being able to save the Z-Offset.

But the sample-bigtreetech-eddy.cfg has a new G28 and SET_Z_FROM_PROBE macro. What does this do when replacing G28 to call SET_Z_FROM_PROBE macro.? If my bed is too low or too high, I still need to use SET_GCODE_OFFSET and add it to my START_PRINT?

Maybe its an early morning for me and Im getting confused so forgive me

looxonline, 2024-07-02:
> I need some clarification, I see that the Readme.md has been updated with the new macro SET_GCODE_OFFSET Z= regards to not being able to save the Z-Offset.
> 
> But the sample-bigtreetech-eddy.cfg has a new G28 and SET_Z_FROM_PROBE macro. What does this do when replacing G28 to call SET_Z_FROM_PROBE macro.? If my bed is too low or too high, Do I still need to use SET_GCODE_OFFSET and add it to my START_PRINT?
> 
> Maybe its an early morning for me and Im getting confused so forgive me

Hey. The readme has a section at the end that (I hope) explains it fairly well. It explains why we don't recommend hacking the macro to include an offset and why we just recommend using the macros as they currently stand (but definitely using them). Please take some time to read through that section, carefully as well as the new macros and let me know if it is still not clear.

Kooramah, 2024-07-02:
> > I need some clarification, I see that the Readme.md has been updated with the new macro SET_GCODE_OFFSET Z= regards to not being able to save the Z-Offset.
> > But the sample-bigtreetech-eddy.cfg has a new G28 and SET_Z_FROM_PROBE macro. What does this do when replacing G28 to call SET_Z_FROM_PROBE macro.? If my bed is too low or too high, Do I still need to use SET_GCODE_OFFSET and add it to my START_PRINT?
> > Maybe its an early morning for me and Im getting confused so forgive me
> 
> Hey. The readme has a section at the end that (I hope) explains it fairly well. It explains why we don't recommend hacking the macro to include an offset and why we just recommend using the macros as they currently stand (but definitely using them). Please take some time to read through that section, carefully as well as the new macros and let me know if it is still not clear.
> 
> Also, if you are having problems with the first layer being inconsistent then I strongly suggest recalibrating the eddy. There should be no need for the thermal recalibration, just the quick eddy calibration.

Got it thank you, its pretty clear now. My morning was just...eh lol. Than you for your suggestions on there for recalibrating. I've actually been doing that already. Before I was just manually updating my z-offset from the old readme.

Thanks again

0xfde9, 2024-07-04:
Hello, sorry I'm a bit lost now, where do I add this? 
`SET_GCODE_OFFSET Z={printer.probe.last_z_result - cf['probe_eddy_current btt_eddy'].z_offset - 0.XX}`
Because trying to run this in the terminal 
`SET_Z_FROM_PROBE`
I got this message:
`Unknown command:"SET_Z_FROM_PROBE"
`

looxonline, 2024-07-04:
> Hello, sorry I'm a bit lost now, where do I add this? `SET_GCODE_OFFSET Z={printer.probe.last_z_result - cf['probe_eddy_current btt_eddy'].z_offset - 0.XX}` Because trying to run this in the terminal `SET_Z_FROM_PROBE` I got this message: `Unknown command:"SET_Z_FROM_PROBE" `

You need all of the new macros from our sample config file. Please take some time to read the comments in the sample config file as they also explain things well.

BTTJUSTIN, 2024-07-07:
> > Hi, thanks for your reporting I get news from our engineer, The "z offset save" issue will be fixed in the end of next weekend (approximately date is 24th )
> 
> @BTTJUSTIN Is it possible to get an update from the Engineer? The fix was supposed to be posted a week ago at this point.

Hi,The eddy has been merge to the klipper .
About the Z offset save issue ,our engineer already has a solution on it ,we need some time to take care of its details.
I will keep on following up .
Thanks for your information.

Subcode, 2024-07-10:
@BTTJUSTIN You say eddy has been merged with klipper, does that mean we should switch to the normal klipper version?
Or are some features not yet implemented.

bigtreetech, 2024-07-11:
> @BTTJUSTIN You say eddy has been merged with klipper, does that mean we should switch to the normal klipper version? Or are some features not yet implemented.

Sorry, @BTTJUSTIN is mistaken. The PR has been submitted to be merged into Klipper but Kevin has asked for some tweaks before it is fully integrated. It should be merged any day now but not yet.

Also, the z-offset issue has been solved. Please use our latest config files and perform the calibration procedures again. The guide has been updated to reflect all of the needed steps.

closed 2024-07-11 by bigtreetech