bigtreetech/Eddy #12: Missing mcu configuration in both the manual and sample configuration making klipper restart unreliable

BrokenAutomap, 2024-05-17:
The default klipper restart method for serial connections is `arduino `
Because Eddy is using usb this isn't a valid restart method causing klipper not to be able to restart without reboot of the host OS.
Instead the correct restart method is `command`
So after the correction the mcu section of the config should be as follows:

```    
[mcu eddy] 
serial: /dev/serial/by-id/ (Refer to the actual ID found in SSH or Mainsail using
the method above)
restart_method: command
```

V1EngineeringInc, 2024-05-20:
What are your symptoms of this? I am willing to give it a test if I know what I am looking for.

looxonline, 2024-05-21:
Thanks for reporting. We will get the documentation updated and close this issue out once the updates are uploaded.

BrokenAutomap, 2024-05-22:
I

> What are your symptoms of this? I am willing to give it a test if I know what I am looking for.

Basically with btt eddy present without this additional line klipper fails to restart. After issuing either firmware restart or klipper restart, whatever you choose it'll stay in this "restarting" state forever. When it happens only rebooting the whole os managed to make the klipper start properly.

dangerzau, 2024-06-06:
there is an R-pi restart method you could try instead of command.

BTTJUSTIN, 2024-06-12:
The documentation has been updated. Thanks again for pointing out the issue

closed 2024-06-12 by BTTJUSTIN