bigtreetech/Eddy #17: Klipper reports shutdown

Chaotic3d, 2024-05-27:
At random times during a print I will receive the error: Klipper reports shutdown lost communication with MCU Eddy. I am running Eddy usb on manta e3ez with CB1. I have tried reinstalling Eddy a few different times always with the same end result. All prints fail at some point do to loosing communication with Eddy.

looxonline, 2024-05-28:
@BTTJUSTIN Please escalate this to the engineering team ASAP. No need to wait for the weekly review meeting.

BTTJUSTIN, 2024-05-28:
Thanks for reporting. We will get our engineering  team to look into this.

Elwin232, 2024-05-28:
HI，I  met this question as well. 
But it result from the looseconnection the usb cable btween the manta board usb port .Are you able to get the usb id of the eddy via this command ls /dev/serial/by-id at the SSH interface?

BTTJUSTIN, 2024-05-29:
Hi Chaotic3d ,I got some details need to be confirmed with you.
1.Are you able to get the the usb id of the eddy via this command ls /dev/serial/by-id at the SSH interface when  the print to fail at some point during the print ?(Klipper reports shutdown lost communication with MCU Eddy) 
2.or it only shows up  the usb id of the eddy  I run a restart ?
3.Could you share me your klippy log file?

Chaotic3d, 2024-05-29:
Let me make sure I understand so once it has failed and shows lost
communication with MCU Eddy ssh into and see if it can see Eddy before
restarting klipper? I will try that as soon as I can I'm waiting on a part
for the printer due to my hotend fan dying due to it going full blast after
klipper lost communication with Eddy.

BTTJUSTIN, 2024-05-29:
Hi, Yes, once it has failed and shows lost communication with MCU Eddy,please  ssh into and see if it can see the id of the Eddy before restarting klipper

looxonline, 2024-05-29:
> Let me make sure I understand so once it has failed and shows lost communication with MCU Eddy ssh into and see if it can see Eddy before restarting klipper? I will try that as soon as I can I'm [...]

Could you also tell me a little more about your setup? 

What board is being used?
What SBC is being used?
Can you post some pics of your wire routing?
Have you coiled any extra length of the Eddy wire near to high electronic noise sources?
Is the Eddy wire running parallel to high electronic noise sources over a long distance?

BTTJUSTIN, 2024-06-03:
Hi Chaotic3d , do you have time to make a try on this ------once it has failed and shows lost communication with MCU Eddy,please ssh into and see if it can see the id of the Eddy before restarting klipper.@Chaotic3d

closed 2024-06-14 by BTTJUSTIN

wjbillett, 2024-07-09:
Ever figure this out? I have same problem. It happens randomly.  I am using eddy on a hermit crab 2 CAN setup.

bigtreetech, 2024-07-09:
> Ever figure this out? I have same problem. It happens randomly. I am using eddy on a hermit crab 2 CAN setup.

Are you using it (Eddy) over a USB connection? So Eddy and not Eddy Coil.

bigtreetech, 2024-07-09:
@wjbillett @Chaotic3d @Elwin232 We have a new cable that we would like to send to you to test. Please could you send me an email at luke@bigtree-tech.com so that we can get this arranged.

hakankok, 2024-11-23:
I have an eddy sensor that I have been using for about 5 months.The sensor works via USB connected to the Manta M8P platform.However, recently, after performing the table calibration, it stops the printing process by giving a message that the connection is lost during the printing process.However, recently, after performing the table calibration, it stops the printing process by giving a message that the connection is lost during the printing process.

hakankok, 2024-11-23:
I have an eddy sensor that I have been using for about 5 months.The sensor works via USB connected to the Manta M8P platform.However, recently, after performing the table calibration, it stops the printing process by giving a message that the connection is lost during the printing process.When I connect via ssh, it appears that the id is registered. Can you help me solve the problem?

hakankok, 2024-11-23:
I wonder if the problem was solved for the people you sent new cables to?