bigtreetech/Eddy #18: Simple question around LDC_CALIBRATE_DRIVE_CURRENT CHIP=btt_eddy output

dangerzau, 2024-05-28:
Hi There,

Apologies if this has been answered somewhere else as I've been unable to find the answer - The question is should the output of the command LDC_CALIBRATE_DRIVE_CURRENT CHIP=btt_eddy change if I was to move the probe away or closer to the bed?  as I find even if I was to hold the probe at say head hight in the air and issue the command the answer back is always "15"

If I then place the probe directly on the bed - 15
2mm away - 15

I'm assuming this isn't correct?

V1EngineeringInc, 2024-05-28:
Odd, mine said 15 at 20mm away, I did it at 2mm and it said 16...and worked better. I never tried right on the bed.

Sorry, I don't have an answer, but I hope to follow along to learn as well.

dangerzau, 2024-05-28:
I too tried it at 2mm and got 15 as well - I did find a post somewhere online that mentioned the drive current should have been known to the manufacturer at the time of creating the device - with the recommendation of 16 being the value to utilise (Found it: [here](https://klipper.discourse.group/t/eddy-current-sensor-homing-and-calibration-problems/16670) So I used that in my config (Simply changed the setting where autoconf had left its mark) and that seems to have worked fine - Not that 15 didn't as I never actually tried to use it due to receiving usb timeout issues at that point - I took the horrible course of action and changed the timeout value to .5 and thats hidden that issue well enough for now that I can at least try out the new toy... and OMG RAPID SCAN IS AMAZING!!!! WHere has it been all my life.. lol

ifreislich, 2024-05-29:
TLDR; I think 16 is the correct value.

I've done a ton of debugging on this. The calibrated return value depends initial IDRIVE0 register value. During calibration, Klipper chooses 15 for this the first time. If you change the value for `reg_drive_current` in the saved config section you can determine the range. These are my results:

initial IDRIVE0=2-13 → cal result: 14
initial IDRIVE0=14 → cal result: 14
initial IDRIVE0=15 → cal result: 15
initial IDRIVE0=16-31 → cal result: 16

For me, I'm running Klipper master with a patch generated from this fork so I could keep getting updates. Kevin added some error checking based on values returned in the 4 most significant bits of the sample data from the LDC1612 and it broke homing and scanning for me (error was Low Amplitude) until I increased `reg_drive_current` to 16. I’ve also found that IDRIVE0=16 produces sporadic low amplitude errors when the sensor temperature is around 70C, but IDRIVE0=17 results in high amplitude errors when the sensor temperature is below 56C.

From my reading of the datasheet, `LDC_CALIBRATE_DRIVE_CURRENT CHIP=btt_eddy` **is not the correct way** to determine the sensor drive current. Drive current is something that should come out of the design process since it should be picked from TABLE42 using the parasitic resistance of the of the sensor coil. In other words, this is a value that BTT should provide. However, until BTT does the engineering, this calibration is the best we have. 

You need to perform `PROBE_EDDY_CURRENT_CALIBRATE CHIP=btt_eddy` and thermal drift calibration if you change the drive current.

BTTJUSTIN, 2024-06-04:
Thanks for making this suggestion. We will discuss it internally and let you know if it will be scheduled for a future firmware release

andymeade, 2025-08-26:
Was there anything decided on this?  Brand new BTT Eddy Duo and I'm having all sorts of issues that most people seem to point to the drive current.