bigtreetech/Eddy #19: Documentation Question

dangerzau, 2024-05-28:
Heading the warning at the top of the page - And Step 13 being how to change to the BTT fork of Klipper, should that be step 1 that you change to that repo and then flash the BTT Klipper fork onto the MCU of both the printer and the eddy?

The reason I ask is if I do leave it until step13 (Trying to diagnose issues I'm having with Comms timeouts) Klipper throws a mcu protocol error due to the difference in Klipper version - BTT being V0.23.0-166-g957475421 and Mainline being v0.12.0-206-g3078912f

So in essence - 
Step 0.5: Change to BTT fork 
Step 1: Guide goes here etc....

Or am I just missing something like a setting somewhere and I should be able to have a different version of Klipper on the printer mainboard vs the Eddy (Don't think that's the case iirc)

krautech, 2024-05-29:
So it's just a warning that further down the line you'll be asked to change to BTTs fork of klipper that's all.

It's behind the main fork so you'll get a "detached" notification in mainsail/fluidd that lets you know you are on a different version

You shouldn't get an MCU protocol error been on a different version however.

You should be on mainline klipper until step 13. Although off the top of my head, it shouldn't matter if you do it prior 👍

Anakin20097, 2024-06-02:
I'm having the same issue. My Klipper still thinks it's on the mainline version but eddy is on the BTT version, hence getting MCU error and won't allow me to move forward

[image]

Anakin20097, 2024-06-02:
Nevermind, I reflashed the software into eddy and it went back to version 12.0-166 and it's working now !

BTTJUSTIN, 2024-06-05:
Hi,it is glad to hear that you figured it out.

closed 2024-06-05 by BTTJUSTIN

dangerzau, 2024-06-06:
I didn't figure it out in the sense of the only way I could get it to work was to reflash all devices to use the BTT eddy fork of klipper.... reading through the comments here... I went back into mainline - flashed both the PRinters mainboard and the EddyUSB with mainline klipper - Switched branch/fork as per the instructions at step 13... and get the folllowing error:

Klipper reports: ERROR

MCU Protocol error

This is frequently caused by running an older version of the
firmware on the MCU(s). Fix by recompiling and flashing the
firmware.

Your Klipper version is: v0.12.0-166-g957475421
MCU(s) which should be updated:
mcu: Current version v0.12.0-207-g6cd17420
eddy: Current version v0.12.0-207-g6cd17420
Up-to-date MCU(s):
<none>

Once the underlying issue is corrected, use the "RESTART"
command to reload the config and restart the host software.

mcu 'eddy': Command format mismatch: ldc1612_setup_home oid=%c clock=%u threshold=%u trsync_oid=%c trigger_reason=%c vs ldc1612_setup_home oid=%c clock=%u threshold=%u trsync_oid=%c trigger_reason=%c error_reason=%c

So either something still isn't correct here, or am I missing something STILL?