bigtreetech/Eddy #20: MCU Protocol error

diegomtnz, 2024-05-29:
Klipper reports: ERROR

MCU Protocol error

This is frequently caused by running an older version of the
firmware on the MCU(s). Fix by recompiling and flashing the
firmware.

Your Klipper version is: v0.12.0-166-g957475421
MCU(s) which should be updated:
mcu: Current version v0.12.0-178-g7e8c7f46
eddy: Current version v0.12.0-206-g3078912f
Up-to-date MCU(s):
<none>

Once the underlying issue is corrected, use the "RESTART"
command to reload the config and restart the host software.

mcu 'eddy': Command format mismatch: ldc1612_setup_home oid=%c clock=%u threshold=%u trsync_oid=%c trigger_reason=%c vs ldc1612_setup_home oid=%c clock=%u threshold=%u trsync_oid=%c trigger_reason=%c error_reason=%c

krautech, 2024-05-29:
Update your MCU versions of klipper, it's a few versions behind

diegomtnz, 2024-05-29:
Solved, first I did step 13 change to BTT klipper:
`git remote add eddy https://github.com/bigtreetech/klipper
git fetch eddy
git checkout eddy/eddy`

and then I flashed EDDY

`SSH into raspberry PI or your host device
Type
cd ~/klipper
make menuconfig
Use these settings to compile the firmware. Firmware Image
Once set, hit 'Q' and when asked, select yes to save.
Type to compile.make
Disconnect power to Eddy
Push and hold boot button on Eddy (Its next to where the cable plugs in) and at the same time, plug in the cable to your Raspberry Pi Boot Image
SSH into host device
Type into the command line. You should see eddy.lsusb
LSUSB Image

Type into command linecd ~/klipper
Type Remember to change 2e8a:0003 to your device ID you found in step 9make flash FLASH_DEVICE=2e8a:0003
Type into the command line. The found device will be what you enter into your klipper config under [mcu eddy] for the Serial variable.ls /dev/serial/by-id/*`

krautech, 2024-05-29:
Glad it's sorted

I'm going to add to the guide to make sure klipper is up-to-date before attempting this 👍

BTTJUSTIN, 2024-06-04:
Hi， thanks for your reporting , it is glad to hear that you figured it out.

closed 2024-06-04 by BTTJUSTIN