bigtreetech/Eddy #22: Z Endstop not working (Must calibrate probe_eddy_current first )

diegomtnz, 2024-05-30:
Manta-M8P V2 + CB1

mcu(stm32h723xx)
Version: v0.12.0-178-g7e8c7f46

mcu eddy(rp2040)
Version: v0.12.0-166-g957475421
Host(aarch64, 64bit)

Version: v0.12.0-166-g957475421
OS: BTT-CB1 2.3.4 Bullseye

printer.cfg:
`# This file contains common pin mappings for the BIGTREETECH Manta M8P V2.0
# To use this config, the firmware should be compiled for the
# STM32H723 with a "128KiB bootloader" "25 MHz crystal"
# and "USB (on PA11/PA12)", "CAN bus (on PD0/PD1)" or Serial (on USART1 PA10/PA9).

# See docs/Config_Reference.md for a description of parameters.
[include dyze500.cfg]
#[include fluidd.cfg]
[include mainsail.cfg]

#  VIRTUAL SDCARD
#[virtual_sdcard]
#path: /home/biqu/printer_data/gcodes
#on_error_gcode: CANCEL_PRINT

# Motor1
[stepper_x]
step_pin: PE6
dir_pin: PE5
enable_pin: !PC14
microsteps: 16
rotation_distance: 40
endstop_pin: ^!PF4 #tmc2240_stepper_x:virtual_endstop   ## PF4
position_endstop: 0
position_min:0
position_max: 300
homing_speed: 300
second_homing_speed: 10
homing_retract_dist: 15
homing_retract_speed: 50

# Motor2
[stepper_y]
step_pin: PE2
dir_pin: PE1
enable_pin: !PE4
microsteps: 16
rotation_distance: 40
endstop_pin: ^!PF3 #tmc2240_stepper_y:virtual_endstop   ## PF3
position_endstop: 0
position_min:0
position_max: 300
homing_speed: 250
second_homing_speed: 10
homing_retract_dist: 15
homing_retract_speed: 50

# Motor3
[stepper_z]
step_pin: PB8
dir_pin: !PB7
enable_pin: !PE0
microsteps: 16
rotation_distance: 8
endstop_pin: probe:z_virtual_endstop
#position_endstop: 0
position_min: -5.0
position_max: 270
homing_speed: 35
second_homing_speed: 5
homing_retract_dist: 10

# Motor4
[stepper_z1]
step_pin: PB4
dir_pin: !PB3
enable_pin: !PB6
microsteps: 16
rotation_distance: 8
endstop_pin: probe:z_virtual_endstop

# Motor5
[extruder]
step_pin: PG13
dir_pin: PG12
enable_pin: !PG15
microsteps: 16
rotation_distance: 33.500
nozzle_diameter: 0.4
filament_diameter: 1.75
heater_pin: PA0 # HE0
sensor_pin: PB0 # T0
sensor_type: Dyze500
control: pid
pid_Kp: 22.2
pid_Ki: 1.08
pid_Kd: 114
min_temp: 0
max_temp: 300

## End-Stop 5
#[filament_switch_sensor material_0]
#switch_pin: PF0

## Motor6
#[extruder1]
#step_pin: PG9
#dir_pin: PD7
#enable_pin: !PG11 
#heater_pin: PA1 # HE1 
#sensor_pin: PC5 # T1
#...

## End-Stop 6
#[filament_switch_sensor material_1]
#switch_pin: PC15

## Motor7
#[extruder2]
#step_pin: PD4
#dir_pin: PD3
#enable_pin: !PD6
#heater_pin: PA3 # HE2
#sensor_pin: PC4 # T2
#...

## Motor8
#[extruder3]
#step_pin: PC7
#dir_pin: PC8
#enable_pin: !PD2
#heater_pin: PA5 # HE3
#sensor_pin: PA7 # T3
#...

[heater_bed]
heater_pin: PF5
sensor_pin: PB1 # TB 
sensor_type: ATC Semitec 104GT-2
control: watermark
min_temp: 0
max_temp: 130

#[fan_generic soc-fan]
#pin: host:gpio79  #CB1
#pin: host:gpio26  #CM4

# Fan0
[fan]
pin: PF7

## Fan1
#[heater_fan fan1]
#pin: PF9

## Fan2
#[heater_fan fan2]
#pin: PF6

## Fan3
#[heater_fan fan3]
#pin: PF8

## Fan4
#[heater_fan fan4]
#pin: PA4

## Fan5
#[heater_fan fan5]
#pin: PA6
#tachometer_pin: PC2

## Fan6
#[heater_fan fan6]
#pin: PA2
#tachometer_pin: PC1

[mcu]
serial: /dev/serial/by-id/usb-Klipper_stm32h723xx_320016001951313434373135-if00
[temperature_sensor CB1]
sensor_type: temperature_host

[mcu eddy]
serial: /dev/serial/by-id/usb-Klipper_rp2040_45474E621A8C887A-if00
[temperature_sensor btt_eddy_mcu]
sensor_type: temperature_mcu
sensor_mcu: eddy
min_temp: 10
max_temp: 100
[temperature_probe btt_eddy]
sensor_type: Generic 3950
sensor_pin: eddy:gpio26
horizontal_move_z: 2

[printer]
kinematics: cartesian
max_velocity: 500
max_accel: 5000
max_z_velocity: 75
max_z_accel: 150

########################################
# TMC2240 configuration
########################################

# Motor1
[tmc2240 stepper_x]
cs_pin: PC13
spi_software_mosi_pin: PG6
spi_software_miso_pin: PG7
spi_software_sclk_pin: PG8
#diag0_pin: ^!PF4
run_current: 0.6
stealthchop_threshold: 999999
#driver_SGT: 2
interpolate: True

# Motor2
[tmc2240 stepper_y]
cs_pin: PE3
spi_software_mosi_pin: PG6
spi_software_miso_pin: PG7
spi_software_sclk_pin: PG8
#diag0_pin: ^!PF3
run_current: 0.6
stealthchop_threshold: 999999
#driver_SGT: 2
interpolate: True

# Motor3
[tmc2240 stepper_z]
cs_pin: PB9
spi_software_mosi_pin: PG6
spi_software_miso_pin: PG7
spi_software_sclk_pin: PG8
#diag1_pin: PF2
run_current: 0.6
stealthchop_threshold: 999999

# Motor4
[tmc2240 stepper_z1]
cs_pin: PB5
spi_software_mosi_pin: PG6
spi_software_miso_pin: PG7
spi_software_sclk_pin: PG8
#diag1_pin: PF1
run_current: 0.6
stealthchop_threshold: 999999

## Motor5
#[tmc2240 extruder]
#cs_pin: PG14
#spi_software_mosi_pin: PG6
#spi_software_miso_pin: PG7
#spi_software_sclk_pin: PG8
#run_current: 0.800
#stealthchop_threshold: 999999

## Motor6
#[tmc2240 extruder1]
#cs_pin: PG10
#spi_software_mosi_pin: PG6
#spi_software_miso_pin: PG7
#spi_software_sclk_pin: PG8
#run_current: 0.800
#stealthchop_threshold: 999999

## Motor7
#[tmc2240 extruder2]
#cs_pin: PD5
#spi_software_mosi_pin: PG6
#spi_software_miso_pin: PG7
#spi_software_sclk_pin: PG8
#run_current: 0.800
#stealthchop_threshold: 999999

## Motor8
#[tmc2240 extruder3]
#cs_pin: PC6
#spi_software_mosi_pin: PG6
#spi_software_miso_pin: PG7
#spi_software_sclk_pin: PG8
#run_current: 0.800
#stealthchop_threshold: 999999

[board_pins]
aliases:
    # EXP1 header
    EXP1_1=PE7, EXP1_2=PG1,
    EXP1_3=PG0, EXP1_4=PF15,
    EXP1_5=PF14, EXP1_6=PF13,    # Slot in the socket on this side
    EXP1_7=PF12, EXP1_8=PF11,
    EXP1_9=<GND>, EXP1_10=<5V>,

    # EXP2 header
    EXP2_1=PE13, EXP2_2=PE12,
    EXP2_3=PE15, EXP2_4=PE11,
    EXP2_5=PE10, EXP2_6=PE14,      # Slot in the socket on this side
    EXP2_7=PE8, EXP2_8=<RST>,
    EXP2_9=<GND>, EXP2_10=<NC>

# See the sample-lcd.cfg file for definitions of common LCD displays.

#[adxl345]
#cs_pin: PA15
#spi_bus: spi3a

[probe_eddy_current btt_eddy]
sensor_type: ldc1612
z_offset: 1.0
#i2c_address:
i2c_mcu: eddy
i2c_bus: i2c0f
x_offset: 0 # Set according to the actual offset relative to the nozzle
y_offset: 20 # Set according to the actual offset relative to the nozzle
data_rate: 500

#[bltouch]
#sensor_pin: PD13
#control_pin: PD12
#z_offset: 1

[safe_z_home]
home_xy_position: 150, 150
z_hop: 10
z_hop_speed: 5
speed: 250

[bed_mesh]
horizontal_move_z: 2
speed: 300
mesh_min: 10, 10
mesh_max: 290, 290
probe_count: 9, 9
algorithm: bicubic

## Proximity switch
#[probe]
#pin: PD8

#[output_pin ps_on_pin]
#pin: PD14

#[neopixel my_neopixel_1]
#pin: PD15

#[hall_filament_width_sensor]
#adc1: PC0
#adc2: PF10

#*# <---------------------- SAVE_CONFIG ---------------------->
#*# DO NOT EDIT THIS BLOCK OR BELOW. The contents are auto-generated.
#*#
#*# [probe_eddy_current btt_eddy]
#*# reg_drive_current = 18
`

I run the command:
`LDC_CALIBRATE_DRIVE_CURRENT CHIP=btt_eddy`
`SAVE_CONFIG`
add me to printer.cfg:
`#*# <---------------------- SAVE_CONFIG ---------------------->
#*# DO NOT EDIT THIS BLOCK OR BELOW. The contents are auto-generated.
#*#
#*# [probe_eddy_current btt_eddy]
#*# reg_drive_current = 18`
But when I try to home in the z axis I always get this message:
`Must calibrate probe_eddy_current first`

ifreislich, 2024-05-30:
You've missed a step. You still need to `PROBE_EDDY_CURRENT_CALIBRATE CHIP=btt_eddy`. You should also re-flash your stm32 mcu with the same version as your host.

krautech has an excellent guide that's easier to follow. Section 3 is where I think you're at.
[krautech's eddy setup guide](https://github.com/krautech/btt-eddy-guide?tab=readme-ov-file#z-endstop)

krautech, 2024-05-30:
Hey,

Make sure you're placing eddy 20mm above the bed when you're doing the LDC_CALIBRATE_DRIVE_CURRENT CHIP=btt_eddy command.

It should auto generate in your printer.cfg the drive current. Yours looks like it's 18 which is very wrong so something has gone wrong in one of your earlier steps.

Reflash firmware and make sure the base of your eddy probe is exactly 20mm above the centre of your bed before you do it.

If you still have issues, come back and let me know

diegomtnz, 2024-05-30:
https://github.com/bigtreetech/Eddy/assets/15325404/349f05ee-21c8-464f-aa40-3d6c529718ce

you can see in video,thanks

diegomtnz, 2024-05-30:
SOLVED:
I need z offset calibration,my printer.cfg now:
#*# <---------------------- SAVE_CONFIG ---------------------->
#*# DO NOT EDIT THIS BLOCK OR BELOW. The contents are auto-generated.
#*#
#*# [probe_eddy_current btt_eddy]
#*# reg_drive_current = 16
#*# z_offset = 0.850
#*# calibrate =
#*# 	0.050000:3219502.999,0.100000:3218637.895,0.150000:3217813.333,
#*# 	0.200000:3217004.405,0.250000:3216097.985,0.300000:3215259.082,
#*# 	0.350000:3214539.599,0.400000:3213774.576,0.450000:3213022.870,
#*# 	0.500000:3212292.122,0.550000:3211543.117,0.600000:3210880.607,
#*# 	0.650000:3210167.339,0.700000:3209524.022,0.750000:3208887.566,
#*# 	0.800000:3208274.048,0.850000:3207685.914,0.900000:3207083.796,
#*# 	0.950000:3206523.057,1.000000:3205914.178,1.050000:3205389.873,
#*# 	1.100000:3204884.781,1.150000:3204363.035,1.200000:3203869.725,
#*# 	1.250000:3203366.151,1.300000:3202853.605,1.350000:3202398.485,
#*# 	1.400000:3201919.492,1.450000:3201480.975,1.500000:3201053.256,
#*# 	1.550000:3200594.807,1.600000:3200165.753,1.650000:3199709.337,
#*# 	1.700000:3199286.008,1.750000:3198910.560,1.800000:3198514.903,
#*# 	1.850000:3198150.712,1.900000:3197758.758,1.950000:3197390.839,
#*# 	2.000000:3197031.282,2.050000:3196679.911,2.100000:3196353.086,
#*# 	2.150000:3196030.354,2.200000:3195703.645,2.250000:3195367.403,
#*# 	2.300000:3195083.491,2.350000:3194790.406,2.400000:3194493.079,
#*# 	2.450000:3194226.396,2.500000:3193900.197,2.550000:3193628.500,
#*# 	2.600000:3193342.894,2.650000:3193092.407,2.700000:3192836.853,
#*# 	2.750000:3192572.462,2.800000:3192304.585,2.850000:3192034.831,
#*# 	2.900000:3191823.611,2.950000:3191589.977,3.000000:3191351.190,
#*# 	3.050000:3191152.487,3.100000:3190903.756,3.150000:3190681.405,
#*# 	3.200000:3190468.077,3.250000:3190259.802,3.300000:3190050.001,
#*# 	3.350000:3189850.527,3.400000:3189653.491,3.450000:3189470.947,
#*# 	3.500000:3189255.185,3.550000:3189065.961,3.600000:3188919.215,
#*# 	3.650000:3188703.005,3.700000:3188511.192,3.750000:3188373.851,
#*# 	3.800000:3188182.358,3.850000:3188007.329,3.900000:3187863.877,
#*# 	3.950000:3187703.384,4.000000:3187579.746,4.050000:3187403.387
#*# calibration_temp = 41.480823

ifreislich, 2024-05-30:
Glad you figured it out. I was trying to tell you that you missed this step.

diegomtnz, 2024-05-30:
> Me alegro de que lo hayas descubierto. Estaba tratando de decirte que te perdiste este paso.

thanks

diegomtnz, 2024-05-30:
> Glad you figured it out. I was trying to tell you that you missed this step.

thanks

Lebeao, 2024-05-31:
Hello, I also need your help, I'm starting to despair, I'm dealing with exactly the same problem "Must calibrate probe_eddy_current first" until then everything is clear, 
what happens next with the step "Z Offset Calibration" if that's all When I enter "PROBE_EDDY_CURRENT_CALIBRATE CHIP=btt_eddy" I get the default value 290mm z axis, what happens next?The 290mm is the max height of the z axis, if I enter a different value here it will be displayed. What am I doing wrong, please give me a tip what I can and should do, I'm new to klipper and have just finished building the Voron.
sorry for my English, I had it translated.

best regards Marco
 
[image]
[image]
[image]

ifreislich, 2024-05-31:
You need to complete the `PROBE_EDDY_CURRENT_CALIBRATE CHIP=btt_eddy` step. It doesn't know where zero is so use the dialog and paper test to move the bed or nozzle to set the zero position.

closed 2024-06-04 by BTTJUSTIN