bigtreetech/Eddy #23: Error during homing z: Eddy current sensor error

lusl13, 2024-05-31:
when is eddy mcu temp below 40°C, works fine. but when is above 40°C, is problem with homing z axis. console says "Error during homing z: Eddy current sensor error" 
[klippy (3).log](https://github.com/user-attachments/files/15515403/klippy.3.log)

ifreislich, 2024-05-31:
What's your value for `reg_drive_current` in the saved config section? The autocalibration doesn't pick the best value. You probably need to increase it by 1, but changing the drive current requires full Z offset and temperature drift calibration.

krautech, 2024-05-31:
> when is eddy mcu temp below 40°C, works fine. but when is above 40°C, is problem with homing z axis. console says "Error during homing z: Eddy current sensor error" 
> [klippy (3).log](https://github.com/user-attachments/files/15515403/klippy.3.log)
> 

I'll look through your klippy logs when I'm at a PC.

In the meantime go over your USB connections and reinstall the firmware from the start to see if it makes a difference.

Is it consistent? Like anything over 40c it won't home? Or just sometimes?
Does it ever happen UNDER 40c?

krautech, 2024-05-31:
> What's your value for `reg_drive_current` in the saved config section? The autocalibration doesn't pick the best value. You probably need to increase it by 1, but changing the drive current requires full Z offset and temperature drift calibration.

Changing the drive current after he has done temperature drift will throw out all the temperature drift calculations, I don't recommend this.

ifreislich, 2024-05-31:
> Changing the drive current after he has done temperature drift will throw out all the temperature drift calculations, I don't recommend this.

It can always be set back to the original value if it doesn't solve the higher temperature homing.

lusl13, 2024-06-01:
@ifreislich reg_drive_current = 15 
@krautech 

37°C ok
39°C ok
40°C ok
41°C ok
42°C ok 
43°C ok
44°C ok
45°C ok
46°C not ok - Error during homing z: Eddy current sensor error
47°C not ok - Error during homing z: Eddy current sensor error

everything worked fine with klipper version v0.11, but after update to v0.12 have this problem. 

i cant make temp compensation calibration. when i send " PROBE_DRIFT_CALIBRATE PROBE=btt_eddy TARGET=80 STEP=5" to console, klipper want to z-offset calibration..

i dont use this command:
git remote add eddy https://github.com/bigtreetech/klipper
git fetch eddy
git checkout eddy/eddy

because when i use that command, klipper wont start because klipper in that btt url is older than my mcus

krautech, 2024-06-01:
I think I've seen your messages via Facebook

You need to go and update all your mcus with the BTT version of klipper. So run BTT klipper, then go and do all the make menuconfig etc to rebuild your firmware for your devices using the BTT version of klipper as the base

lusl13, 2024-06-01:
@krautech i updated all to 
host: v0.12.0-166-g957475421-dirty
all mcus: v0.12.0-166-g957475421

now testing...
47°C ok
50°C ok
60°C ok

so.. its working fine. thank you

krautech, 2024-06-01:
Great to hear!! Glad we could get it sorted for you.

BTTJUSTIN, 2024-06-04:
Hi，it is glad to hear that you figured it out

closed 2024-06-04 by BTTJUSTIN