bigtreetech/Eddy #26: Need to home Z to calibrate, need to calibrate to home Z

helloITdepartment, 2024-06-06:
I'm up to section 3, step 22 of the readme, where it says to run `PROBE_EDDY_CURRENT_CALIBRATE chip=btt_eddy`. Doing so gives the familiar gui prompt with the + and - to move the nozzle up and down, but clicking either aborts and gives me the error `Must home axis first: 165.000 130.000 10.300 [0.000]`. Fair enough, I'll just home all the axes. But when it gets to homing Z, it gives me the error `Must calibrate probe_eddy_current first` (homing X and Y without Z work fine) which seems to be this endless cycle. Unless I'm missing something? Either from the documentation or otherwise?

[image]

krautech, 2024-06-06:
G28 X Y

And then

PROBE_EDDY_CURRENT_CALIBRATE chip=btt_eddy

helloITdepartment, 2024-06-06:
Yeah I did that, sorry the G28 X Y was cut off in the first image. Just ran it again, and still same issue

[image]

krautech, 2024-06-06:
Do you have any custom homing macros or anything?

helloITdepartment, 2024-06-06:
Nope, just double checked. I'm trying to figure out the issue, but I can't seem to see from the code when the z homing was expected to happen. Unless the manual z offset isn't supposed to need calibration? But that doesn't sound right

helloITdepartment, 2024-06-07:
Thought it might have had to do with the fact that I previously had a BLTouch installed, but I removed all reference to it in the printer.cfg and even just unplugged it just to be sure. Should I reinstall the original Z endstop microswitch just so it has something to home Z against? Does Eddy not do Z homing? Isn't that the point of the `endstop_pin: probe:z_virtual_endstop` line?

krautech, 2024-06-07:
It definitely does z homing..I use it all day.

Something in your config is causing it to register the z height wrong, which throws out the calibration from being able to be adjusted.

Can you upload your config? Thanks

helloITdepartment, 2024-06-07:
Sure, here's both
I have the file for eddy separated out for modularity so I've included both (also made into txt so GH will accept it)
[eddy.txt](https://github.com/user-attachments/files/15737527/eddy.txt)
[printer.txt](https://github.com/user-attachments/files/15737528/printer.txt)

krautech, 2024-06-07:
remove `z_offset: 1.0` from eddy config and also you have 

`[bed_mesh]`
and
`[safe_z_home]`

sections in both your configurations, choose whichever is more appropriate for your setup.
(My advice, use the one in Eddy but adjust your mesh_min and mesh_max to your appropriate size.

BTTJUSTIN, 2024-06-07:
Thanks for reaching out. Our support team does not operate using the GitHub platform but we would be happy to help you via
email based ticketing system. Please visit https://biqu.equipment/pages/technical-support and submit a ticket and we will be in touch with you soon

krautech, 2024-06-07:
> No luck unfortunately I'm just trying to understand the process. When running the `PROBE_EDDY_CURRENT_CALIBRATE CHIP=btt_eddy`, is it first supposed to go down and probe Z to home and then figure out what the offset it? I think that's how I remember the probe_calibrate for the BLTouch working, but my Eddy doesn't do that

No, its supposed to stay motionless at the location you start the calibration and then youre meant to adjust it down to paper height yourself. Its slightly different to the BL touch calibration

helloITdepartment, 2024-06-07:
Hmm curious. Then I guess the question is why does it require Z to be homed if it's just going adjust it with the offset? Or rather, how is it supposed to know where Z0 is? Does the Eddy just take a reading from wherever it is and go "oh looks like the bed is 20mm away"? In which case wouldn't some calibration need to happen first?

krautech, 2024-06-07:
Painfully annoying but yes lol 

You're supposed to home X and Y and then Z will be at "0" then you lower it to your lowest Z point you have configured.

So start the calibration as close to the bed as you can with a little bit of wiggle room. Otherwise you'll hit the movement out of range error.

krautech, 2024-06-07:
I have a print finishing in an hour or so, once its finished I'll video the process.

krautech, 2024-06-07:
Could you try comment out your

`screw_tilt_adjust`

and see if it makes any difference

helloITdepartment, 2024-06-07:
> Painfully annoying but yes lol
> 
> You're supposed to home X and Y and then Z will be at "0" then you lower it to your lowest Z point you have configured.
> 
> So start the calibration as close to the bed as you can with a little bit of wiggle room. Otherwise you'll hit the movement out of range error.

Then I think we've narrowed down the problem. Mine doesn't seem to do that bit where it says "okay we're doing the calibration, let me mark this position as Z0 and tell the system that Z is homed"

PavelTajdus, 2024-06-09:
You can use Klippers `SET_KINEMATIC_POSITION Z=0` and even with negative number, so than your printer think its homed. But be carefull with moving the printers toolhead.

https://www.klipper3d.org/G-Codes.html?h=set_kinematic_position#set_kinematic_position

closed 2024-06-14 by BTTJUSTIN

helloITdepartment, 2024-06-16:
> You can use Klippers `SET_KINEMATIC_POSITION Z=0` and even with negative number, so than your printer think its homed.

So this worked to get the Eddy to calibrate. Then I `SAVE_CONFIG`ed, and went to rehome XYZ to see if it worked, and:
`Error during homing z: Eddy current sensor error`

[image]

Even went back and recalibrated the current (section 2 step 17) and still same issue

krautech, 2024-06-16:
I'd stop bothering

There's too many issues with BTT eddy to devote time and energy to it, get a beacon or my preference cartographer, and be done with it.