bigtreetech/Eddy #27: Heat problems or error in the temperature compensation

Sentinel-sys, 2024-06-07:
I use the EDDY USB in my Voron 2.4, where it replaces the Z-Endstop as well as my Klicky NG.
Installation was very easy and the setup, although extensive, was very well documented.
Unfortunately, I seem to have a problem that I can't quite put my finger on.
When cold, everything works perfectly. Homing, QGL and also the creation of a heightmap. But as soon as I heat up and the EDDY gets hotter, the problems start. Depending on what I am doing I get different error messages, but it always boils down to the fact that the EDDY can no longer read the Z-height correctly.
For example, I have now done a temperature calibration for the fifth time and heated up the pressure chamber in 5C° steps and always leveled it manually in between. I do this until the EDDY shows me 70 C° because I want to heat the heating bed and the extruder quite high (ABS, ASA, etc.). Then I save the config and restart Klipper.
If I now try homing while the print chamber is still hot, I get the following error: “Endstop z still triggered after retract”
Similar when I want to print. Everything heats up, I want to do a final QGL and EDDY doesn't seem to be able to recognize the Z-height correctly.

Is this a known problem? And does it have something to do with the temperature compensation or is the EDDY generally too sensitive to heat?

krautech, 2024-06-07:
Hey,

Could you please share your printer.cfg and any other Configs please?

I'll take a look for you

Sentinel-sys, 2024-06-07:
Hi,

@krautech 

here the printer.cfg

`# This file contains common pin mappings for the Fysetc Spider board.
# To use this config, the firmware should be compiled for the STM32F446.
# When calling "menuconfig", enable "extra low-level configuration setup"
# and select the "12MHz crystal" as clock reference
# For flashing, write the compiled klipper.bin to memory location 0x08000000

# See docs/Config_Reference.md for a description of parameters.

## Voron Design VORON2 250/300/350mm Spider TMC2209 UART config

## *** THINGS TO CHANGE/CHECK: ***
## Spider version                       Please search this file for "SpiderVersion" and uncomment the appropriate line for each
## MCU paths                            [mcu] section
## Thermistor types                     [extruder] and [heater_bed] sections - See 'sensor types' list at end of file
## Z Endstop Switch location            [safe_z_home] section
## Homing end position                  [gcode_macro G32] section
## Z Endstop Switch  offset for Z0      [stepper_z] section
## Probe points                         [quad_gantry_level] section
## Min & Max gantry corner postions     [quad_gantry_level] section
## PID tune                             [extruder] and [heater_bed] sections
## Fine tune E steps                    [extruder] section
[include mainsail.cfg]
#[include KAMP/*cfg]
[include clean_nozzle.cfg]
[include led-effects.cfg]
#[include klicky-probe.cfg]
#[include autoz.cfg]

[mcu eddy]
serial: /dev/serial/by-id/usb-Klipper_rp2040_45474E621B0B035A-if00

[temperature_sensor btt_eddy_mcu]
sensor_type: temperature_mcu
sensor_mcu: eddy
min_temp: 10
max_temp: 100

[probe_eddy_current btt_eddy]
sensor_type: ldc1612
z_offset: 1.0
#i2c_address:
i2c_mcu: eddy
i2c_bus: i2c0f
x_offset: 0 # Set according to the actual offset relative to the nozzle
y_offset: 20 # Set according to the actual offset relative to the nozzle
data_rate: 500

[temperature_probe btt_eddy]
sensor_type: Generic 3950
sensor_pin: eddy:gpio26
horizontal_move_z: 2

[bed_mesh]
horizontal_move_z: 2
speed: 300
mesh_min: 10, 10
mesh_max: 270, 270
probe_count: 9, 9
algorithm: bicubic

[safe_z_home]
home_xy_position: 150, 150
z_hop: 10
z_hop_speed: 25
[... 824 lines cut ...]
#*# points =
#*# 	-0.048009, -0.045255, -0.029759, -0.050424, -0.042768
#*# 	-0.062493, -0.048896, -0.025467, -0.042343, -0.024506
#*# 	-0.052223, -0.031892, 0.000000, -0.001135, -0.006808
#*# 	-0.065935, -0.037047, -0.004500, -0.030573, -0.037214
#*# 	-0.056955, -0.017280, 0.004712, -0.003812, -0.007616
#*# x_count = 5
#*# y_count = 5
#*# mesh_x_pps = 2
#*# mesh_y_pps = 2
#*# algo = bicubic
#*# tension = 0.2
#*# min_x = 40.0
#*# max_x = 260.0
#*# min_y = 40.0
#*# max_y = 260.0
`

Sorry, maybe a little mixed up. I'm just in the process of organizing everything and possibly optimizing it in one place or another.

What other files do you need?

Thank you very much

Joe

Sentinel-sys, 2024-06-07:
[config-20240607-164011.zip](https://github.com/user-attachments/files/15741242/config-20240607-164011.zip)

krautech, 2024-06-07:
Make sure you comment out

`[probe]`

and dont leave it there.. has the potential to cause issues.

also remove 

`
#*# [stepper_z]
#*# position_endstop = 2.370
`

From your DO NOT EDIT section, this will definitely cause you issues.

Sentinel-sys, 2024-06-07:
Okay, changed the config. 
Cold everything is working. Now it´s heating up.

Fingers crossed... :-)

krautech, 2024-06-07:
[printer.zip](https://github.com/user-attachments/files/15741418/printer.zip)

This is all the changes i suggested, using this, rerun all steps from step 16 of the guide onwards and see how you go.

Sentinel-sys, 2024-06-07:
Okay,

I had now made the first two changes you made. I had only left #*# [probe_eddy_current btt_eddy] in.

Printer heated up and then homing and QGL. During QGL this error occurred: No trigger on probe after full movement

I will now use the printer.cfg file you provided and do the calibration process again.

Hope that works then.

Greetings
Joe

krautech, 2024-06-07:
Theres a good chance that stepper_z part in the do not edit section caused the calibration process to produce weird calibration defaults. Hopefully this will sort it otherwise I shall look deeper for you.

Sentinel-sys, 2024-06-07:
I'll just try it again from step 16 completely fresh. It just takes time to heat up and down.

krautech, 2024-06-07:
Yeah no worries.

Also i just noticed.. you have 2 x enabled `[bed_mesh]` sections with different variables, you might like to fix that

krautech, 2024-06-07:
Oh some big changes there. Yeah i'd probably start completely fresh and go through it one by one, it does seem like a wire is been crossed somewhere within the configuration but its hard to pin point.

If you need more help, reach out on discord. I use the same name there as here and I reply pretty quick. It can be slow moving here.

Sentinel-sys, 2024-06-07:
I hope you get involved with BTT when you do their support. I've been writing to support for four days and they haven't even looked at my printer.cfg yet.

Sentinel-sys, 2024-06-07:
Unfortunately, I am currently stuck at step 25. When I send the command, I get the following error:
Move out of range: 10.000 -10.000 2.005 [0.000]

krautech, 2024-06-07:
I'm not at my PC atm but I'll have a look over your configuration again.

Normally when I've seen this happen it's along the lines of your probe is too close to the bed when you're trying to calibrate. Perhaps move it a little further away first.

stubbs41, 2024-06-08:
I'm having similar issues that when cold everything is fine... when it has warmed up to say 45 or more but below 60 degrees the eddy usb becomes very unstable. I have a ratrig vcore 3 and run ztilt on 3 axis. 0.02 is my margin and cold it's fine. Hot it will fail 20 times in a row crashing into my nozzle on each run on one of the corners. it will read on one pass Z1 1.143 then 2nd pass 0.456 then the 3rd pass is 1.289 and it's typically Z1. I've gone over this for days. 

I then just home it and it crashes on homing. I've gone through everything I can think of and what I can bug others to reach out with. I'm happy to assist in anyway possible. I have one more Eddy I was going to install on my Voron 0 for fun.
[klippy (11).log](https://github.com/user-attachments/files/15746065/klippy.11.log)
log is uploaded should include the weird info along with me just testing probe a bunch.

krautech, 2024-06-08:
Temperature compensation is useless in its current state. I wouldn't run BTT eddy on anything with QGL or Bed Tilt until this is all resolved 

@stubbs41 I think your issue is slightly different in that you can pass calibration and @Sentinel-sys cannot.

@stubbs41 when cold QGL runs fine correct?

Sentinel-sys, 2024-06-08:
Hi everyone,

@krautech:

So if I understand your last statement to stubbs41 correctly you think I can't complete the temperature calibration successfully. However, I can do this very well. 

> I think your issue is slightly different in that you can pass calibration and @Sentinel-sys cannot.

What does not work is homing or QGL in hot state afterwards.

What exactly do you mean by “further away”? I have set “horizontal_move_z: 2” for homing and QGL. Or do you mean the built-in distance in the X-Carrier?

Sentinel-sys, 2024-06-08:
By the way, I have now ordered a Cartographer. I'll either use this or simply install my Klicky probe again. The BTT EDDY is simply too unreliable for me.
Nevertheless, I would be interested to know why. I would prefer to simply deactivate the temperature compensation to see if everything works. But the support team has not yet replied to my query.

Another general question. Is there actually a possibility to do an “Automatic Z-Calibration” with the EDDY sensors like it was possible with the Klicky Probe? I don't think so, do you?

looxonline, 2024-07-13:
> By the way, I have now ordered a Cartographer. I'll either use this or simply install my Klicky probe again. The BTT EDDY is simply too unreliable for me. Nevertheless, I would be interested to know [...]

The new thermal calibration and macros provide a completely stable experience through the entire temperature range. Please give them a test and let us know if you still have issues.

closed 2024-07-13 by looxonline