bigtreetech/Eddy #28: No valid measurements found at coordinate.

bigblzrd, 2024-06-11:
I've been banging my head against a wall with this thing for a couple of weeks now.

First off I had zero success with the temperature calibration and persevered with it for days and it wouldn't keep its Z offset reliably or more often than not would just error on its first home with the bed at temp.

After giving up with that I'm now using a Z endstop and I thought after installing that, everything would be smooth sailing. Instead I'm constantly met with a 'No valid measurements at coordinate' error every single time I do a mesh. It's definitely not because its too close to an edge, I've tried countless values for my bed mesh min and max XY but without fail it throws the error wherever the toolhead starts the mesh.

I'm completely out of ideas to find a solution for this, I can't even find documentation on the error itself at it doesnt appear to exist in the source code for BTTs fork of Klipper or the main branch.

[image]

krautech, 2024-06-11:
Hey, sounds about right lol I'll try help where I can but unfortunately you're not going to get any help from BTT here other than to "email a ticket to them"

- What printer is it?
- What bed type are you using? 
- Does it have a giant magnetic sticker? Small magnets?
- When you place your toolhead over X20 Y25 can you show me a photo of where it sits?
- Can you please upload your printer.cfg (you'll need to compress it as a zip to upload it)

I'm not actively working on Eddy anymore, as I think BTT has more or less abandoned it until Kevin @ Klipper implements his probe scripts fully and I personally had way to many issues for it to be worth it not to mention the 30-40 people I've know helped get it "running" with work arounds and missing features.

bigblzrd, 2024-06-11:
> Hey, sounds about right lol I'll try help where I can but unfortunately you're not going to get any help from BTT here other than to "email a ticket to them"
>
>     * What printer is it?
>
>     * What [...]

Hey man! 

It's an Enderwire with the standard Creality bed and full magnetic sheet (not embedded magnets).

I've played with moving the min XY in loads to make sure it wasn't caused by it being close to an edge and yeah wherever the minimums set it'll always error :(

My config might be a little bit messy in some areas where I've been experimenting with things so I apologise in advance. Every sections labeled though hahah

[image]

[printercfg.zip](https://github.com/user-attachments/files/15794665/printercfg.zip)

krautech, 2024-06-11:
Even if you set 

```
mesh_min: 100,100
mesh_max: 180,180
```

Does it give the same no measurement?

bigblzrd, 2024-06-11:
> Even if you set
> 
> ```
> mesh_min: 100,100
> mesh_max: 180,180
> ```
> 
> Does it give the same no measurement?

Yep so I just tried and it errors here...

```No valid measurements found at coordinate : [100.0, 75.0]```

krautech, 2024-06-11:
Weird, it shouldn't even be scanning 100, 75

Hmm,

My only real suggestion is starting completely fresh with your printer.cfg
Which, as someone with a similar modified printer, is easier said than done.
Honestly, unless someone has an immediate solution, I wouldn't bother
trying to wrap your head around it anymore unless you don't care if your
printer is unusable?

Could be firmware, could be hardware. The error code doesn't help at all,
except to say it's not getting anything whilst scanning lol

My final thought is this..

Delete [bed_mesh] and [probe_eddy_current] sections from the very bottom of
your printer.cfg

Redo the calibration steps for eddy. DONT do the temperature compensation
steps. But you must do the z height calibration which looking at it, you
seem to not have configured.

Then once that's finished, it will add a more detailed [probe_eddy_current]
section, and then try run a bed mesh.

bigblzrd, 2024-06-11:
> Weird, it shouldn't even be scanning 100, 75 Hmm, My only real suggestion is starting completely fresh with your printer.cfg Which, as someone with a similar modified printer, is easier said than [...]

See I wondered about the Z calibration stuff, if I'm using a Z endstop is that still necessary? 

Until the thermal stuff gets sorted I planned to treat it as if I had the Coil version. As you said though I think I'll just have to start everything Eddy and bed mesh related from scratch and pray something fixes itself. I'll switch back to the main Klipper build and completely start again with reflashing etc.

If none of that works then I'm well and truly lost lol. I'll give it a go in the morning (It's midnight here) and report back if I get any further.

Really appreciate the input so far though, thank you!

krautech, 2024-06-11:
Yeah you need to complete all steps up until step 26.

It still needs to know where eddy is in relation to your nozzle so that it can provide a bed mesh.

So my guess is now 90% this is what you're missing from been able to complete a bed mesh

bigblzrd, 2024-06-12:
> Yeah you need to complete all steps up until step 26.
> 
> It still needs to know where eddy is in relation to your nozzle so that it can provide a bed mesh.
> 
> So my guess is now 90% this is what you're missing from been able to complete a bed mesh

So update, I've got the thing printing. 

So the issue turned out to be exactly what you said, all I needed to do was calibrate the Z offset of the probe and I haven't actually touched anything else or reflashed. However I've discovered if I do that Z calibration with the bed cold, as soon as it's at temp and I do a mesh it throws the error. If I do the Z calibration at temp it meshes fine so I'm now wondering if I should just live with it and hope it stays consistent at meshing when its at temp or still start from scratch and maybe iron that small caveat out.

In all honesty I don't feel like I'll be able to get it much better so while I wait for the full Klipper implementation I think it's best to just cross my fingers and make the most of it while it works hahah

krautech, 2024-06-12:
Glad to hear it's mostly sorted!!

Calibrate it while at your print temp, that's what I did while still using eddy and for the most part it worked.

BTTJUSTIN, 2024-06-14:
Thanks for reaching out. Our support team does not operate using the GitHub platform but we would be happy to help you via email based ticketing system. Please visit https://biqu.equipment/pages/technical-support and submit a ticket and we will be in touch with you soon.

JustACrouton, 2024-06-27:
I had the same problem. idk what fixed it but make sure "horizontal_move_z: 2"
and try running the commands in the Klipper documentations instead of the BTT ones. ("BED_MESH_CALIBRATE METHOD=scan")

closed 2024-07-13 by looxonline