bigtreetech/Eddy #29: An idea - use the rapid scan idea for gantry levelling.

dangerzau, 2024-06-12:
Had a thought....  If rapid scan can rapidly scan from left to right and send that data to the host MCU... Could the probe stay in the same x and y position and for those of us with say a gantry and dual z motors take a single reading of height from say the left side.  Move to the right, then stay there in x and y and move the left z motors up or down to level the gantry to the bed accurately and rapidly.  Rather than having to go back and forth, I mean sure go back and check it has aligned properly but this seems like such a simple feature I wish I had the smarts to implement it myself :)

krautech, 2024-06-12:
It will always have to go back and forth to take a new height but here's a fast QGL macro

krautech, 2024-06-12:
```
[gcode_macro QUAD_GANTRY_LEVEL]
rename_existing: BASE_QUAD_GANTRY_LEVEL
gcode:
        BASE_QUAD_GANTRY_LEVEL RETRY_TOLERANCE=1
        BASE_QUAD_GANTRY_LEVEL horizontal_move_z=2
        
```

BTTJUSTIN, 2024-06-14:
Thanks for making this suggestion. We will discuss it internally and let you know if it will be scheduled for a future firmware release

BTTJUSTIN, 2024-09-29:
After an internal discussion we decided not to implement.

closed 2024-09-29 by BTTJUSTIN

bigtreetech, 2024-09-29:
To elaborate, there is little that can be done to optimise the existing klipper implementation. It is already highly optimal and resampling is always required.