bigtreetech/Eddy #30: Klipper lost communication with MCU 'eddy'

diegostf, 2024-06-12:
During a long print (more than an hour), klipper loses communication with Eddy (Klipper reports shutdown lost communication with MCU Eddy).
I'm using Eddy on the Hurakan.
After the error, it is possible to get the usb id of the eddy via this command ls /dev/serial/by-id at the SSH interface (before restarting klipper).

mcu(stm32g0b1xx)
Version: v0.11.0-479-g95747542

mcu eddy(rp2040)
Version: v0.11.0-479-g95747542

Host(aarch64, 64bit)
Version: v0.11.0-479-g95747542
OS: Debian GNU/Linux 11 (bullseye)

[klippy.log](https://github.com/user-attachments/files/15807572/klippy.log)

krautech, 2024-06-12:
Hmm went over you klippy logs, can't see anything out of the ordinary.

Temps look fine 

- home your printer
- when it's in standby, physically move your canbus cable especially at the connector as close to eddy as possible, does it force a disconnect/lost communication?

Are you using eddy USB or eddy coil?
What mainboard and toolhead are you using?

diegostf, 2024-06-12:
> Hmm went over you klippy logs, can't see anything out of the ordinary.
> 
> Temps look fine
> 
> * home your printer
> * when it's in standby, physically move your canbus cable especially at the connector as close to eddy as possible, does it force a disconnect/lost communication?
> 
> What mainboard is it?
  
 Hurakan uses BIQU BTT Manta M4P. Eddy is USB version.
 I moved the cable a lot, but there was no disconnection/lost communication.

BTTJUSTIN, 2024-06-14:
Thanks for reporting. We will get our engineering team to look into this.

looxonline, 2024-07-04:
@diegostf Hey there! I am trying to track down people who are having this issue. Is it still happening with you? If it does, could you please type dmesg into the command line and save the response?

diegostf, 2024-07-05:
> @diegostf Hey there! I am trying to track down people who are having this issue. Is it still happening with you? If it does, could you please type dmesg into the command line and save the response?

Yes, it still has this problem. I moved the USB cable, changed the USB cable, plugged the USB cable directly into the MANTA (BIQU HURAKAN) and nothing helped.

Should I type this command after the error occurs?

looxonline, 2024-07-05:
> Yes, it still has this problem. I moved the USB cable, changed the USB cable, plugged the USB cable directly into the MANTA (BIQU HURAKAN) and nothing helped.
>
> Should I type this command after the [...]

If you could do that and then capture the response it would be really helpful. It will help us to see if the probe is actually disconnecting from USB. If it is, then it is going to be super interesting to try and figure out why on Earth it is doing that.

diegostf, 2024-07-05:
> If you could do that and then capture the response it would be really helpful. It will help us to see if the probe is actually disconnecting from USB. If it is, then it is going to be super [...]

OK. I will do this.

For information: I had a FYSETC-Portable Input Shaper connected to USB too and it disconnected and caused the same error. I removed it to avoid problems. So the error started with EDDY.
These problems started happening after I started using the BBT Klipper fork.

diegostf, 2024-07-05:
> > > @diegostfOlá! Estou tentando rastrear pessoas que estão tendo esse problema. Isso ainda está acontecendo com você? Se acontecer, você poderia digitar dmesg na linha de comando e salvar a resposta?
> > 
> > 
> > Sim, ele ainda tem esse problema. Mudei o cabo USB, troquei o cabo USB, conectei o cabo USB diretamente no MANTA (BIQU HURAKAN) e nada ajudou.
> > Devo digitar este comando depois que o erro ocorrer?
> 
> Se você pudesse fazer isso e então capturar a resposta, seria muito útil. Isso nos ajudaria a ver se a sonda está realmente se desconectando do USB. Se estiver, então será superinteressante tentar descobrir por que diabos ela está fazendo isso.

> If you could do that and then capture the response it would be really helpful. It will help us to see if the probe is actually disconnecting from USB. If it is, then it is going to be super [...]

[image]

Kooramah, 2024-07-06:
Did you check if the cable is properly seated into the Eddy? The first day I had the eddy, I get connection issues but then it turned out that the USB wasn't properly seated into the Eddy. Gotta push all the way until it clicks.

spddqd, 2024-07-07:
Have the same issue when doing I'm doing a bunch of restarts. ADXL accelerometer, which also uses rpi chip, gets disconnected as well.
Logs with only Eddy connected after 2 FIRMWARE_RESTART:
```
klipper@armbian:~$ lsusb
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 002: ID 1a40:0101 Terminus Technology Inc. Hub
Bus 001 Device 003: ID 1a86:7523 QinHeng Electronics CH340 serial converter
Bus 001 Device 004: ID 1a40:0101 Terminus Technology Inc. Hub
Bus 001 Device 005: ID 046d:0825 Logitech, Inc. Webcam C270
Bus 001 Device 008: ID 1d50:614e OpenMoko, Inc. rp2040
klipper@armbian:~$ lsusb
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 002: ID 1a40:0101 Terminus Technology Inc. Hub
Bus 001 Device 003: ID 1a86:7523 QinHeng Electronics CH340 serial converter
Bus 001 Device 004: ID 1a40:0101 Terminus Technology Inc. Hub
Bus 001 Device 005: ID 046d:0825 Logitech, Inc. Webcam C270
```

dmesg log:
```
[74372.948450] usb 1-1.3.4: USB disconnect, device number 8
[74373.349870] usb 1-1.3.4: new full-speed USB device number 9 using xhci-hcd
[74373.557295] usb 1-1.3.4: New USB device found, idVendor=1d50, idProduct=614e, bcdDevice= 1.00
[74373.565730] usb 1-1.3.4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[74373.573434] usb 1-1.3.4: Product: rp2040
[74373.578782] usb 1-1.3.4: Manufacturer: Klipper
[74373.584024] usb 1-1.3.4: SerialNumber: 45474E621B02473A
[74373.653795] cdc_acm 1-1.3.4:1.0: ttyACM0: USB ACM device
[74382.676343] usb 1-1.3.4: USB disconnect, device number 9
[74383.113694] usb 1-1.3.4: new full-speed USB device number 10 using xhci-hcd
[74398.741424] usb 1-1.3.4: device descriptor read/64, error -110
[74414.357168] usb 1-1.3.4: device descriptor read/64, error -110
[74414.649066] usb 1-1.3.4: new full-speed USB device number 11 using xhci-hcd
[74430.232861] usb 1-1.3.4: device descriptor read/64, error -110
```

diegostf, 2024-07-07:
> Did you check if the cable is properly seated into the Eddy? The first day I had the eddy, I get connection issues but then it turned out that the USB wasn't properly seated into the Eddy. Gotta push all the way until it clicks.

Yes, I checked the connector several times. I even changed the USB cable.
But I'll try to change the connector too.

diegostf, 2024-07-08:
> @diegostf Hey there! I am trying to track down people who are having this issue. Is it still happening with you? If it does, could you please type dmesg into the command line and save the response?

Last login: Thu Jun 20 18:01:29 2024 from 10.0.0.93
biqu@Hurakan:~$ cd klipper
biqu@Hurakan:~/klipper$ dmesg
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd034]
[    0.000000] Linux version 5.16.17-sun50iw9 (root@ubuntu) (aarch64-none-linux-                    gnu-gcc (GNU Toolchain for the A-profile Architecture 9.2-2019.12 (arm-9.10)) 9.                    2.1 20191025, GNU ld (GNU Toolchain for the A-profile Architecture 9.2-2019.12 (                    arm-9.10)) 2.33.1.20191209) #2.2.1 SMP Wed Nov 9 10:39:47 HKT 2022
[    0.000000] Machine model: BQ-H616
[    0.000000] NUMA: No NUMA configuration found
[    0.000000] NUMA: Faking a node at [mem 0x0000000040000000-0x000000005fffffff                    ]
[    0.000000] NUMA: NODE_DATA [mem 0x5ffc0040-0x5ffc1fff]
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000040000000-0x000000005fffffff]
[    0.000000]   DMA32    empty
[    0.000000]   Normal   empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000040000000-0x000000004007ffff]
[    0.000000]   node   0: [mem 0x0000000040080000-0x000000005fffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000040000000-0x000000005fffffff]
[    0.000000] cma: Reserved 128 MiB at 0x0000000057400000
[    0.000000] psci: probing for conduit method from DT.
[    0.000000] psci: PSCIv1.1 detected in firmware.
[    0.000000] psci: Using standard PSCI v0.2 function IDs
[    0.000000] psci: MIGRATE_INFO_TYPE not supported.
[    0.000000] psci: SMC Calling Convention v1.2
[    0.000000] percpu: Embedded 18 pages/cpu s36440 r8192 d29096 u73728
[    0.000000] pcpu-alloc: s36440 r8192 d29096 u73728 alloc=18*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
[    0.000000] Detected VIPT I-cache on CPU0
[    0.000000] CPU features: detected: ARM erratum 845719
[    0.000000] Fallback order for Node 0: 0
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 129024
[    0.000000] Policy zone: DMA
[    0.000000] Kernel command line: root=UUID=bbeb24dc-1974-46f7-ac2c-7685cebf5a                    27 rootwait rootfstype=ext4 bootsplash.bootfile=bootsplash.armbian console=ttyS0                    ,115200 console=tty1 consoleblank=0 loglevel=7 ubootpart=69c7d102-01 usb-storage                    .quirks=   cgroup_enable=memory swapaccount=1
[    0.000000] Unknown kernel command line parameters "ubootpart=69c7d102-01 cgr                    oup_enable=memory", will be passed to user space.
[    0.000000] printk: log_buf_len individual max cpu contribution: 4096 bytes
[    0.000000] printk: log_buf_len total cpu_extra contributions: 12288 bytes
[    0.000000] printk: log_buf_len min size: 16384 bytes
[    0.000000] printk: log_buf_len: 32768 bytes
[    0.000000] printk: early log buf free: 14064(85%)
[    0.000000] Dentry cache hash table entries: 65536 (order: 7, 524288 bytes, l                    inear)
[    0.000000] Inode-cache hash table entries: 32768 (order: 6, 262144 bytes, li                    near)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 352904K/524288K available (13312K kernel code, 918K rwdat                    a, 4032K rodata, 2240K init, 279K bss, 40312K reserved, 131072K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] rcu:     RCU restricting CPUs from NR_CPUS=8 to nr_cpu_ids=4.
[    0.000000]  Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 25 jif                    fies.
[    0.000000] rcu: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=4
[    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
[    0.000000] Root IRQ handler: gic_handle_irq
[    0.000000] GIC: Using split EOI/Deactivate mode
[    0.000000] random: get_random_bytes called from start_kernel+0x4c4/0x6a4 wit                    h crng_init=0
[    0.000000] arch_timer: cp15 timer(s) running at 24.00MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles:                     0x588fe9dc0, max_idle_ns: 440795202592 ns
[    0.000001] sched_clock: 56 bits at 24MHz, resolution 41ns, wraps every 43980                    46511097ns
[    0.000572] Console: colour dummy device 120x45
[    0.000921] printk: console [tty1] enabled
[    0.001004] Calibrating delay loop (skipped), value calculated using timer fr                    equency.. 48.00 BogoMIPS (lpj=96000)
[    0.001029] pid_max: default: 32768 minimum: 301
[    0.001111] LSM: Security Framework initializing
[    0.001153] Yama: becoming mindful.
[    0.001242] AppArmor: AppArmor initialized
[    0.001292] Mount-cache hash table entries: 1024 (order: 1, 8192 bytes, linea                    r)
[    0.001313] Mountpoint-cache hash table entries: 1024 (order: 1, 8192 bytes,                     linear)
[    0.002888] rcu: Hierarchical SRCU implementation.
[    0.004033] smp: Bringing up secondary CPUs ...
[    0.004525] Detected VIPT I-cache on CPU1
[... 402 lines cut ...]
[    8.842526] rc rc0: lirc_dev: driver sunxi-ir registered at minor = 0, raw IR                     receiver, no transmitter
[... cut ...]
[   14.885278] usb 2-1.2: Manufacturer: Klipper
[   14.885285] usb 2-1.2: SerialNumber: 45474E621A8637CA
[   14.886496] cdc_acm 2-1.2:1.0: ttyACM1: USB ACM device
[   15.173270] sunxi-gmac 5030000.ethernet eth0: eth0: Type(7) PHY ID 00441400 a                    t 0 IRQ poll (5030000.ethernet-0:00)
[   20.433887] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[   35.054502] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[   37.469982] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready

bigtreetech, 2024-07-09:
@diegostf @spddqd We have a new cable that we would like to send to you for testing. We believe that it will solve the issue. Please could you send me an email at luke@bigtree-tech.com so that we can make the arrangements?

diegostf, 2024-07-09:
> @diegostf @spddqd We have a new cable that we would like to send to you for testing. We believe that it will solve the issue. Please could you send me an email at [luke@bigtree-tech.com](mailto:luke@bigtree-tech.com) so that we can make the arrangements?

I have already changed the Eddy cable 5 times (good quality cables, measuring between 1 and 3m). Everyone had the same problem.

I will send a mail to your. I would really like to resolve this problem.

bigtreetech, 2024-07-09:
> I have already changed the Eddy cable 5 times (good quality cables, measuring between 1 and 3m). Everyone had the same problem.
>
> I will send a mail to your. I would really like to resolve this [...]

Did your replacement cables have an internal shield?

diegostf, 2024-07-09:
> Did your replacement cables have an internal shield?

Yes, sure. I can send photo to you.

bigtreetech, 2024-07-09:
> Yes, sure. I can send photo to you.

Yes, please. We can communicate over email.

diegostf, 2024-07-10:
> Yes, please. We can communicate over email.

I sent you an email yesterday.

Corbin1747, 2024-07-10:
> I sent you an email yesterday.

If this fixes the issue, will BTT be sending out replacement cables to all those who have ordered one?

andremoares, 2024-07-13:
Help install eddy usb in biqu hurakan manta mp4

looxonline, 2024-07-13:
> If this fixes the issue, will BTT be sending out replacement cables to all those who have ordered one?

Only to those who actually have a disconnection issue.

looxonline, 2024-07-13:
> Help install eddy usb in biqu hurakan manta mp4

Thanks for reaching out. This is not a thread that is dealing with your issue. For installation support please use the ticketing tool on biqu.equipment as we do not offer this kind of support via github.

Brenddpaula, 2024-07-28:
Hey, im having this same issue. My unit was working fine, i did some prints with no issues. Then out of nowhere it started to disconnect. I've checked the cable and the connection and there was no problem in sight.
Now i'm constantly receiving this error and cannot finish any prints.

FiroRos, 2024-07-28:
any updates @bigtreetech ? its not the cable because I soldered it straight to eddy's board and still got the same issue so its either a problem with the mcu itself, or a software bug.
I ordered a new eddy sensor too and Im currently waiting for it to arrive.

bigtreetech, 2024-08-12:
> any updates @bigtreetech ? its not the cable because I soldered it straight to eddy's board and still got the same issue, so its either a problem with the mcu itself, or a software bug. I ordered a new eddy sensor too and Im currently waiting for it to arrive.

For customers experiencing this issue, it is cable related. Their machines have too much EMI noise for USB and the new cable shields that noise out. If you are having this issue then we can send you a new cable. Just send a message to service001@biqu3d.com with the subject line: "Request for new EDDY USB cable". Explain that you are getting Eddy disconnect issues and include klipper logs in the email that show the issue happening. They will send you the new cable.

hudrucan, 2024-10-06:
> > any updates @bigtreetech ? its not the cable because I soldered it straight to eddy's board and still got the same issue, so its either a problem with the mcu itself, or a software bug. I ordered a new eddy sensor too and Im currently waiting for it to arrive.
> 
> 
> 
> For customers experiencing this issue, it is cable related. Their machines have too much EMI noise for USB and the new cable shields that noise out. If you are having this issue then we can send you a new cable. Just send a message to service001@biqu3d.com with the subject line: "Request for new EDDY USB cable". Explain that you are getting Eddy disconnect issues and include klipper logs in the email that show the issue happening. Also include the store that you purchased it from, the date of purchase and the order number. They will send you the new cable.

I have the same issue, but not sure I can get this cable for testing or not cuz I bought it of reseller in Vietnam.
Could you provide us more details about the new cable for us can find it localy?

bigtreetech, 2024-10-07:
> > > any updates @bigtreetech ? its not the cable because I soldered it straight to eddy's board and still got the same issue, so its either a problem with the mcu itself, or a software bug. I ordered a new eddy sensor too and Im currently waiting for it to arrive.
> > 
> > 
> > For customers experiencing this issue, it is cable related. Their machines have too much EMI noise for USB and the new cable shields that noise out. If you are having this issue then we can send you a new cable. Just send a message to [service001@biqu3d.com](mailto:service001@biqu3d.com) with the subject line: "Request for new EDDY USB cable". Explain that you are getting Eddy disconnect issues and include klipper logs in the email that show the issue happening. Also include the store that you purchased it from, the date of purchase and the order number. They will send you the new cable.
> 
> I have the same issue, but not sure I can get this cable for testing or not cuz I bought it of reseller in Vietnam. Could you provide us more details about the new cable for us can find it locally?

Send the email to the address and the service team should get you sorted.

hakankok, 2024-11-24:
I have an eddy sensor that I have been using for about 5 months.The sensor works via USB connected to the Manta M8P platform.However, recently, after performing the table calibration, it stops the printing process by giving a message that the connection is lost during the printing process.When I connect via ssh, it appears that the id is registered. Can you help me solve the problem?

I wonder if the problem was solved for the people you sent new cables to?

idiotsguide, 2024-11-28:
> I have an eddy sensor that I have been using for about 5 months.The sensor works via USB connected to the Manta M8P platform.However, recently, after performing the table calibration, it stops the [...]

I am having the same issue and it also just started recently. I have not reached out for a new cable yet but wanted to at least add my experience to the list.

@bigtreetech it also really sucks that the board uses an itty bitty connector that's not common instead of USB (as in the USB version) or something that someone might have their own crimp connectors for in their drawer already. Can you explain what exactly is wrong with the cable provided? I'd like to source my own but I would like to know the exact issue and difference between the cables. The USB cable that came with mine appears to be heavily shielded (very thick cable and somewhat stiff, I'm assuming it's shielding) with the exception of the 1 inch section of wire that has no shielding covering it in order to crimp the tiny connector on. Can you share what's different about the new cable?

638Alex638, 2024-12-01:
Anyone who has already received the new cable, has the Eddy disconnection problem gone away?

user-42, 2024-12-02:
> Anyone who has already received the new cable, has the Eddy disconnection problem gone away?
> 
> I am always happy when Eddy turns off when the printer is turned on - in this case, after rebooting the printer, it can print for many hours without problems. But lately Eddy increasingly turns off during printing. If a small detail is printed, then this is not so bad, but when printing large parts with a print time of several hours (3-5), it is very annoying when the error occurs after 1-2 hours. Support sent me a new cable, I really hope that it will help solve the problem. Otherwise, I will have to remove Eddy and put back the stock slow sensor (

Hey there! Have you already tried new cable?

638Alex638, 2024-12-03:
> Привет всем! Вы уже пробовали новый кабель?

Hi.
Haven't tried it yet. It was sent to me just 2 days ago and I'll probably get it in about 2 weeks.

user-42, 2024-12-04:
Hey guys!

Here is a fix for EMI noise that cause errors and disconnects: use ferrite cores for cables if you can't get shielded cable.

[image]

I also had errros on long prints and ferrite cores seem to fix.  I placed two cores on the eddy's cable.

Andyman14159, 2024-12-06:
Yes I am having this exact same problem! Can someone from BTT please reach out?   -->  stephens45336@gmail.com
Manta E3Ez W/CB1 and BTT HDMI5 & EDDY USB
Same lost connection with MCU EDDY during any print job, multiple repeated failures. All connections tested and good.
Problem has gotten progressively worse over time.

user-42, 2024-12-07:
> Yes I am having this exact same problem! Can someone from BTT please reach out? --> [stephens45336@gmail.com](mailto:stephens45336@gmail.com) Manta E3Ez W/CB1 and BTT HDMI5 & EDDY USB Same lost connection with MCU EDDY during any print job, multiple repeated failures. All connections tested and good. Problem has gotten progressively worse over time.

Try ferrite cores.

638Alex638, 2024-12-09:
While the new cable is still on its way, I ordered ferrite cores. Yesterday I installed them and printed for about 4 hours. During this time, the connection with the Eddy MCU was lost only once. I could have this error once in 5-6 hours before, or it could appear 1-2 times per hour. I will continue to observe.

idiotsguide, 2024-12-19:
I've sent the email as per instructions above in hopes of getting a replacement cable. One thing I'm curious about, why does the Eddy need to talk to the printer at all during a print? I run the homing, Z tilt, and mesh sequence before the print and that's the last time it gets used until the next print. Is there some way it can be ignored during a print or put in a state where it's not constantly talking to klipper until it's needed again?

idiotsguide, 2024-12-20:
FYI for anyone who sees the directions above for getting a replacement USB cable from BIQU/BTT, they now respond telling you to submit a ticket instead. Here's the response I got with the link to the next hoop to jump through in case you want to save time. Looks like I should have just purchased a different product from a different company.

Hi，

We apologize for the inconvenience.

Please submit a ticket https://biqu.equipment/pages/technical-support

Looking forward to your reply.

Best regards

638Alex638, 2025-02-10:
I found the time and replaced the cable with a new one sent by tech support. I had previously compared it with the cable that came with the sensor, but I found no difference. Both cables are the same diameter, both have the same number of cores, and there are no shielding applets. I did not understand where it is more protected from interference, as the manufacturer claims.

I have two ferrite filters installed on the old and new cables. The number of errors associated with loss of connection with the sensor's MCU has significantly decreased, but they have not completely disappeared. I have a Neptune 4. The printer can print all day without errors, or it can lose connection with the sensor after 1-3 hours of printing and the part goes to the trash. I have been monitoring the printer for about 2 months and am increasingly inclined to abandon the eddy and return to the slow stock sensor (

Corbin1747, 2025-02-11:
> I found the time and replaced the cable with a new one sent by tech support. I had previously compared it with the cable that came with the sensor, but I found no difference. Both cables are the [...]

I've had the new cable installed for a few months and it's not failed once. I am on main line klipper and updated a few weeks ago. 
Here is how I've got it set up. No ferrite filters are in place. 
The cable for the eddy is run outside of the main cables. 
Before the cable change I was getting errors all of the time. 

[image]

638Alex638, 2025-02-11:
> Кабель для вихретокового датчика проложен отдельно от основных кабелей.

If in this case the cable is laid separately from the other cables, what will it look like if it is laid together with the others?

My old and new cables are laid away from all the others and approach the head cables only in the head itself. In this case, errors should not occur at all, but they do occasionally, but they do occur.

Even despite this, I am generally dissatisfied with the sensor, because it has too large an error in measuring the table level. I have repeatedly calibrated in an attempt to set the ideal 3-offset. During calibration, everything is set perfectly, but as soon as I start printing, I immediately get an elephant's foot on the lower layers. If I adjust the offset minimally, the first layer stops sticking to the table. This has never happened with the stock sensor, I will have to return it and get used to a long preparation for printing (

bigtreetech, 2025-02-11:
> If in this case the cable is laid separately from the other cables, what will it look like if it is laid together with the others?
>
> My old and new cables are laid away from all the others and [...]

Use this new version of the software.

638Alex638, 2025-02-11:
> Используйте эту новую версию программного обеспечения.

Do you mean the test settings of the sensor, when it determines the offset by poking the nozzle into the table?

bigtreetech, 2025-02-11:
> > Используйте эту новую версию программного обеспечения.
> 
> Do you mean the test settings of the sensor, when it determines the offset by poking the nozzle into the table?

It will solve any thermal drift issues you may be having.

cyaneo, 2025-02-27:
I also have this issue with my Sovol SV08 Mainline - when powering on, after a short time (15-25min) the "Lost communication with MCU 'eddy' error" occurs.

MarkusNiewoehner, 2025-02-28:
> I also have this issue with my Sovol SV08 Mainline - when powering on, after a few minutes the "Lost communication with MCU 'eddy' error" occurs.

same here...

user-42, 2025-03-01:
> Hey guys!
> 
> Here is a possible fix for EMI noise that causes errors and disconnects: use ferrite cores for cables if you can't get shielded cable.
> 
> [image]
> 
> I also had errros on long prints and ferrite cores seem to fix. I placed two cores on the eddy's cable and made several 3-4h prints. No issues as for now.
> 
> Would be good if somebody else tests ferrite cores.

I have no issues for last months. Ferrite cores solve problem.

cyaneo, 2025-03-02:
Really?
To save a few cents, BTT apparently includes an inadequate shielded USB cable? Shame on you!

Just like with the SV06 clipper screen, the cable had to be far away from other electronic devices :-(

looxonline, 2025-03-02:
> Really? To save a few cents, BTT apparently includes an inadequate shielded USB cable? Shame on you!
> 
> Just like with the SV06 clipper screen, the cable had to be far away from other electronic devices :-(

The cable is actually very costly. It comes with an internal shield that has to be custom manufactured since shields are not usually designed to fix to a picoblade terminating connector. We included the ferrites on the board itself at either end. There are RF chokes designed to reflect HF RF interference integrated directly into our SBCs and the Eddy itself. If someone is using another SBC without these integrated chokes then it may allow some through. Also, if a user has an extremely noisy environment then additional chokes may be required.

cyaneo, 2025-03-02:
OK - are there ferrits on both ends? Where are they? And why do additional ferrite cores seem to fix the problem (see previous post)? I tested it: When routing the USB cable outside the printers (and using the external USB connector on the SV08), I tested - 1 hour of running time - no error...

[image]

Witchd0ct0r, 2025-03-23:
Having the same problem. Currently using Neptune 4 Max printers running OpenNeptune. All sensors are producing the same random disconnects. Put ferrite cores to both ends of the cables. It reduces the problem somewhat. We are currently having over 1000 EUR worth of failed prints and materials. This is the worst product ever. If anyone is reading this issue please DO MOT BY BTT EDDY. I repeate DO NOT BUY. This is currently an unsolved problem for a year. We are switching to another sensor on our printers. Imagine the printer stopping in the middle of a 4 day printjob with 2 kg filament gone from a 3 kg spool. The error also shots down the heated bed. There is no way to restart the print. Also due to the searila USB connection there is no way to make klipper to ignore the disconnect.

The sensor end of the cable is clearly unshielded. Also on the replacement cable sent by btt the same spot is also unshielded. Probably the EMI noise from the extruder is throwing off the serial communication. It is absolutely clear that the cable was not tested for noise resistance and the manufacturer takes 0 responsibility.

jiridoko, 2025-03-28:
Same here. I have made a custom toolboard with USB hub board right next to my EBB36, only USB data and DC power (so 4 wires total) going to the toolhead), and then driving both the EBB36 and Eddy Duo (USB mode) from it. Never had any problems with EBB36 on USB on that machine but the Eddy sensor disconnects after about an hour of print. And the wire going from the Eddy probe to the USB hub is about 5cm, not more than that. I am suspecting this may be related to the probe temperature though - it's sitting at about 70C reported temperature when it disconnects. I will try switching over to CAN bus to see if this resolves this issue. Love the idea of the product but the details like that are quite annoying. But if everything just worked it would be boring I guess 😃

yael-lorenzo, 2025-04-03:
I have the same problem as well. I went back to the old probe that works the same in my experience.  I didn't gain much and lost a lot. Eddy communication failures stops printing. 
I am seriously thinking of investing more energy to put the ferrite rings.

jiridoko, 2025-04-03:
I think I solved it. I swapped the USB wire going from my rpi4 to the toolhead for a thicker one with heavy shielding (shield connected to ground on either end), I put a ferrite ring on it and I set it at a distance from any 24V PWM wires (bed heating, hotend heating, stepper motors). And now I have finished a 8h print without any issues. Seems to have fixed the issue!

[image]
[image]
[image]

benster310, 2025-04-28:
I tested my Eddy USB by removing it from the print head. When I switch on my Voron 2.4 Rev D, the Eddy does not power up immediately. I started to twist the wire and suddenly the Eddy probe would get power. I checked the Mainsail dashboard, and while the Eddy has power, all is ok. As soon as I move the able around, the Eddy loses its power, and then Klipper complains it has lost ommunication with MCU_Eddy.

I will be cuting a piece of the cable out and replace it and do the same test again. Hopefully this works. I think that the cable provided with the Eddy USB, might be sub standard as mine is running through a dragg chain, which may be the cause of it no holding up under all the movement.

bigtreetech, 2025-04-28:
> I tested my Eddy USB by removing it from the print head. When I switch on my Voron 2.4 Rev D, the Eddy does not power up immediately. I started to twist the wire and suddenly the Eddy probe would [...]

Sounds like you have a loose connection with one of the terminals. Rigid strain relief is an absolute must with any moving installation like this otherwise connections will become loose with time.

benster310, 2025-04-28:
I do not fully agree with your statement. The Voron kit has a Stealth burner hot end. The BTT Eddy is in the place of the Omron probe which means that the cable is securely mounted in a channel on the Stelth Burner, so there is no movement where the cable is plugged into the BTT. The rest of the cable runs in the drag chain, but I suspect that one of the thin wires, has broken and with movement disconnects and then connects again when I twist the cable. Unfortunately I cannot a buy a replacement cable for it, as no one sells them. I will take a bet that replacing the cable will solve the problem and I am going to do my own test to prove that it was in fact the cable that is faulty, and not a loose joint. This is a brand new build, and I have only printed some calibrations samples, and now have to fix this mess before I can continue. I have not even started to print proper 3D prints.

bigtreetech, 2025-04-28:
> I do not fully agree with your statement. The Voron kit has a Stealth burner hot end. The BTT Eddy is in the place of the Omron probe which means that the cable is securely mounted in a channel on [...]

Check the terminals. There is very little chance of a cable breaking internally with such little movement. The only possibility of that would be if the cable is pulled up or pushed hard against the cable chains so that it is stretched or compressed when they move. It should always sit at the mid point of the chain. I have had several running on machines for over a year now without a break using the same cable.

benster310, 2025-04-28:
How do you propose I open the BTT Eddy to check the Terminals? Or are you referring to the terminal where the wires go in to see if there is something loose?

bigtreetech, 2025-04-28:
> How do you propose I open the BTT Eddy to check the Terminals? Or are you referring to the terminal where the wires go in to see if there is something loose?

No, no. Not inside the Eddy. Just the terminals at the end of the wire that plug into the Eddy. Possible that the wire to terminal or terminal to pin joint is a little loose. You could test it by getting the Eddy loose and then just fiddling with that plug to see if it causes disconnects.

[20 more replies]