bigtreetech/Eddy #31: Klipper reports: ERROR  Option 'data_rate' is not valid in section 'probe_eddy_current btt_eddy'

diegostf, 2024-06-13:
I'm installing a clean version on the Hurakan (MANTA M4P), as the previous version was losing communication with the Eddy mcu, after an hour of printing (https://github.com/bigtreetech/Eddy/issues/30).
  During installation, klipper is showing several errors in [probe_eddy_current btt_eddy] and [temperature_probe btt_eddy].

Section 'temperature_probe btt_eddy' is not a valid config section
Option 'data_rate' is not valid in section 'probe_eddy_current btt_eddy'

I also had to comment on the  [probe] section (a lot of erros). 

[klippy.log](https://github.com/user-attachments/files/15827790/klippy.log)

krautech, 2024-06-13:
You need to be using BTTs klipper

diegostf, 2024-06-14:
> You need to be using BTTs klipper

Yes, I'm using it. I followed the installation tutorial (https://github.com/bigtreetech/Eddy) until step 15.

krautech, 2024-06-14:
According to your update manager you're using the master branch of klipper and not BTTs fork.

BTTJUSTIN, 2024-06-14:
Thanks for reaching out. Our support team does not operate using the GitHub
platform but we would be happy to help you via email based ticketing system. Please visit https://biqu.equipment/pages/technical-support and submit a ticket and we will be in touch with you soon

closed 2024-06-17 by hwisgold

Rotifox, 2025-01-03:
> According to your update manager you're using the master branch of klipper and not BTTs fork.

Please can you explain how to do what you say? I have the same problem