bigtreetech/Eddy #32: Z Offset Calibration - Movement out of range.

PigeonFX, 2024-06-14:
Hi, 

Following along with the guide and get to the selection 3. Z Offset Calibration.

> Home X and Y axes with command G28 X Y
> Make sure you dont have a bed heightmap loaded.
> Move Nozzle to Centre of the bed with G0 X125 Y125 F6000 (adjust for your bed size)
> Start Manual Z-Offset Calibration by typing PROBE_EDDY_CURRENT_CALIBRATE CHIP=btt_eddy  (TIP: Go a little lower than you normally would)

All works fine until I put in PROBE_EDDY_CURRENT_CALIBRATE CHIP to the terminal, and the Z offset TESTZ Z=- window pops up:

[image]

regardless of what I press, I get the Movement out of range error and it closes out of TESTZ Z=-

[image]

Here is my printer config, with the Eddy setting at the top, could someone point out what I am doing wrong, as I'm 98% sure it is user error! 

[Printer + eddy.txt](https://github.com/user-attachments/files/15832886/Printer.%2B.eddy.txt)

krautech, 2024-06-14:
Try add the following to your printer.cfg

```
[force_move]
# Enable commands that force potentially unsafe movement
enable_force_move: True
```

BTTJUSTIN, 2024-06-14:
Thanks for reaching out. Our support team does not operate using the GitHub platform but we would be happy to help you via email based ticketing system. Please visit https://biqu.equipment/pages/technical-support and submit a ticket and we will be in touch with you soon

closed 2024-06-18 by PigeonFX