bigtreetech/Eddy #37: z tilt adjust readings way off.

crackmonkeyy, 2024-06-29:
I have a strange issue with eddy, it homes and generates bed meshs fine but when I do z tilt adjust it gives crazy readings. with the bed a few mm away it will say things like 90+plus mm then next spot 60 something and ever time it's completely random. 
[image]

eselcuker, 2024-07-03:
i have same issue, impossible to make z tilt, because of this cant print anything, any solution ?

crackmonkeyy, 2024-07-03:
> i have same issue, impossible to make z tilt, because of this cant print anything, any solution ?

i managed to get the z tilit working by setting the z offset to 0, there seams to be a bug which kept adding my new z_offset to the old one and eventually my z_offset was at 90 odd mm and that messed with the z tilit. the only was i can get the z_offset working properly was offsetting the gcode using prusaslicers z_offset feature.

eselcuker, 2024-07-03:
Thanks for your repply but when i make z offset 0 under probe settings in cfg i am getting Invalid probe_eddy_current height error cant do z tilt

looxonline, 2024-07-03:
@eselcuker @crackmonkeyy @
Some questions and some things to consider:

Question(s):
- What printers are you all using?

Consider:
- The Eddy will read reliable data up to about 15mm before the edge of the bed. Make sure that you are not setting your probing points within that safety boundary for the z-tilt adjust.

eselcuker, 2024-07-04:
> * adjust

mine is a voron trident, BTT EDDY is a fully bullshit, i took it apart, threw it away and i don't plan on using it again, useless product. i did dispute order on aliexpress,

looxonline, 2024-07-04:
> > * adjust
> 
> mine is a voron trident, BTT EDDY is a fully bulls***, i took it apart, threw it away and i don't plan on using it again, useless product. i did dispute order on aliexpress,

How close to the edge of the bed were you trying to probe?

crackmonkeyy, 2024-07-06:
hey, I have mecury zero g and the issue for me was the z offset being ignored by eddy, each time I would lower the nozzle -0.5mm and  it save to the config. but when the next print starts its around 0.5mm to high from the bed again. i lowered nozzle again and saved then eventually my z offset was near 90mm. which then caused the z tilt errror. to reproduce the issue set z offset to 90mm at the bottom of the printer.cfg . as a work around ive set the offset to 2mm for eddy in printer.cfg and used prusaslicer to modify the gcode to adjust the z offset. im still not getting a reliable reading for eddy as a z endstop though, it's not reliable at all. 😔 hopefully u guys can get it sorted with an update?

bigtreetech, 2024-07-06:
> hey, I have mecury zero g and the issue for me was the z offset being ignored by eddy, each time I would lower the nozzle -0.5mm and it save to the config. but when the next print starts its around [...]

I know exactly what was happening with your machine and why. I am 100% confident that you can get great results out of it by following the steps below:

1. Bring in all of the latest macros from our latest sample.cfg file.
2. Perform the eddy probe calibration and thermal calibration again.
3. Have a read through this to understand how a z-offset works with an Eddy current probe and why your changing and saving did what it did: https://gist.github.com/bigtreetech/484380b26be613b9139bc537510393df

Please let me know the results after you apply point 1 and 2. I am sure they will solve your issues.

crackmonkeyy, 2024-07-10:
hey @bigtreetech , Thanks for your help, I tried your suggestions today.
1. i loaded all the new macros from the beta config. only issue i had was due to me having a ecrf v2 and the multiple call for save_variables but that was a simple fix to merge the folders.
2. i redid the probe calibration and thermal calibration, i had slight issues with the thermal compensation calibration as my printer is open air and eddy only gets to 46 degrees c. it only did 3 calibrations even when i changed the steps to every 2 degrees? the thermal compensation seams to be working fine though so not a big issue.
3. eddy was saving the z offset in printer.cfg, the new beta is saving it in the save_variables folder correctly now and loads it when starting a print correctly.

looxonline, 2024-07-13:
Closing since this seems to be resolved now.

closed 2024-07-13 by looxonline