bigtreetech/Eddy #38: Unable to home - !! Error during homing z: Eddy current sensor error

geekyhawkes, 2024-07-06:
I have tried both the BTT fork and main klipper but have issues with both using eddy:
On the BTT fork I run into error: Internal error during connect: split() takes no key word arguments.

On main klipper things go better but when i home i get the error:
!! Error during homing z: Eddy current sensor error

I believe this is because eddy is in a state of error during homing but am unable to verify or work out how to get around this issue.

Kooramah, 2024-07-06:
did you check all of your configs and did them according to the readme?

geekyhawkes, 2024-07-06:
I did, followed everything to the letter but I think the issue is because
using core klipper it has problems with an error statement of eddy when
homing but the BTT fork gave me other issues.

spddqd, 2024-07-07:
python --version?

geekyhawkes, 2024-07-07:
Was 2 but also installed 3 to see if it helped. No change sadly.

spddqd, 2024-07-07:
Did you completely remove python2 from your system? ``python`` might still link to old python2. Though that's a shot in the dark.
I was able to compile everything using KIAUH with custom branch yesterday. Might give that one a try.

geekyhawkes, 2024-07-07:
Thank you, I didn't remove python 2 just installed 3 as the guidance from
python was not to remove.

I did try KIAUH for the switch but maybe worth another try and see where I
get to.

geekyhawkes, 2024-07-07:
Interestingly updating to the latest klipper main fork and everything is
now working without issue. Thanks to the klipper team who have pulled and
implemented eddy.

bigtreetech, 2024-07-08:
> Interestingly updating to the latest klipper main fork and everything is now working without issue. Thanks to the klipper team who have pulled and implemented eddy.
> […](#)

The thermal compensation algorithm is almost implemented in main too. Should just be a matter of days now.

bigtreetech, 2024-07-10:
> Interestingly updating to the latest klipper main fork and everything is now working without issue. Thanks to the klipper team who have pulled and implemented eddy.
> […](#)

Going to close this since it is sorted.

closed 2024-07-10 by bigtreetech