bigtreetech/Eddy #40: BTT Eddy Probe error - Internal error during connect: split() takes no keyword arguments suggest guide update

geekyhawkes, 2024-07-16:
Hello,

I followed the guide provided here but ended up with a klipper error Internal error during connect: split() takes no keyword arguments. installing python-is-python3 didnt resolve the error but upgrading my piosLite to 12 (bookworm) from 11 (bullseye) and then following your guide fixed the issue.  

Can I suggest the start of the guide at least states the min required debian version (seems to be 12 based on my experience) but also maybe a link out to a guide on how to upgrade the distro?

Everything is working great now after a bit of back and forth so I might not be the only one out there that missed the issues with bullseye / deb11.

bigtreetech, 2024-07-16:
This is not so much an issue with your distro version in as much as it is an issue with your python version. You were running python 2.x with your first attempt but somehow your python was updated to 3.x with whatever you did along the way. The split argument only supports keywords from 3.x onwards and this is what klipper tries to use with it. 

I'll update the guide to indicate that users need to be 100% sure that they are using python 3.x with their klipper installation. Guiding them through that process is a tricky one and outside of the scope of the Eddy guide but KIAUH would be the easiest for most people. Just back up existing configs, remove any existing instances of klipper, install again using KIAUH and tell it to use python 3.

bigtreetech, 2024-07-16:
Thanks for taking the time to give us a heads up BTW. Updating the guide now and closing this off.

closed 2024-07-16 by bigtreetech