bigtreetech/Eddy #41: Triggers above bed

sanpedrox2, 2024-07-17:
Hi.
when trying to get my eddy working i have hit a wall.
after the temperature tuning step the eddy triggers a few cm above the bed when homing and ofc cant get a scan off the bed from that height. if i try printing obviously that doesnt work either from a few cm above the bed :)

as the bed cools down it triggers closer and closer, i get a feeling that some numbers are calculated wrong somewhere.

any tips would be greatly appreciated :)

looxonline, 2024-07-17:
I feel like your thermal tuning must be off. Are you sure that you followed the guide 100%? I would suggest:
1. Back up your klipper config.
2. Delete the autosaved thermal compensation at the end of the klipper config so that it has no effect.
3. Check if you have the same issue of it triggering up cm above the bed when it is hot.
4. If the problem has gone away then there was something that was not done right during the thermal comp step. Remember that you need to run a paper level at each point during the thermal comp.

sanpedrox2, 2024-07-17:
i did delete the settings and started over, with the same result. didnt try it inbetween though.
and yes, i know to do the papertest, i have followed the guide 100% both of the times. especially the second ;)

gonna have a go with a clean slate and see

sanpedrox2, 2024-07-17:
ok.. it does increase. but not so extreme. just a couple of mm instead of cm. guess ill retry the temp tune :/

looxonline, 2024-07-17:
> ok.. it does increase. but not so extreme. just a couple of mm instead of cm. guess ill retry the temp tune :/

I have noticed that bad data in the autosave part of the config can really mess with things. Maybe it is best to just clear out the whole autosave part that has anything to do with probes and start fresh.

sanpedrox2, 2024-07-17:
yes, i did that several times now. still get the same result. 
trying to home after a save_config, results in "endstop z still triggered after retract". the nozzle is 2 cm above bed at this point because of z lift before home.

eddy at 60c triggers a few cm above bed. at 55 its about one cm.

pretty sure it doesnt use temperature compensation at all

bigtreetech, 2024-07-17:
> yes, i did that several times now. still get the same result. trying to home after a save_config, results in "endstop z still triggered after retract". the nozzle is 2 cm above bed at this point because of z lift before home.
> 
> eddy at 60c triggers a few cm above bed. at 55 its about one cm.
> 
> pretty sure it doesnt use temperature compensation at all... or even compensating in the wrong direction?

Could you upload your config files to pastebin and post them. Please also upload some pictures of your installation here, clearly showing the height of the eddy above the nozzle.

sanpedrox2, 2024-07-17:
https://pastebin.com/xZbrmwja
https://pastebin.com/FrzJGb6b

sanpedrox2, 2024-07-17:
[image]
[image]

looxonline, 2024-07-17:
Thanks. It is super late here now. I'll have a look in the morning.

sanpedrox2, 2024-07-18:
i think i may have fixed it by changing #*# reg_drive_current = 15
to #*# reg_drive_current = 16
as i saw some others had succes with that.

so far it seems to be working and not throwing any errors when homing/meshing

closed 2024-07-18 by sanpedrox2

looxonline, 2024-07-18:
> i think i may have fixed it by changing #_# reg_drive_current = 15 to #_# reg_drive_current = 16 as i saw some others had succes with that.
> 
> so far it seems to be working as expected and not throwing any errors when homing/meshing
> 
> _edit_ after further testing i can see that it works really well now. case closed :) phew... i was so close to just giving up and reinstall my bltouch

Thanks for letting us know about the resolution. We have not seen this sort of behaviour on other installations so I will add something to the FAQ to share your experience with others.

sanpedrox2, 2024-07-27:
No problem :-)
i can also report that i did try to revert  ## reg_drive_current = 16 back to 15 out of curiosity. to check if it really was the solution as i had tinkered with a few other settings.

Can confirm that the problem came back instantly with 15, and dissapeared as soon as i went back to 16.
i did the current calibration with eddy 20mm from the bed as per the instructions and got 15 every time. its not right in my case atleast 👍