bigtreetech/Eddy #42: PROBE_EDDY_CURRENT_CALIBRATE Error

DrahuoynillorT, 2024-07-18:
6:56 PM
Move out of range: 125.000 125.000 270.300 [0.000]
6:56 PM
TESTZ Z=-
6:56 PM
Z position: ?????? --> 270.000 <-- ??????
6:56 PM
Starting manual Z probe. Use TESTZ to adjust position.
Finish with ACCEPT or ABORT command.
6:56 PM
PROBE_EDDY_CURRENT_CALIBRATE CHIP=btt_eddy

This is the error I receive. I am not sure what I am doing wrong window pops up in mainsail to adjust every time I hit the negative or even the advanced section of numbers below the window exits and I receive this error even when I increase the Z max it does the same thing : Move out of range: 125.000 125.000 300.300 [0.000]. It says to adjust with TESTZ but I dont know the command for this other than just using the PROBE_EDDY_CURRENT_CALIBRATE CHIP=btt_eddy which doesnt allow me to adjust anything and keeps kicking me out of the window.

looxonline, 2024-07-18:
What is the max Z height on your printer?

DrahuoynillorT, 2024-07-18:
> What is the max Z height on your printer?

270 is my max height. I have fixed the issue though. I was connecting remotely from my pc as it is easier for me to read instructions and copy paste the commands etc. My printer is in a garage, and I was running back and forth when adjusting for the 20mm height. The problem was my printer has a touch screen attached with Klipper screen worked fine from screen to adjust height but hated doing it from the remote connection for some reason.

looxonline, 2024-07-18:
> 270 is my max height. I have fixed the issue though. I was connecting remotely from my pc as it is easier for me to read instructions and copy paste the commands etc. My printer is in a garage, and [...]

It should still work from the remote terminal though. According to the error it was trying to go 0.3mm above your max height which seems to suggest that + was being hit instead of -. Are you using the latest guide in the readme and the latest macros?

DrahuoynillorT, 2024-07-18:
> It should still work from the remote terminal though. According to the error it was trying to go 0.3mm above your max height which seems to suggest that + was being hit instead of -. Are you using [...]

https://bttwiki.com/en/Eddy.html   Im using the guide from the btt wiki I assume it is the latest.

looxonline, 2024-07-18:
> > > > What is the max Z height on your printer?
> > > 
> > > 
> > > 270 is my max height. I have fixed the issue though. I was connecting remotely from my pc as it is easier for me to read instructions and copy paste the commands etc. My printer is in a garage, and I was running back and forth when adjusting for the 20mm height. The problem was my printer has a touch screen attached with Klipper screen worked fine from screen to adjust height but hated doing it from the remote connection for some reason.
> > 
> > 
> > It should still work from the remote terminal though. According to the error it was trying to go 0.3mm above your max height which seems to suggest that + was being hit instead of -. Are you using the latest guide in the readme and the latest macros?
> 
> https://bttwiki.com/en/Eddy.html Im using the guide from the btt wiki I assume it is the latest. Also yes, I understand the point is to get the eddy as close as possible.

No, we still need to update that. Use the guide from the readme in this repo.

closed 2024-07-18 by looxonline