bigtreetech/Eddy #43: Eddy Touch

0xfde9, 2024-07-20:
Hello team.

Any update regarding "Eddy Touch" I'm still getting from time to time the inaccurate 'z offset' 

Regards.

looxonline, 2024-09-11:
This is in the pipeline

igorscheffer, 2024-12-14:
any news about eddy touch

HeavilyModdedEnder3, 2025-01-07:
Happy New Year! Just wondering if there has been any update since September

HeavilyModdedEnder3, 2025-01-16:
[image]

looxonline, 2025-01-16:
It may be sooner than you think. A bright member of the community has helped with the work and it is going into testing.

jbenzaquen42, 2025-01-22:
https://hackmd.io/yEF4CEntSHiFTj230CdD0Q

Looks like we can test it now!

looxonline, 2025-01-22:
Yup. It is currently in beta testing by some of our beta testers and things are looking good. Please give all credit to the author of this. Vlad is a gifted person!

jbenzaquen42, 2025-01-22:
> Yup. It is currently in beta testing by some of our beta testers and things are looking good. Please give all credit to the author of this. Vlad is a gifted person!

All credit to Vlad for sure! I was expecting some extra wiring, but this looks much better than I expected!

blinkitogaming, 2025-01-23:
+1 on this feature and really thanks to Vlad. Can't wait to see this working.