bigtreetech/Eddy #44: Failed calibration - frequency not increasing each step

JonasBouchard, 2024-07-22:
When I run `PROBE_EDDY_CURRENT_CALIBRATE_AUTO CHIP=btt_eddy` , I get the error `Failed calibration - frequency not increasing each step` . I already ran through drive current calibration and the z offset between the nozzle and the probe is very small.

jeffstark, 2024-07-24:
The command isn't working for me either. After I accept the paper test in the very first step, the console says `temperature_probe btt_eddy: calibration aborted`

If I proceed by changing the bed and nozzle temp, the script never asks me to do the paper test again.

BTTJUSTIN, 2024-07-26:
Thanks for reaching out. Our support team does not operate using the GitHub platform but we would be happy to help you via email based ticketing system. Please visit https://biqu.equipment/pages/technical-support and submit a ticket and we will be in touch with you soon

bigtreetech, 2024-07-26:
@BTTJUSTIN This is not a support request. This is a problem and it should follow that flow. I am reopening it and I want you to please follow the correct flow.

bigtreetech, 2024-08-05:
> The command isn't working for me either. After I accept the paper test in the very first step, the console says `temperature_probe btt_eddy: calibration aborted`
> 
> If I proceed by changing the bed and nozzle temp, the script never asks me to do the paper test again.
> 
> And trying to manually execute the next step with `PROBE_DRIFT_NEXT` tells me `Unknown command:"PROBE_DRIFT_NEXT"`
> 
> EDIT: I got a little further by hitting accept on klipperscreen (didnt even notice it pop up on the screen before) instead of mainsail, but as I continue, klipper crashes telling me `Internal error on command:"PROBE_CALIBRATE"`

This is not the same command that the OP is talking about. You are speaking about thermal comp and the OP is speaking about the frequency mapping. Have you performed your frequency mapping before you try to perform your thermal comp?

bigtreetech, 2024-08-05:
> When I run `PROBE_EDDY_CURRENT_CALIBRATE_AUTO CHIP=btt_eddy` , I get the error `Failed calibration - frequency not increasing each step` . I already ran through drive current calibration and the z offset between the nozzle and the probe is very small.

Some questions:

1. Are you using Eddy or Eddy coil?
2. What was the result of the LDC calibration? Should be between 14 and 16.

bigtreetech, 2024-08-13:
> When I run `PROBE_EDDY_CURRENT_CALIBRATE_AUTO CHIP=btt_eddy` , I get the error `Failed calibration - frequency not increasing each step` . I already ran through drive current calibration and the z offset between the nozzle and the probe is very small.

Closing because we are not getting any response.

closed 2024-08-13 by bigtreetech

SinglerGold2, 2024-10-05:
Well, I do have the problem now...

BTT Eddy USB
LDC Calibration =  15

MJD19994, 2024-10-17:
I am having the same issue with an Eddy USB, it fails the calibration every time and my LDC Calibration came in at 16 within the range of the above comment. I submitted a support ticket but it so far that has been very little help. It seems like my printer is losing steps on the z axis and not actually moving as it should during the calibration. The lost steps only happen during the calibration action, during normal operations my z axis moves as it should.

Sharkooon42, 2024-12-10:
I am experiencing the exact same issue as described in the last comment. My Eddy USB sensor fails calibration every time. The Z-axis seems to lose steps during the calibration process and doesn’t move correctly, even though it operates perfectly fine during normal use. My LDC calibration also resulted in a value within the correct range, so I would rule out a hardware issue.

It would be great if someone could share a solution or suggest any approaches to resolve this problem. I’ve already tried adjusting some settings, but without success. Are there any further tips or updates that address this issue?

AleCo3lho, 2024-12-12:
I am having the same problem, Eddy Coil. LDC 16

jotaperez3, 2025-01-19:
Was anyone able to solve this problem, I have not been able to solve it in any way.

AleCo3lho, 2025-01-19:
> Was anyone able to solve this problem, I have not been able to solve it in any way.

In my case I am using a Eddy Coil so it may be different from yours, but basically it was a problem with the wiring.