bigtreetech/Eddy #45: Invalid probe_eddy_current height

Zaskeller, 2024-07-25:
probe is 1.2mm above nozzle. done probe calibrate on the EDDY USB version. python is updated klipper fork is used and have tested without klipper fork still get error.

"Invalid probe_eddy_current height"

notable this error triggers instantly if say doing a screw tilt calculate or bed mesh. if machine is homed. doing a bed mesh moves the probe to the first point but the error triggers before the probe has moved. 

have this issue on two machines i have tested every thing i can think of and just can not get it working.

bigtreetech, 2024-07-25:
Please try to raise the probe up to about 4mm above the nozzle and then try again. Please let me know how it goes. I believe that in some cases users actually need to mount their probes higher than usual and if so then I would like to add this to the manual/guide.

flavio-costa, 2024-08-09:
My Btt Eddy USB started to present the same problem, even placing it 4mm from the table, redoing the entire configuration from scratch, it doesn't work.

flavio-costa, 2024-08-09:
Apparently, I managed to solve my problem.
I went to printer.cfg and uncommented the z offset line in: [probe_eddy_current btt_eddy]
and I redid the calibration, it worked normally again.

bigtreetech, 2024-08-13:
> Apparently, I managed to solve my problem. I went to printer.cfg and uncommented the z offset line in: [probe_eddy_current btt_eddy] and I redid the calibration, it worked normally again.

Yes, this setting is required.

closed 2024-08-13 by bigtreetech

bigtreetech, 2024-09-26:
> > Apparently, I managed to solve my problem. I went to printer.cfg and uncommented the z offset line in: [probe_eddy_current btt_eddy] and I redid the calibration, it worked normally again.
> 
> Thank you, this saved me a full reinstall :) I just replaced the nozzle and heat break on my printer, so as I usually do with any probes I went into the printer.cfg and in the "do not edit" section I reset the probe offset to 0.000 so I could start from scratch. That triggered the same issue you had, so I removed the # before the line you mentioned and bingo :) I was about to copy off my configs and reinstall :)
> 
> BTW - Yes it has just occurred to me I didn't need to edit that offset in the "do not edit" section :) I've done it for so long I can't remember why I actually started, probably a left over from Marlin or something :)

I think I'll try to add this point to the FAQ to let people know that it's better to keep it in the standard config than in the do not touch section.

serbiansky, 2024-10-23:
What should be the value for z_offset then as I am facing issues with my nozile is to low what ever the settings in z_offset I put?

DaveTheBuskerUK, 2024-10-23:
> What should be the value for z_offset then as I am facing issues with my nozile is to low what ever the settings in z_offset I put?

You tune that live, you don't enter a value into the printer.cfg

The tool is called PROBE_CALIBRATE you can launch it by typing that command into the console

Here's the Klipper docs page for it - https://www.klipper3d.org/Probe_Calibrate.html

Does that help ?

serbiansky, 2024-10-24:
I am trying to use Eddy on my Ender 3v2 modified printer as probe and z end stop and in the configuration example they added z-offset=2.5, if I comment this part then klipper is complaining. Whatever the value I put in the probe is to low. Yesterday I put 2.5 and nozzle banded the bed. If I reduce to range 0.1-1.5 is to low and 0.0 klipper is complaining. Obviously I am doing something wrong.
https://github.com/bigtreetech/Eddy/blob/master/sample-bigtreetech-eddy-homing.cfg

motion-mind, 2025-02-06:
I'm still getting this error. When I run PROBE_EDDY_CURRENT_CALIBRATE_AUTO CHIP=btt_eddy, the height that it starts with is 346mm no matter what. Shouldn't this start closer to 0? After I complete this calibration and try to HOME, the toolhead position says it's at 500mm+ and I get the **Invalid probe_eddy_current height** and HOME fails.