bigtreetech/Eddy #46: PROBE_DRIFT_CALIBRATE Problem

ej8g82p, 2024-07-26:
I'm having an issue performing the temperature calibration. After running PROBE_DRIFT_CALIBRATE PROBE=btt_eddy TARGET=50 STEP=5 and making the last adjustment, I get a "probe_eddy_current btt_eddy: invalid calibration detected, curve at index 5 overlaps previous curve at temp 75C" message.
Anybody else run into this problem before, or have any ideas?

jeffstark, 2024-07-26:
At least a couple of us are experiencing this issue with no solution in sight. BTT keeps closing support tickets and telling us to email them.

ej8g82p, 2024-07-26:
I see. Thanks for the heads up

bigtreetech, 2024-07-26:
> At least a couple of us are experiencing this issue with no solution in sight. BTT keeps closing support tickets and telling us to email them.

There is *one* issue that has been closed like that with the others having received thorough discussion and assitance.

ej8g82p, 2024-07-26:
What's the proper way to go about getting assistance?

Brenddpaula, 2024-07-27:
Hey, I was having the same problem and I've solved it by erasing the saved configs related to eddy from my printer.cfg and started the process again. 
What I've noticed is that this issue was happening when the value of testz during temperature raise were lower than the previous ones. After deleting the config and starting all again it did not find any reductions in value while raising the bed temperature and was able to complete the process.

ej8g82p, 2024-07-27:
> Hey, I was having the same problem and I've solved it by erasing the saved configs related to eddy from my printer.cfg and started the process again. What I've noticed is that this issue was [...]

Thank you for you help. I've tried doing that multiple times, but it still fails for me. What I'm confused about is that it detects a problem with a previous curve at 75C, and I can't seem to get my eddy to go past 54C when doing the calibration 🤷‍♂️

ph0tic, 2024-08-10:
im having this problem too

jeffstark, 2024-08-10:
> im having this problem too
> 
> 

Same here

bigtreetech, 2024-08-12:
Hey all. We are looking into this and will update this thread as soon as we have a better understanding of what may be causing this.

bigtreetech, 2024-08-12:
@jeffstark @ej8g82p @ph0tic @Brenddpaula Could you all let me know what temp range you are trying to calibrate to (the command you are initially sending) and what temperature you are actually able to hit during the calibration?

Arksine, 2024-08-12:
Hi everyone, I have pushed a branch that should help alleviate this issue.

https://github.com/Arksine/klipper/tree/dev-temperature_probe-20240812

The problem is due to the relatively low range you are able to achieve during calibration.  The calibration is capable of extrapolating values outside of the calibration range, however if your printer is not enclosed it is unlikely that the probe's coil will ever reach temperatures that high.  This patch adds a new `max_valid_temp` option that defaults to 60C.  This value should allow you to pass validation.

If some users could test this branch and confirm it resolves their issue I will create a pull request for it.

jeffstark, 2024-08-12:
> @jeffstark @ej8g82p @ph0tic @Brenddpaula Could you all let me know what temp range you are trying to calibrate to (the command you are initially sending) and what temperature you are actually able to hit during the calibration?

I'm just following the guide step by step, so whatever is in the official guide is what I did. I see that we can now use mainline klipper, so im gonna wipe the eddy stuff again and try it from scratch. If that doesnt work, I'll try Arksine's build.

ej8g82p, 2024-08-12:
> @jeffstark @ej8g82p @ph0tic @Brenddpaula Could you all let me know what temp range you are trying to calibrate to (the command you are initially sending) and what temperature you are actually able to hit during the calibration?

I think the max temp I was able to get to is around 54C, and I was calibrating to 50C with the last step being around 51-52C

bigtreetech, 2024-08-12:
> I think the max temp I was able to get to is around 54C, and I was calibrating to 50C with the last step being around 51-52C

Yea then your cali range would have been so low that the validation algo would have hit some errors when trying to validate the extrapolated results around 75C and upwards. The new branch should fix it. Once you and others validate, Arksine will issue the PR.

ej8g82p, 2024-08-12:
Sounds like @Arksine is dead on with the solution. I most definitely can't reach the higher temperatures with my max being around 54C. I will try the 60C max branch and will report back as soon as I get a chance

jeffstark, 2024-08-12:
I'm trying @Arksine's klipper build, and Im up to step 32 in the guide, which is what I couldn't proceed beyond before.

Step 32 is enter `PROBE_DRIFT_CALIBRATE PROBE=btt_eddy TARGET=56 STEP=4` in the console. When I do that it now says `Unknown command:"PROBE_DRIFT_CALIBRATE"`

It's entirety possible I messed up somewhere, but that's where I am. With the btt klipper fork, I could enter this command, but never finish the calibration steps. This build seems to be missing it, unless I took a wrong turn somewhere.

bigtreetech, 2024-08-12:
> I'm trying @Arksine's klipper build, and Im up to step 32 in the guide, which is what I couldn't proceed beyond before.
>
> Step 32 is enter `PROBE_DRIFT_CALIBRATE PROBE=btt_eddy TARGET=56 STEP=4` in [...]

Ah wait. I forgot to update that command. One sec...read the guide in two minutes.

jeffstark, 2024-08-12:
> Ah wait. I forgot to update that command. One sec...read the guide in two minutes.

I can confirm that @Arksine's klipper build along with the new command `TEMPERATURE_PROBE_CALIBRATE PROBE=btt_eddy TARGET=56 STEP=4` works.

I have finally completed this step.

Thanks!

bigtreetech, 2024-08-12:
Going to close this one out for now and ask Arksine to send through a PR.

closed 2024-08-12 by bigtreetech

Arksine, 2024-11-25:
> The link for the branch does work anymore, can you update it? Thanks in advanced

The changes in that branch were merged to Klipper master, so the branch was deleted.

1jgrodrigues, 2024-11-25:
> > The link for the branch does work anymore, can you update it? Thanks in advanced
> 
> The changes in that branch were merged to Klipper master, so the branch was deleted.

Do you mean BTT version of klipper or the oficial klipper version? 
Were do you configure the option max_valid_temp?

Arksine, 2024-11-25:
> Do you mean BTT version of klipper or the oficial klipper version?
Were do you configure the option max_valid_temp?

The official Klipper branch.  The `max_validation_temp` option is configured in the [temperature_probe](https://www.klipper3d.org/Config_Reference.html#temperature_probe) section.