bigtreetech/Eddy #47: homing issues after PROBE_DRIFT_CALIBRATE

cjhmdm, 2024-07-28:
Using the USB version of Eddy. Going through the config steps in this wiki. After running PROBE_DRIFT_CALIBRATE, when I home my z gets set to 99.900 and I am unable to manually (or otherwise) move it to 0

cjhmdm, 2024-07-28:
[image]
screenshot of the issu. Notice Z is set to 99.900, this is immediately after homing and after going through PROBE_DRIFT_CALIBRATE

CzarnyRafal, 2024-08-08:
It looks like there is problem with saving variables.
Are you using this Zoffset-beta config file?

I have similar problem, when you copy this "Result..." value (without minus) to the variables.cfg file it should set it near zero.

bigtreetech, 2024-08-08:
> Using the USB version of Eddy. Going through the config steps in this wiki. After running PROBE_DRIFT_CALIBRATE, when I home my z gets set to 99.900 and I am unable to manually (or otherwise) move it to 0

There is something messed up with your calibration. It thinks that it is at a crazy low value when it is just probing the bed. Could you do the following:

1. Clear out anything in the klipper auto save part of your config that is related to probing or the eddy. Make sure that you are careful about leaving the format in a way that klipper can still read the other parts of the autosave after you have removed that stuff.
2. Start with the LDC calibration and then run the mapping calibration. Don't run the thermal comp just yet.
3. Pop your config file into a pastebin link as soon as you have done the mapping and send the link here.
4. Run the thermal comp.
5. Pop the new config file into another pastebin link as soon as you are done with the thermal comp and then send the link here.
6. Look inside variables.cfg and also copy that into a pastebin link.

cjhmdm, 2024-08-08:
> > Using the USB version of Eddy. Going through the config steps in this wiki. After running PROBE_DRIFT_CALIBRATE, when I home my z gets set to 99.900 and I am unable to manually (or otherwise) move it to 0
> 
> There is something messed up with your calibration. It thinks that it is at a crazy low value when it is just probing the bed. Could you do the following:
> 
> 1. Clear out anything in the klipper auto save part of your config that is related to probing or the eddy. Make sure that you are careful about leaving the format in a way that klipper can still read the other parts of the autosave after you have removed that stuff.
> 2. Start with the LDC calibration and then run the mapping calibration. Don't run the thermal comp just yet.
> 3. Pop your config file into a pastebin link as soon as you have done the mapping and send the link here.
> 4. Run a probe and take a screenshot of the probe results. Post the screenshot here.
> 5. Run the thermal comp.
> 6. Pop the new config file into another pastebin link as soon as you are done with the thermal comp and then send the link here.
> 7. Run another probe and take a screenshot of the probe results. Paste the screenshot here.
> 8. Look inside variables.cfg and also copy that into a pastebin link.

Hello,

After having gone through the entire calibration process from start to finish several times, the z-position being set to 99.900 specifically happens after running the thermal compensation, each and every time. When I have some more time available later today, I'll go through and post back the information you're requesting.

bigtreetech, 2024-08-08:
> > > Using the USB version of Eddy. Going through the config steps in this wiki. After running PROBE_DRIFT_CALIBRATE, when I home my z gets set to 99.900 and I am unable to manually (or otherwise) move it to 0
> > 
> > 
> > There is something messed up with your calibration. It thinks that it is at a crazy low value when it is just probing the bed. Could you do the following:
> > 
> > 1. Clear out anything in the klipper auto save part of your config that is related to probing or the eddy. Make sure that you are careful about leaving the format in a way that klipper can still read the other parts of the autosave after you have removed that stuff.
> > 2. Start with the LDC calibration and then run the mapping calibration. Don't run the thermal comp just yet.
> > 3. Pop your config file into a pastebin link as soon as you have done the mapping and send the link here.
> > 4. Run a probe and take a screenshot of the probe results. Post the screenshot here.
> > 5. Run the thermal comp.
> > 6. Pop the new config file into another pastebin link as soon as you are done with the thermal comp and then send the link here.
> > 7. Run another probe and take a screenshot of the probe results. Paste the screenshot here.
> > 8. Look inside variables.cfg and also copy that into a pastebin link.
> 
> After having gone through the entire calibration process from start to finish several times, the z-position being set to 99.900 after homing happens after running the thermal compensation, each and every time. When I have some more time available later today, I'll go through and post back the information you're requesting.

OK, this is useful info. There may be a bug with the thermal comp algo that we need to investigate. Thanks for working with us on it.

cjhmdm, 2024-08-08:
> There is something messed up with your calibration. It thinks that it is at a crazy low value when it is just probing the bed. Could you do the following:
>
> 1. Clear out anything in the klipper auto [...]

1. Completed
2.  Completed
3. [printer.cfg](https://gist.github.com/cjhmdm/1047503532acd12d3815e50692e54dfb) & [eddy.cfg](https://gist.github.com/cjhmdm/3b27958191819303f161412e16ab5e51)

4. Unsure what you mean by "run a probe" so including homing results after auto calibration, manual probe calibration results, and bed mesh results. If this is not what you meant, please clarify:

4a. Screenshot after homing (Note: Z-offset is 0.35mm too high from initial calibration at this stage)
[image]

4b. Screenshots and [printer.cfg](https://gist.github.com/cjhmdm/4888fbdc0ccddcaabafd565cf2f0a1a1) after BED_MESH_CALIBRATE METHOD=scan SCAN_MODE=rapid
[image]
[image]

4c. Screenshot and [printer.cfg](https://gist.github.com/cjhmdm/405248a359e6cd893f93f0b11b013cc5) after PROBE_EDDY_CURRENT_CALIBRATE CHIP=btt_eddy after meshing (this unloads the bed mesh profile, however).
Note: Z-offset is 0.375mm too low at this stage
[image]

5. Completed

6. Thermal comp [printer.cfg](https://gist.github.com/cjhmdm/81e754e32d0893192abff41e1c844892)

NOTE: Interesting discovery. When homing directly after running the thermal comp the Z position is arbitrarily set to 99.900 as before, but when homing after allowing everything to cool, the Z position appears to be set correctly, which on my machine is ~3.651, which, again, is 0.35mm too low 

7a. Homing after thermal comp and adjusting z-offset
[image]

7b. Screenshot and [printer.cfg](https://gist.github.com/cjhmdm/1e1cd03e72248aa0340c4818d460fb3f) after running PROBE_EDDY_CURRENT_CALIBRATE CHIP=btt_eddy again
[image]

8. [variables.cfg](https://gist.github.com/cjhmdm/fae6e6781ef8edee69c9d0c0923fb1fa)

bigtreetech, 2024-08-08:
> > > Using the USB...

How high above the bed is the probe when your nozzle is just touching the bed? Also, could you post of a pic of the probe mount? 

From your pictures I am seeing strange behaviour when homing. The z-offset with an Eddy based probe does not define the difference between the nozzle and bed when the probe triggers as is the case with normal probes. This is because eddy probes are constantly streaming data telling klipper the current height so there is no defined trigger point. Instead, the z-offset defines that the nozzle should stop during a probing dive. You have set yours to 2mm and therefore the nozzle should stop at 2mm above the bed during a probing dive. Your output says that it is stopping between 1.2 and 1.3mm which is not right. This indicates that there is either something wrong with your mapping calibration (are you pinching the paper correctly?) or your mounting (is there too much metal around the probe? Is it too close or too far from the bed when the nozzle is resting on the bed?).

Pictures of the mount will help to understand this more. Pictures of the probe height with the nozzle touching the bed will also be good.

cjhmdm, 2024-08-08:
> How high above the bed is the probe when your nozzle is just touching the bed? Also, could you post of a pic of the probe mount?
>
> From your pictures I am seeing strange behaviour when homing. The [...]

I have the probe mounted 2mm above the nozzle using your stealthburner files for the EBB SB2209 CAN (RP2040) in combination with the standard Stealthburner files and Rapido HF hotend. The Eddy probe is mounted in the center channel of the SB X-Carriage mount.

cjhmdm, 2024-08-08:
> How high above the bed is the probe when your nozzle is just touching the bed? Also, could you post of a pic of the probe mount?
>
> From your pictures I am seeing strange behaviour when homing. The [...]

[image]
[image]

cjhmdm, 2024-08-08:
after running through all of the calibrations and letting things cool, then doing some manual z-offset tuning, I'm able to print pretty well... though the images above don't really do the print justice since petg is always messy for me

bigtreetech, 2024-08-08:
I'm running the identical setup to you here on my test rig. My probe is probably mounted a touch higher if anything. I'm able to hit print and forget about it until it is done. I am confident that you can do the same but it will take calibrating properly.

Be sure to zero out the offset value between calibrations as the saved variable may still have old offsets in there. You can use the standard button on the UI to clear the z=offset.

cjhmdm, 2024-08-08:
> I'm running the identical setup to you here on my test rig. My probe is probably mounted a touch higher if anything. I'm able to hit print and forget about it until it is done. I am confident that [...]

I'm mostly fine now, I just need to tinker with the z-offset a bit more (and to be fair, when using the Eddy Coil on my EBB42 and NOT using the beta z-offset, I had the same issues). 

The discovery I made today with the temperature comp issue more or less solves that issue (letting everything cool down before trying to use the printer again).

bigtreetech, 2024-08-13:
> > I'm running the identical setup to you here on my test rig. My probe is probably mounted a touch higher if anything. I'm able to hit print and forget about it until it is done. I am confident that you can do the same but it will take calibrating properly.
> > Be sure to zero out the offset value between calibrations as the saved variable may still have old offsets in there. You can use the standard button on the UI to clear the z offset.
> 
> I'm mostly fine now, I just need to tinker with the z-offset a bit more (and to be fair, when using the Eddy Coil on my EBB42 and NOT using the beta z-offset, I had the same issues).
> 
> The discovery I made today with the temperature comp issue more or less solves that issue (letting everything cool down before trying to use the printer again).
> 
> Side note: Would love to see I2C added to the EBB SB2209 CAN (RP2040) so I can use the Eddy Coil... so far, all things considered, I've had a much better experience with the coil over the usb.

We needed to drop some ports on that board to be able to fit the larger connectors on it. You can use the EBB SB2209 USB that will be released soon as that will allow you to connect the USB eddy to the toolboard as if it were an I2C eddy. 

Closing for now as this looks to be resolved.

closed 2024-08-13 by bigtreetech