bigtreetech/Eddy #49: Failed calibration - incomplete sensor data

Erop1928, 2024-07-28:
Good afternoon. There was a problem with the calibration of the Eddy Coil sensor.
During calibration, after the command "PROBE_EDDY_CURRENT_CALIBRATE CHIP=btt_eddy" and aligning the nozzle with the piece of paper, the sensor begins to calibrate and then displays the error "Failed calibration - incomplete sensor data"
[klippy.txt](https://github.com/user-attachments/files/16405320/klippy.txt)

Erop1928, 2024-07-29:
Okay, this was solved simply by recalibrating from 50-100 times. Why? Who the hell knows

closed 2024-07-29 by Erop1928

bigtreetech, 2024-07-29:
This could be due to:

The sensor being a little too close to the bed (raising by 1mm could solve it).
The LDC current calibration data not being correct (if it detected 15 then try to force it to 16).

Erop1928, 2024-08-13:
Unfortunately, both options did not help, the height was from 0.5mm to 4mm (in increments of 0.5 using a probe) relative to the nozzle and changing the LDC current calibration value between 13 and 17 (by unit)
Calibration passes 1 time out of 100. the same with bed mesh calibration, and any interaction with the sensor.
I changed the wires, made a copper screen for them, crimped new wires and connectors
Nothing helped

Today I tried it all day on the new version of klipper (v0.12.0-278-g3f2ef88e). Of course, all MCUs are also updated to this version.
My setup:
BTT CB2 2GB 32GB
Main MCU: Manta M8p 1.0
CAN Adapter: BTT IO2CAN
Toolhead: BTT EBB36 v1.2
MMU: BTT MMB v1.2
And  Eddy Coil

bigtreetech, 2024-08-15:
> Unfortunately, both options did not help, the height was from 0.5mm to 4mm (in increments of 0.5 using a probe) relative to the nozzle and changing the LDC current calibration value between 13 and [...]

Could you take a pic of your mount and post it?

bigtreetech, 2024-08-15:
OK. I feel like you may have a bad Eddy Coil. Please email luke@bigtree-tech.com and I will get you in touch with the support team.

Erop1928, 2024-08-19:
> OK. I feel like you may have a bad Eddy Coil. Please email luke@bigtree-tech.com and I will get you in touch with the support team.

How long does it take to respond?

bigtreetech, 2024-08-19:
> > OK. I feel like you may have a bad Eddy Coil. Please email [luke@bigtree-tech.com](mailto:luke@bigtree-tech.com) and I will get you in touch with the support team.
> 
> How long does it take to respond?

No email has arrived. Please check that you used the correct address.

Erop1928, 2024-08-20:
> No email has arrived. Please check that you used the correct address.
Seems to be correct (Wrote again)
[image]
Maybe you can write to me at erop1928@gmail.com?

Erop1928, 2024-09-02:
[image]
For some reason, when it comes to Luke, you stop responding. Apparently your "write a letter to Luke about this problem" means "go to hell with your problems"

bigtreetech, 2024-09-03:
> ![browser_24yDBi5r7H](https://private-user-images.githubusercontent.com/33271224/363787729-5ba80880-aed3-4f1b-898e-a0cf765dffbf.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjUzNDQyNjUsIm5iZiI6MTcyNTM0Mzk2NSwicGF0aCI6Ii8zMzI3MTIyNC8zNjM3ODc3MjktNWJhODA4ODAtYWVkMy00ZjFiLTg5OGUtYTBjZjc2NWRmZmJmLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA5MDMlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwOTAzVDA2MTI0NVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWIxZjQyZDMxZTk4NDk2ODA5ZWI3ZGJlOGU2Nzc4Zjg3Nzk0ZWRjM2RhYzlhYzAzZDBhN2Y2MDUyZWJiOGZkZjMmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.8Dv5FuFLHu5Tu99A6T86AGc562VLXwiAr7HY9Pss9jY) For some reason, when it comes to Luke, you stop responding. Apparently your "write a letter to Luke about this problem" means "go to hell with your problems"

I am the CTO of a company and have more to do every single day than almost every other person I meet. Your message has arrived and will be tended to as soon as it reaches the top of the queue.

Erop1928, 2024-09-03:
Ok, thanks for the answer, asking for expectation is always better than uncertainty.
If there is a possibility that this is a marriage, it is much easier for me to buy a new sensor.