bigtreetech/Eddy #50: Eddy Coil - Error during homing probe: Eddy current sensor error

tome9111991, 2024-07-30:
Hello,
some times i got the error "Error during homing probe: Eddy current sensor error" durring QGL on my V2.4.
Mostly when my chamber temperature is >45°C
With a cold chamber there are actually no problems.

what could be the problem?

tome9111991, 2024-07-31:
i have set `reg_drive_current = 16`  and PROBE_EDDY_CURRENT_CALIBRATE CHIP=btt_eddy again works better now.

tome9111991, 2024-08-01:
hmm okay,
i have cold printer today i just power on and want a "QUAD_GANTRY_LEVEL" but not working i got the error "Error during homing probe: Eddy current sensor error"

"BED_MESH_CALIBRATE METHOD=scan SCAN_MODE=rapid" and "PROBE_ACCURACY" works without problems

bigtreetech, 2024-08-01:
> hmm okay, i have cold printer today i just power on and want a "QUAD_GANTRY_LEVEL" but not working i got the error "Error during homing probe: Eddy current sensor error"
> 
> I use the default Voron 2.4 Z-endstop for z-home
> 
> "BED_MESH_CALIBRATE METHOD=scan SCAN_MODE=rapid" and "PROBE_ACCURACY" works without problems

How close were you to the bed when you went to do the QGL? I have noticed that the latest klipper merge has some very strict checks in place that we are going to have to discuss with Kevin to see if there is a better way to implement. If the oscillator is clipping when the QGL is requested then it throws this error. The oscillator will clip if the probe is too close to the bed before the QGL starts. 

It may be a good idea for you to raise the probe up a bit and leave the drive current at 16. Of course, you'll need to run the Eddy calibration again after raising it but that is quite quick.

tome9111991, 2024-08-01:
for test i have  drive current set to 15, works but bit unstable.

i got the error right after the z-homing. the printhead is not even in its qgl position.
I'm z10 after z homing.
maybe the coil detect the metall pin from z-endstop and then the error appears?

bigtreetech, 2024-08-01:
> for test i have drive current set to 15, works but bit unstable.
> 
> i got the error right after the z-homing. the printhead is not even in its qgl position. I'm z10 after z homing. maybe the coil detect the metall pin from z-endstop and then the error appears?

It could be this. It will throw the error when the coil is giving a bad reading. I don't like the way that it is currently implemented since the coil may be absolutely anywhere before a QGL starts and there is no way of guaranteeing the location. We will work with Kevin to try and improve the algorithm. What happens if you manually move the toolhead over the bed after homing and then tell it to perform a QGL?

tome9111991, 2024-08-01:
currently i use 
`[gcode_macro G32]
gcode:
    SAVE_GCODE_STATE NAME=STATE_G32
    G90
    G28
    QUAD_GANTRY_LEVEL
    G28 Z
    G0 X150 Y150 Z30 F3600
    RESTORE_GCODE_STATE NAME=STATE_G32
`

I will try to move to a "save" position before start "QUAD_GANTRY_LEVEL"

tome9111991, 2024-08-01:
i switch back to drive current 16.
toolhead is X150 Y150 Z150 i use "QUAD_GANTRY_LEVEL" command and got instantly the "Error during homing probe"
after the message toolhead moves to first QGL position but till Z150
then QGL aborts

freakydude, 2024-08-03:
I have the Eddy Coil version. It's connected to EBB42 CAN and that's connected via CAN to Manta M8P+CB1.
It's installed on an Artillery Sidewinder X2 with H2 V2S Extruder.

I seem to be having the same problems. 

Running `LDC_CALIBRATE_DRIVE_CURRENT` with 20mm coil spacing also gives me a `reg_drive_current = 15`. 
Running the PaperTest with `PROBE_EDDY_CURRENT_CALIBRATE` seems to work fine.
After saving the config and restarting the Klipper firmware, I always get the `Error during homing probe: Eddy current sensor error' message.

After reading this issue, I increase `reg_drive_current` to 16 and the error goes away.

So if I can help/provide information, let me know

_By the way - at least for me there is also a lot of confusion. The descriptions here https://github.com/bigtreetech/Eddy on how to use Eddy with Klipper differ from those in the Klipper Docu itself. (And from things that the search engine spits out)_

_In the Klipper Docu no additional macros are needed for homing - here they are. Seems that this here is correct.
It would be really good if you could describe this without 100 Important, Tips and Notes. Just straight forward. Just the normal case step by step. Once for the coil version and again for the USB+Tempsensor version_

bigtreetech, 2024-08-12:
> i switch back to drive current 16. toolhead is X150 Y150 Z150 i use "QUAD_GANTRY_LEVEL" command and got instantly the "Error during homing probe" after the message toolhead moves to first QGL position but till Z150 then QGL aborts
> 
> chamber is below 30°C
> 
> EDIT: After heating the hotend few min at 150°C QGL is working
> 
> EDIT2: QGL not works with cold hotend but works if you close to bed (bed is cold)
> 
> maybe a new implemition for QGL will fix this? The coil should start measuring when the toolhead is in the QGL position.

When I mentioned to raise your probe I meant to physically move the distance between it and the nozzle. We have revised our recommended installation height to be between 2mm and 3mm above the nozzle. If yours is too low then this could be causing the errors.

tome9111991, 2024-08-12:
> When I mentioned to raise your probe I meant to physically move the distance between it and the nozzle. We have revised our recommended installation height to be between 2mm and 3mm above the nozzle. If yours is too low then this could be causing the errors.

Eddy is on highest postion as possible in my V2.4
i use a Dragon HF hotend maybe heatblock is to close?

bigtreetech, 2024-08-12:
> > When I mentioned to raise your probe I meant to physically move the distance between it and the nozzle. We have revised our recommended installation height to be between 2mm and 3mm above the nozzle. If yours is too low then this could be causing the errors.
> 
> Eddy is on highest postion as possible in my V2.4 i use a Dragon HF hotend maybe heatblock is too close?

Use a few thin objects (such as business cards or similar) and stack them between the eddy and the bed when the nozzle is just touching the bed. Then remove them and measure their thickness. Let me know the distance and I'll let you know if it could be your issue.

tome9111991, 2024-08-12:
The offset between nozzle and eddy is 2.50 - 3mm

bigtreetech, 2024-08-12:
> The offset between nozzle and eddy is 2.50 - 3mm. That's what I measured.

OK then let's try the opposite with you. Could you try to lower it to between 1.5mm or 2mm? Use business cards again to get the value right. Once that is done, see if it is more stable across the temp range. If not then lower the drive current to 15. Also, ensure that the side of the Eddy with the circuitry (the back) is facing away from the hotend.

tome9111991, 2024-08-12:
i have change to 1.7mm maybe but not work with drive current 16 is printer cold
drive current 15 is working

bigtreetech, 2024-08-12:
> i have change to 1.7mm maybe but not work with drive current 16 is printer cold drive current 15 is working

The closer you get to the bed, the lower the drive current that is needed. Does it continue to work with 15 when you warm the printer?

tome9111991, 2024-08-12:
drive current 15 not work with warm printer :(

my workaround:
set drive current to 16
preheat hotend at 150°C for 5min 
and eddy works well.

bigtreetech, 2024-08-12:
> drive current 15 not work with warm printer :(
> 
> my workaround: set drive current to 16 preheat hotend at 150°C for 5min and eddy works well.

Is it installed in the standard location on the 2.4?

tome9111991, 2024-08-12:
yes

bigtreetech, 2024-08-13:
Yours is a very strange case. In every other case we have worked with, there is a combination of mounting height and LDC drive current that allows for use over the entire temperature range. Yours is the only one where we have seen this. Please send an email to luke@bigtree-tech.com so that I can discuss this with you further.

freakydude, 2024-08-13:
@bigtreetech Just a theoretical question:

Your Eddy probe bottom has an angle of 1° degree. So 2.5mm z-offset measured with business cards would measure the deepest point, the back of the probe. The coil is (according to your sketchups) not in the middle, but near to it. 

for a bracket design, would it be more correct to use the exact coil position and adjust it to a 2.5mm z offset - or (as measured with business cards) the bottom back of the eddy probe?
or is it ultimately not a calculated figure, but one that works well in practice?

thanks

bigtreetech, 2024-08-13:
> @bigtreetech Just a theoretical question:
>
> Your Eddy probe bottom has an angle of 1° degree. So 2.5mm z-offset measured with business cards would measure the deepest point, the back of the probe. [...]

The guidance on 2.5mm is actually specifying the height of the lowest part of the casing and not the coil. The guidance factors into consideration the slope on the casing (note that the slope is not present on the PCB) and the distance between the PCB and the case.