bigtreetech/Eddy #51: Does the readme needs updating now that it seems like most/all? patches are in main klipper?

zingo, 2024-08-06:
E.g. Is this still true? And if not what is missing in the main repo? Most patches seem to have been merged, only a z-level patch seem to be missing that might not be needed?

```
As it stands, Eddy requires the use of BTT's fork of klipper found [HERE](https://github.com/bigtreetech/klipper). This is included in the guide as the first step.

The pull request to merge this into mainline klipper has been made and it should be integrated shortly. Until then this is a STRICT REQUIREMENT.
```

zingo, 2024-08-06:
As a note  I (and some others) run on the main klipper repo and it seem to work ok after the pull request went in around 17-18 July :)

bigtreetech, 2024-08-06:
We are in the process of updating it right now. Updates should all come through before the end of the week.

closed 2024-08-06 by bigtreetech

zingo, 2024-08-06:
Great thanks a lot!

Kooramah, 2024-08-09:
hi there, any update if this is on the main klipper repo now?

smwoodward, 2024-08-09:
> hi there, any update if this is on the main klipper repo now?

It is, and I've been using it for a while on the main branch of Klipper.  There was one command during setup/calibration that was different in the main line version of Klipper, but I don't remember what it was now.

Kooramah, 2024-08-09:
> > hi there, any update if this is on the main klipper repo now?
> 
> It is, and I've been using it for a while on the main branch of Klipper. There was one command during setup/calibration that was different in the main line version of Klipper, but I don't remember what it was now.

perfect, let me give it a shot and figure out which command you're talking about