bigtreetech/Eddy #52: "Error during homing z: Eddy current sensor error" after updating to mainline klipper

cjhmdm, 2024-08-14:
I saw that Eddy is now supported in mainline klipper so figured I'd update everything with the hopes of it resolving some of my z-offset issues.

After rebuilding and flashing klipper for all of my devices (BTT Manta M8P V2.0, BTT EBB SB2209 CAN (RP2040) with Knomi and SB toolhead, and Eddy) and going through configuration steps 1-21 then trying to home for steps 27-29 I receive the following errors:

4:04 PM Error during homing z: Eddy current sensor error
4:04 PM Error during homing z: Eddy current sensor error

Here's my updated [printer.cfg](https://gist.github.com/cjhmdm/d26374865b635cb2fdd06b007634cb93) and [eddy.cfg](https://gist.github.com/cjhmdm/acd87ccfaac0bf66ba23cdc790e788aa)

cjhmdm, 2024-08-14:
Forgot to note, I followed the steps in the FAQ related to this issue, but it persists.

cjhmdm, 2024-08-15:
update: I saw the closed thread on this, increasing reg_drive_current from 15 to 16 didn't initially provide positive results. I then powered down my printer for several hours while at my son's practice, then tried again when I got home. I was able to home okay, and even run a bed mesh as well as the temperature compensation. However, once saving the config and restarting the firmware, I was unable to home again, receiving the same error. My bed temp at this time went from 100-90c. 

I then tried changing reg_drive_current from 16 to 17, which allowed me to home again.

For further testing, I'm going to move the probe up from 2mm to 3mm once my space finishes printing.

bigtreetech, 2024-08-15:
> update: I saw the closed thread on this, increasing reg_drive_current from 15 to 16 didn't initially provide positive results. I then powered down my printer for several hours while at my son's [...]

Are you using a standard voron mount? 

There will be a combination of mount height and LDC current that works best given your mounting setup. Try LDC current at 17 and see if it is stable across your temp range at the current height but bear in mind that you need to run the mapping each time you change the drive current. Find a drive current at the mounting height that gives you good homing across the temp range and then run the thermal cali.

cjhmdm, 2024-08-15:
> Are you using a standard voron mount?
>
> There will be a combination of mount height and LDC current that works best given your mounting setup. Try LDC current at 17 and see if it is stable across [...]

Yes, I'm using the standard Voron x-carriage mount for CW2 & SB.

After more testing, I appear to have been successful in finding a solution to this issue by increasing the eddy offset to print bed from 2mm to 3mm and setting the drive current back to 16 (down from 17 when mounted at 2mm). I have to do more testing to ensure everything is stable again, but it has been so far. 

For testing purposes, is there a max safe LDC current?

Another thing to note is the beta z-offset still appears to need work. I can't pinpoint exactly what's happening, but it appears as if the nvm_offset in variables.cfg seems to be overridden, specifically when it comes to running bed mesh calibration.

Lastly, I think there needs to be a correction in your sample configs. For the BED_MESH_CALIBRATE macro, the note says to run BED_MESH_SCAN which doesn't appear to be a valid command.

ahmedbebars, 2024-08-19:
> I saw that Eddy is now supported in mainline klipper so figured I'd update everything with the hopes of it resolving some of my z-offset issues.
>
> After rebuilding and flashing klipper for all of my [...]

I have the same problem

ahmedbebars, 2024-08-19:
> I saw that Eddy is now supported in mainline klipper so figured I'd update everything with the hopes of it resolving some of my z-offset issues.
>
> After rebuilding and flashing klipper for all of my [...]

Please attach file macro 
And start print

cristiansen-12, 2024-08-25:
I am also running into this error using BTT Eddy.

Started off with the suggested  reg_drive_current = 16 and z-offset  for  probe_eddy_current = 0.5. -> Would not work at all. Just got the error all the time.
Try #1: I increased current up to 19 with no luck. Then changed the z-offset for probe_eddy_current  from 0.5mm to 2mm
and it was the first time I was able to reliably do a Home-Z! Ran the temperature drift calibration up to 80 degrees with no problem. Did a Home-All and I get again the error.
Try #2: Restored reg_drive_current to 16. Again I was able to do a Home-All. Ran the temperature drift calibration up to 80 degrees with no problem. Did a Home-All and I get again the error. Three times in a row the same. :| 

I changed nothing with the setup, but let the BTT Eddy temperature cool down from the circa  80 degrees where it was to circa 46, and again I was able again to Home-All axis. Unless it is a coincidence, it looks to me that the probe temperature itself plays a part into this generic error.

For refrence, I have attached my eddy.cfg and printer.cfg 
[config-20240825-203546.zip](https://github.com/user-attachments/files/16741526/config-20240825-203546.zip)

duckarp, 2024-09-09:
Maybe this will help debugging this issue further, here's my experience:

I followed the FAQ and used the mount from the folder B1 (as I have a BIQU B1 printer). This mount places the probe about 5mm from the nozzle. It worked somewhat, but was giving me wrong z-offsets, so I decided to trust the guide and modify the mount so the probe sits 2-3mm from the nozzle. I finished first two steps, but was facing this error "Error during homing z: Eddy current sensor error". I experimented a little and it seems it works with current = 16 and z-offset 3. Finally I can home again. We'll see how it goes further...

freakydude, 2024-09-13:
I also have the Eddy Coil version. I have this problem often after a printer power off/on cycle. Just try to home. It will home x & y and instead of moving down the z axis, I immediately get the mentioned `Error during homing z: Eddy current sensor error`. 
It does not matter if the printer is at room temperature or if the print bed is at 70°C.

Sometimes it helps to restart the firmware, sometimes it helps to manually move the extruder to a shorter z-distance. 

I have not found a valid reason yet - but I can say that after these steps it works without changing the config or the mount position. So there is a problem - in hardware or software ... i don't know

Generally I use it with a current of 16 and a real world z-offset of 2.5mm connected to the ebb can v1.2 pcb

kdm303, 2024-11-11:
This issue has been a recurring problem on one of my printers. Increasing timeout appears to get around it but when you check for updates it will mark the repo dirty. Recover the repo, update and change it again if need be, at least I'm printing.

File: ~/klipper/klippy/mcu.py

Change: TRSYNC_TIMEOUT from 0.025 to 0.050

hammelpein, 2024-11-24:
I had the same error message, but I doubt it is the same issue.
After digging the source and the documentation for the LDC and adding a decoded output for the error bits, I found the reason was **amplitude_low** error. Which kind of makes sense as the gantry tends to be way lower at the z endstop. Causing this error to pop up when trying to measure the front corners.

The fix to this problem was to make a config file setting for the error config register. I doubt that this works for everyone, and even is required for everyone, but for me clearing bit 11 got me printing again.
The only thing that really helped was adding a simple decoding for the error bits.  :)

freakydude, 2024-11-24:
@hammelpein, you could try making a pull request to the klipper project. The fact that there is more information about the bug, and the LDC has more information about what is wrong, makes a pull request worthwhile.

hammelpein, 2024-11-26:
> @hammelpein, you could try making a pull request to the klipper project. The fact that there is more information about the bug, and the LDC has more information about what is wrong, makes a pull request worthwhile.

Might consider it as soon as it's up again. Made some major hardware changes and it's currently laying on the side.

yael-lorenzo, 2025-03-13:
What temperature do you have in the BTT Eddy (and MCU) while you are doing the process?

real-artswan, 2025-05-29:
I have Eddy Coil (no mcu on the eddy) and facing the same issue. In a hot environment Eddy doesn't work. Here are a set of screens I made after:
1. Printed a model in ABS  (260/100 C) starting from a cold printer
2. Started the next model, got the error
3. Switched printer off/on, waited some time

[image]
[image]
[image]

After some cooldown I was able to home and start print, but it was failed during the second pass of Z-tilt.

real-artswan, 2025-05-29:
[image]

Finally it's started after some more cooldown

[image]

real-artswan, 2025-05-30:
A similar issue with a cartographer: https://github.com/Cartographer3D/cartographer-klipper/issues/209. When I switched from it, eddy was fine and I printed in series many times, but then I updated Klipper. May be it's not related to a Klipper update, I can't nail it down.

bigtreetech, 2025-05-30:
We strongly recommend using the new eddyNg which will get you tap functionality. It will also help you to determine appropriate mounting heights and coil drive currents to ensure that you get stable operation across the temp range. https://github.com/vvuk/eddy-ng

real-artswan, 2025-06-11:
so, to be comfotable somehow, one must use a `Kalico` fork with that. so-so

real-artswan, 2025-06-11:
They mean that they would do coarse homing? But it doesn't work at all if a coil is hot. The error appears very before any movement.
That "ng" try is a good try, but I don't believe it will help. Even more (now if I want to use it) must care about tails of plastic on the nozzle.

real-artswan, 2025-06-11:
and with that you'll get a custom klipper which is not updatable