bigtreetech/Eddy #55: Nozzle crash after performing Z_TILT_ADJUST with the Eddy coil

cssi-dev, 2024-08-20:
I tried printing the first part after switching to the new probe. I was suprised to learn, that my print was canceled during the print start macro.
Error:

> probe triggered prior to movement

with `horizontal_move_z: 5 `

So I switched it to `horizontal_move_z: 10`(even though In the bed section the horizontal_move_z is just 2 ), to then get the nozzle crash because the printer after getting all the readings moved all z axes way to far.

I get that you are not supposed to use the eddy coil for homing because of the temprature drift. But please let us use the Tilt adjust. I don't want to switch every 5-10 prints between the probes just to readjust the z axis when it gets out of wack.

Steps to reproduce:

Home the Printer
Start `Z_TILT_ADJUST`
Be ready to shutdown the printer befor something bad happens
[klippy.log](https://github.com/user-attachments/files/16667168/klippy.log)

cssi-dev, 2024-08-23:
Since it seams that I won't get an answer any time soon, can you at least point me to the correct file so I can at least try to creat a fork with a fix. Thanks already

BTTJUSTIN, 2024-09-06:
Thanks for reporting. We will get our engineering team to look into this.

cssi-dev, 2024-09-07:
Thank you for your response. The klipper support and I already found a solution.(The initial z offset was way to low). I would advice to change the guide so that it is no longer inside the Probe itself and that it has to be calibrated with the Probe query command (which is still missing). 
By the way is temprature compensation apperently always requiert since ztilt only works when you do the calibration(even  if you would not need it)

bigtreetech, 2024-09-09:
> Thank you for your response. The klipper support and I already found a solution.(The initial z offset was way to low). I would advice to change the guide so that it is no longer inside the Probe [...]

That is not how Z-offset works with these probes. See the FAQ section of the readme in this repo for a link to an explanation of how z-offset works with these probes. It is very different to standard z-offset and the query_probe command has no meaning.

closed 2024-09-09 by bigtreetech