bigtreetech/Eddy #57: Pausing at points during bed mesh after going to klipper main release

Wakinyantanka, 2024-08-23:
So everything is working great except while doing bed mesh before each print. During the scan it pauses at equal points along the scan path. Scan is good and first layers are perfect it's just really slow now with the pausing.

tome9111991, 2024-08-25:
You can use 

BED_MESH_CALIBRATE METHOD=rapid_scan

Wakinyantanka, 2024-08-26:
I'm using that command and it still pauses at each point along the scan path.

krautech, 2024-08-26:
There's a whole section at the bottom of the guide devoted to this btw.

Also 

Without your printer.cfg and klippy.logs I feel like you're asking us to fix your printer with no idea what's happening lol

Wakinyantanka, 2024-08-26:
Perhaps I wasn't clear enough, the printer is not broken and the scan is NOT z hopping during the bed mesh. The printer doesn't need fixed so I didn't think the klippy log or printer.cfg was necessary. The scan simply pauses at specific points along the scan path. I wasn't sure if this had something to do with going to the klipper main release or not. 
Also, there's nothing in the documentation about this...lol

bigtreetech, 2024-08-29:
Please ensure that you update all of your macros to the latest in the repo after updating to the klipper main branch. Your issue is almost certainly because your macro is calling an old version of the bed mesh command. This is exactly the behaviour that will be observed when that is happening.

closed 2024-08-29 by bigtreetech

mikehatch, 2025-11-12:
Just did some internet spelunking around this issue - my eddy-ng was also stopping at the mesh points while scanning even though I've seen videos (Printasauraus) where is scanned continuously.  

I found that using the following parameters does the bed_mesh_calibrate continuously
`BED_MESH_CALIBRATE METHOD=scan SCAN_TYPE=rapid`

Most examples show the parameter as METHOD=rapid_scan, but this doesn't execute the rapid way...