bigtreetech/Eddy #59: Probe triggered prior to movement after moving to klipper main release

madk71, 2024-08-28:
Hi, I was able to have the eddy working with very few issues using your klipper fork, so I thought that moving to the master release probably I can have some fixes. I modified the repository in kiauh, updated all, and reflashed the eddy and I was unable to run the PROBE_EDDY_CURRENT_CALIBRATE_AUTO, as soon I press any of the buttons to adjust the z I receive the error that axes need to be homed (x and y were homed) and of course, the z can't be homed if I don't run the current calibration. I copied the old value of the PROBE_EDDY_CURRENT_CALIBRATE_AUTO, restarted klipper, and ran PROBE_EDDY_CURRENT_CALIBRATE_AUTO and this time it was working. So I calibrated eddy and saved it. At this point, I did the bed mesh and then tried to level the bed with SCREWS_TILT_CALCULATE after homing of course, but I received the Probe triggered prior to movement error. The same is true for the Z_TILT_ADJUST.
I created a new sd card with a fresh install of everything and reflashed the eddy again, but I received all the same errors and problems.
attached you can find my cfg files and log.

[klippy.log](https://github.com/user-attachments/files/16788858/klippy.log)
[config-files.zip](https://github.com/user-attachments/files/16788897/config-files.zip)

bigtreetech, 2024-08-29:
Please adjust your macro to match the one here and let me know if it helps:

[gcode_macro SET_Z_FROM_PROBE]
gcode:
    {% set cf = printer.configfile.settings %}
    SET_GCODE_OFFSET_ORIG Z={printer.probe.last_z_result - cf['probe_eddy_current btt_eddy'].z_offset + printer["gcode_macro SET_GCODE_OFFSET"].runtime_offset}
    G90
    G1 Z{cf.safe_z_home.z_hop}

madk71, 2024-08-29:
Hi, Thanks. the new code fixed the problem

closed 2024-08-30 by BTTJUSTIN