bigtreetech/Eddy #61: Unknown command:"TEMPERATURE_PROBE_CALIBRATE"

TheZDog, 2024-08-31:
installing btt eddy and im up to 

32. Run TEMPERATURE_PROBE_CALIBRATE PROBE=btt_eddy TARGET=56 STEP=4

and i get this error

Unknown command:"TEMPERATURE_PROBE_CALIBRATE"

Ender 3 V2 (4.2.7)

printer.cfg

`# This file contains pin mappings for the stock 2020 Creality Ender 3
# V2. To use this config, during "make menuconfig" select the
# STM32F103 with a "28KiB bootloader" and serial (on USART1 PA10/PA9)
# communication.

# If you prefer a direct serial connection, in "make menuconfig"
# select "Enable extra low-level configuration options" and select
# serial (on USART3 PB11/PB10), which is broken out on the 10 pin IDC
# cable used for the LCD module as follows:
# 3: Tx, 4: Rx, 9: GND, 10: VCC

# Flash this firmware by copying "out/klipper.bin" to a SD card and
# turning on the printer with the card inserted. The firmware
# filename must end in ".bin" and must not match the last filename
# that was flashed.

# See docs/Config_Reference.md for a description of parameters.
[include bigtreetech-eddy-main.cfg]
[include macro.cfg]
##[include timelapse.cfg]
[include mainsail.cfg]
[gcode_arcs]

[exclude_object]

[display_status]

[pause_resume]

[stepper_x]
step_pin: PB9
dir_pin: PC2
enable_pin: !PC3
microsteps: 16
rotation_distance: 40
endstop_pin: ^PA5
position_endstop: 0
position_max: 239
position_min: 0
homing_speed: 50

[stepper_y]
step_pin: PB7
dir_pin: PB8
enable_pin: !PC3
microsteps: 16
rotation_distance: 40
endstop_pin: ^PA6
position_endstop: -4
position_min: -4
position_max: 224
homing_speed: 50

[stepper_z]
step_pin: PB5
dir_pin: !PB6
enable_pin: !PC3
#endstop_pin: ^PA7
[... 396 lines cut ...]
#    {% endfor %}
#    {% set rawparams=paramList|join(' ') %}
#    SET_GCODE_VARIABLE MACRO=SET_GCODE_OFFSET VARIABLE=runtime_offset VALUE=0
#  {% endif %}
  SET_GCODE_OFFSET_ORIG { rawparams }

# This macro automates a lot of the frequency mapping process and simplifies the steps significantly.
[gcode_macro PROBE_EDDY_CURRENT_CALIBRATE_AUTO]
gcode:
  BED_MESH_CLEAR
  G28 X Y
  G90 # Abs positioning
  G1 X{ printer.toolhead.axis_maximum.x/2 } Y{ printer.toolhead.axis_maximum.y/2 } F6000
  {% if 'z' not in printer.toolhead.homed_axes %}
    SET_KINEMATIC_POSITION Z={ printer.toolhead.axis_maximum.z-1 } # Allows the user to work it down until it touches.
  {% endif %}
  PROBE_EDDY_CURRENT_CALIBRATE {rawparams}

#This macro is optional but useful if you want to run a rapid scan before each print. Simply uncomment it and add BED_MESH_SCAN to your print start code.
#[gcode_macro BED_MESH_CALIBRATE]
#rename_existing: BTT_BED_MESH_CALIBRATE
#gcode:
#  SET_GCODE_VARIABLE MACRO=_KNOMI_STATUS VARIABLE=probing VALUE=True #Only uncomment this line if using a KNOMI and then remove the BED_MESH_CALIBRATE macro from KNOMI.cfg
#  BTT_BED_MESH_CALIBRATE METHOD=rapid_scan
#  SET_GCODE_VARIABLE MACRO=_KNOMI_STATUS VARIABLE=probing VALUE=False #Only uncomment this line if using a KNOMI and then remove the BED_MESH_CALIBRATE macro from KNOMI.cfg`

bigtreetech, 2024-08-31:
Update your host klipper.

closed 2024-08-31 by bigtreetech

Joldiges, 2025-04-14:
> Update your host klipper.

I'm in the latest, 1.13.0, and i'm having the same issue.  
Similarly, it's not allowing for the mesh command as given