bigtreetech/Eddy #62: Eddy Coil throwing error on bed mesh scan: Unable to obtain probe_eddy_current sensor readings

Rainforestnomad, 2024-08-31:
I have created a fresh install of Raspian and Klipper, on a machine which previously I had working with the Eddy Coil. Now though, when I go to complete a bed mesh scan, the probe scans the bed and then throws this error:

Unable to obtain probe_eddy_current sensor readings

What could be causing this issue?

BTTJUSTIN, 2024-09-04:
Thanks for reaching out. Our support team does not operate using the GitHub platform but we would be happy to help you via 
email based ticketing system. Please visit https://biqu.equipment/pages/technical-support and submit a ticket and we will be in touch with you soon

closed 2024-09-04 by BTTJUSTIN

bigtreetech, 2024-09-04:
Just to elaborate on this, our traditional support channel does not operate over github. When a new product is released or when major changes are made to a product, we may service some tickets which are classed as support requests via github because they may actually hold clues to firmware issues that we need to dig deeper into. After some time, we transition to the standard support flow where we ask users to please submit a ticket using the standard ticketing process. If a user has found a genuine firmware or hardware issue after following all of the steps in the manual and has provided klipper logs and/or other data to show that it is an issue then we address it using github issues.

Rainforestnomad, 2025-01-03:
Issue is closed, however I had to migrate from Raspberry Pi Zero 2W to a Pi4, I believe the 512mb ram was insufficient on the Zero2, and the 1gb Ram of the Pi4 allowed the system to work as intended.