bigtreetech/Eddy #67: Eddy Coil vs Knomi v2 big problem

borysne, 2024-09-10:
hi, I have a very rare compatibility problem! Most likely, no one will help me solve it, but I will write this!
My equipment:
Voron 2.4  board Manta-M8P v2 + CB1, head Stealthburner with EBB 2240, Eddy Coil , Knomi v2
Eddy coil is connected to the i2c port on the MP8 v2 board
Knomi v2 is connected as usual via wi-fi
Eddy coil is configured, works fine, scans and levels the table!
"""The essence of the problem: i turn on the 3d printer, while the Klipper is loading, the Knomi v2 loses wi-fi, as soon as i turn off the power of the Eddy coil, the Knomi v2 starts working as it should!"""
I have already tried everything! I changed the Knomi v2 wifi antenna, connected separate 5v. tried different firmware Knomi v2. 
I did a bunch of experiments! As soon as the Knomi v2 is inside the 3D printer, the Wi-Fi connection is immediately lost!
I suspect that the problem is in the large magnetic field!!!
it is also interesting that the CB1 board has no problems with wi-fi!
the CB1 board is under the table, maybe that's why!
Gentlemen, do you have any thoughts???

BTTJUSTIN, 2024-09-11:
Thanks for reaching out. Our support team does not operate using the GitHub platform but we would be happy to help you via email based ticketing system. Please visit https://biqu.equipment/pages/technical-support 
and submit a ticket and we will be in touch with you soon.

looxonline, 2024-09-11:
> hi, I have a very rare compatibility problem! Most likely, no one will help me solve it, but I will write this! My equipment: Voron 2.4 board Manta-M8P v2 + CB1, head Stealthburner with EBB 2240, Eddy Coil , Knomi v2 Eddy coil is connected to the i2c port on the MP8 v2 board Knomi v2 is connected as usual via wi-fi Eddy coil is configured, works fine, scans and levels the table! """The essence of the problem: i turn on the 3d printer, while the Klipper is loading, the Knomi v2 loses wi-fi, as soon as i turn off the power of the Eddy coil, the Knomi v2 starts working as it should!""" I have already tried everything! I changed the Knomi v2 wifi antenna, connected separate 5v. tried different firmware Knomi v2. I did a bunch of experiments! As soon as the Knomi v2 is inside the 3D printer, the Wi-Fi connection is immediately lost! I suspect that the problem is in the large magnetic field!!! it is also interesting that the CB1 board has no problems with wi-fi! the CB1 board is under the table, maybe that's why! """Sometimes an error appears MCU shutdown: i2c timeout !!!""" Gentlemen, do you have any thoughts???

I doubt that it has much to do with the magnetic or electrical field. It is more likely that the 5V power supply is struggling to keep up with the Eddy Coil and the KNOMI at the same time. When you say that you are turning off the power of the Eddy do you mean that you are unplugging it?

borysne, 2024-09-11:
5v power supply. This is the first thing I checked! Everything is fine with
it! KNOMI consumes an average of 5v. 0.2A
I pulled out the knomi and connected it via USB from a laptop, and brought
it closer to the eddy, the Wi-Fi signal was immediately lost! And one more
piece of news! Today I reflashed the firmware on all the boards and it got
better! But the signal is still lost!

borysne, 2024-09-11:
I make video!
https://youtu.be/d3NsCouxJLk

looxonline, 2024-09-12:
> […](#)

[image]

borysne, 2024-09-12:
sorry! try again i opened access to the video

borysne, 2024-09-12:
Regarding the 5v power supply! I changed it, the EBB2240 was giving out 4.95v power. I changed the circuit a bit, installed a 9.45 kOhm resistor, and raised the output voltage to 5.15v.
[image]

borysne, 2024-09-18:
klippy log were!
[klippy (1).log](https://github.com/user-attachments/files/17048391/klippy.1.log)