bigtreetech/Eddy #69: Error on 'PROBE_EDDY_CURRENT_CALIBRATE': missing CHIP

Gansemagne, 2024-09-14:
Hello.

After installing and flashing my Eddy USB and build up my printer i want to Calibrate the EDDY and got this error:
"Error on 'PROBE_EDDY_CURRENT_CALIBRATE': missing CHIP"

BTT Justin, I would rather get this solved than have to go through a ticket process again, let me put this up.

My config file from Eddy is:

[include mainsail.cfg]

[mcu host]
serial: /tmp/klipper_host_mcu

#serial: /dev/serial/by-id/usb-Klipper_rp2040_12345-if00

[mcu]
serial: /dev/serial/by-id/usb-Klipper_stm32g0b1xx_500042001450415833323520-if00

# The MCU section only applies to the Eddy USB. For Eddy Coil you will use the MCU name of the toolboard that you connected the Eddy Coil to.
[mcu eddy]
serial: /dev/serial/by-id/usb-Klipper_rp2040_5044340310D4C61C-if00 # This is the serial address of your eddy probe. This can be found by using the terminal of your klipper instance (typically through SSH) and using the command ```ls /dev/serial/by-id``` 
restart_method: command

[temperature_sensor btt_eddy_mcu]
sensor_type: temperature_mcu # Sets the type of sensor for Klipper to read
sensor_mcu: eddy # Sets the MCU of the eddy probe tempereature sensor
min_temp: 10 # Sets the minimum tempereature for eddys tempereature sensor to operate
max_temp: 100 # Sets the maximum tempereature for eddys tempereature sensor to operate

[probe_eddy_current btt_eddy]
sensor_type: ldc1612
z_offset: 2.5
#i2c_address:
i2c_mcu: eddy  # This value is good for the Eddy USB but would need to be adjusted for the Eddy Coil according to the MCU you have used.
i2c_bus: i2c0f # This value is good for the Eddy USB but would need to be adjusted for the Eddy Coil according to the I2C bus you have used.
# Measure the offsets below using the method described here: https://www.klipper3d.org/Probe_Calibrate.html#calibrating-probe-x-and-y-offsets
x_offset: -34.25
y_offset: -14.66

# This section is only relevant for Eddy USB. Comment it out for Eddy Coil.
[temperature_probe btt_eddy]
sensor_type: Generic 3950
sensor_pin: eddy:gpio26
horizontal_move_z: 2

[bed_mesh]
horizontal_move_z: 2
speed: 200
mesh_min: 10, 10  
mesh_max: 339, 350 
probe_count: 9, 9
algorithm: bicubic

[safe_z_home]
home_xy_position: 170, 170
z_hop: 3
z_hop_speed: 25
speed: 200

# Eddy Macros
#This secton contains macros and related config sections. Some should be used, others are optional. Read the comment above each one to find out whether or not to uncomment it for your installation.

[save_variables]
filename: ~/printer_data/config/variables.cfg

[force_move]
enable_force_move: True 

[delayed_gcode RESTORE_PROBE_OFFSET]
initial_duration: 1.
gcode:
  {% set svv = printer.save_variables.variables %}
  {% if not printer["gcode_macro SET_GCODE_OFFSET"].restored %}
    SET_GCODE_VARIABLE MACRO=SET_GCODE_OFFSET VARIABLE=runtime_offset VALUE={ svv.nvm_offset|default(0) }
    SET_GCODE_VARIABLE MACRO=SET_GCODE_OFFSET VARIABLE=restored VALUE=True
  {% endif %}

[gcode_macro G28]
rename_existing: G28.1
gcode:
  G28.1 {rawparams}
  {% if not rawparams or (rawparams and 'Z' in rawparams) %}
    PROBE
    SET_Z_FROM_PROBE
  {% endif %}

[gcode_macro SET_Z_FROM_PROBE]
gcode:
    {% set cf = printer.configfile.settings %}
    SET_GCODE_OFFSET_ORIG Z={printer.probe.last_z_result - cf['probe_eddy_current btt_eddy'].z_offset + printer["gcode_macro SET_GCODE_OFFSET"].runtime_offset}
    G90
    G1 Z{cf.safe_z_home.z_hop}

[gcode_macro Z_OFFSET_APPLY_PROBE]
rename_existing: Z_OFFSET_APPLY_PROBE_ORIG
gcode:
  SAVE_VARIABLE VARIABLE=nvm_offset VALUE={ printer["gcode_macro SET_GCODE_OFFSET"].runtime_offset }

[gcode_macro SET_GCODE_OFFSET]
rename_existing: SET_GCODE_OFFSET_ORIG
variable_restored: False  # Mark whether the var has been restored from NVM
variable_runtime_offset: 0
gcode:
#  {% if params.Z_ADJUST %}
#    SET_GCODE_VARIABLE MACRO=SET_GCODE_OFFSET VARIABLE=runtime_offset VALUE={ printer["gcode_macro SET_GCODE_OFFSET"].runtime_offset + params.Z_ADJUST|float }
#  {% endif %}
#  {% if params.Z %}
#    {% set paramList = rawparams.split() %}
#    {% for i in range(paramList|length) %}
#      {% if paramList[i]=="Z=0" %}
#        {% set temp=paramList.pop(i) %}
#        {% set temp="Z_ADJUST=" + (-printer["gcode_macro SET_GCODE_OFFSET"].runtime_offset)|string %}
#        {% if paramList.append(temp) %}{% endif %}
#      {% endif %}
#    {% endfor %}
#    {% set rawparams=paramList|join(' ') %}
#    SET_GCODE_VARIABLE MACRO=SET_GCODE_OFFSET VARIABLE=runtime_offset VALUE=0
#  {% endif %}
  SET_GCODE_OFFSET_ORIG { rawparams }

[gcode_macro PROBE_EDDY_CURRENT_CALIBRATE_AUTO]
gcode:
  BED_MESH_CLEAR
  G28 X Y
  G1 X{ printer.toolhead.axis_maximum.x/2 } Y{ printer.toolhead.axis_maximum.y/2 } F6000
  {% if 'z' not in printer.toolhead.homed_axes %}
    SET_KINEMATIC_POSITION Z={ printer.toolhead.axis_maximum.z-1 } # Allows the user to work it down until it touches.
  {% endif %}
  PROBE_EDDY_CURRENT_CALIBRATE {rawparams}

[gcode_macro BED_MESH_CALIBRATE]
rename_existing: BTT_BED_MESH_CALIBRATE
gcode:
#  SET_GCODE_VARIABLE MACRO=_KNOMI_STATUS VARIABLE=probing VALUE=True #Only uncomment this line if using a KNOMI and then remove the BED_MESH_CALIBRATE macro from KNOMI.cfg
  BTT_BED_MESH_CALIBRATE METHOD=rapid_scan
#  SET_GCODE_VARIABLE MACRO=_KNOMI_STATUS VARIABLE=probing VALUE=False #Only uncomment this line if using a KNOMI and then remove the BED_MESH_CALIBRATE macro from KNOMI.cfg

# Acceleration Test Config

[adxl345]
cs_pin: host:gpio67
spi_bus: spidev0.0
axes_map: x,y,z

[resonance_tester]
accel_chip: adxl345
probe_points: 
  180,180,10 
  90,90,10 
  90,270,10 
  270,270,10 
  270,90,10

[gcode_macro Resonance_Finder]
gcode:
  G28
  TEST_RESONANCES AXIS=Y
  TEST_RESONANCES AXIS=X

[input_shaper]
shaper_freq_x: 43.2
shaper_type_x: mzv
shaper_freq_y: 31.4
shaper_type_y: mzv

# Printer and Speeds

[printer]
kinematics: corexy
max_velocity: 500
max_accel: 10000
max_z_velocity: 15
max_z_accel: 100
square_corner_velocity: 20

[stepper_x]
step_pin: PB13
dir_pin: PB12
enable_pin: !PB14
microsteps: 128
rotation_distance: 40
endstop_pin: PC0
position_endstop: -2
position_max: 374
position_min: -2
homing_speed: 300

[tmc2209 stepper_x]
uart_pin: PC11
tx_pin: PC10
uart_address: 0
run_current: 1

[gcode_arcs]
resolution: .2

[stepper_y]
step_pin: PB10
dir_pin: PB2
enable_pin: !PB11
microsteps: 128
rotation_distance: 40
endstop_pin: PC1
position_endstop: -8
position_min: -8
position_max: 367

homing_speed: 200

[tmc2209 stepper_y]
uart_pin: PC11
tx_pin: PC10
uart_address: 2
run_current: 1

[stepper_z]
step_pin: PB0
dir_pin: !PC5
enable_pin: !PB1
microsteps: 128
rotation_distance: 4
endstop_pin: probe:z_virtual_endstop
position_min: -10
position_max: 300
homing_speed: 10.0

[tmc2209 stepper_z]
uart_pin: PC11
tx_pin: PC10
uart_address: 1
run_current: .5

[extruder]
step_pin: PB3
dir_pin: !PB4
enable_pin: !PD1
microsteps: 16
rotation_distance: 7.8
nozzle_diameter: 0.400
filament_diameter: 1.750
heater_pin: PC8
sensor_type: EPCOS 100K B57560G104F
sensor_pin: PA0
control: pid
pid_Kp: 23.851
pid_Ki: 2.338
pid_Kd: 60.821
pressure_advance: 0.06
pressure_advance_smooth_time: 0.040
min_temp: 0
max_temp: 300
min_extrude_temp: 100
max_extrude_only_distance: 200

[tmc2209 extruder]
uart_pin: PC11
tx_pin: PC10
uart_address: 3
run_current: 0.5

[heater_bed]
heater_pin: PC9
sensor_type: EPCOS 100K B57560G104F
sensor_pin: PC4
control: pid
pid_Kp: 690.34
pid_Ki: 111.47
pid_Kd: 1068.83
min_temp: 0
max_temp: 120

[heater_fan heatbreak_cooling_fan]
pin: PC6

[multi_pin my_fan]
pins: PC7,PB15

[fan]
pin = multi_pin:my_fan
max_power: 1

[neopixel Lights]
pin: PA8
chain_count: 3
color_order: GRB
initial_RED: .5
initial_BLUE: .5
initial_GREEN: .5

[firmware_retraction]
retract_length: 1
# The length of filament (in mm) to retract when G10 is activated,
# and to unretract when G11 is activated (but see
# unretract_extra_length below). The default is 0 mm.
retract_speed: 35
# The speed of retraction, in mm/s. The default is 20 mm/s.
unretract_extra_length: 0
# The length (in mm) of *additional* filament to add when
# unretracting.
unretract_speed: 35
# The speed of unretraction, in mm/s. The default is 10 mm/s. 

[board_pins]
aliases:
    # EXP1 header
    EXP1_1=PB5,  EXP1_3=PA9,   EXP1_5=PA10, EXP1_7=PB8, EXP1_9=<GND>,
    EXP1_2=PA15, EXP1_4=<RST>, EXP1_6=PB9,  EXP1_8=PD6, EXP1_10=<5V>

[gcode_macro UNLOAD_FILAMENT]
gcode:
  M83
  G1 E-100 F600
  M83
  G1 E-100 F600

[gcode_macro LOAD_FILAMENT]
gcode:
  M83
  G1 E100 F600
  M83
  G1 E100 F600

#*# <---------------------- SAVE_CONFIG ---------------------->
#*# DO NOT EDIT THIS BLOCK OR BELOW. The contents are auto-generated.
#*#
#*# [probe_eddy_current btt_eddy]
#*# reg_drive_current = 16

BTT Justin, I would rather get this solved than have to go through a ticket process again, let me put this up.

bigtreetech, 2024-09-14:
As mentioned before, it appears as if you are typing the calibrate command incorrectly and leaving out the chip parameter. This is a support request and not a product software/hardware issue. Out ticketing process handles support requests on more mature products as it is actually quicker for the end user since repos are no longer monitored by support staff after a specified time and only monitored by engineers who specifically look for hardware/software concerns. If you would like to get it solved quicker then simply enter the correct command that includes the chip parameter as per the manual/github readme. Copy and paste the commands you see there.

closed 2024-09-14 by bigtreetech

Gansemagne, 2024-09-14:
I have been working on this error message for the past 3 hours, in which
time I have reflashed the firmware exactly to spec 5 times, searched
through the klipper eddy config files and repasted the command exactly from
the guide around 114 times. I did a klippy log search and made sure of the
number. Your support staff ghosted me after I submitted my .config file. I
really like bigtreetech, I run an skr mini 3 v3 and a btt pi, I really want
this to work, but I'm very disappointed in the customer service.

bigtreetech, 2024-09-14:
> I have been working on this error message for the past 3 hours, in which time I have reflashed the firmware exactly to spec 5 times, searched through the klipper eddy config files and repasted the [...]

Post the exact gcode command that you are sending. I can replicate your error with 100% repeatability by sending the wrong command missing the chip parameter.