bigtreetech/Eddy #71: Eddy Coil Settings for EBB36 (solution in post)

QuinFromCanada, 2024-09-16:
The documentations do not cover the Eddy Coil setup well, can the example please be added to sample config, or the manual, thank you.

**SOLUTION:**

1. Update your EBB36 firmware, eddy current probe native support was added to mainline klipper recently, and this will not work if you have an old version on your toolhead board.  I got it to work no problems with mainline klipper version 0.12.0-297

2. Use whatever BTT sample file you like, but make the following changes for Eddy Coil:

Leave [MCU] section commented out since Eddy Coil is a sensor and not an MCU.
`#[mcu eddy]`

I used an EBB36 v1.2 so I used the following settings:
```
[probe_eddy_current btt_eddy]
sensor_type: ldc1612
#i2c_address:
i2c_mcu: ebb36  # whatever you called you EBB36 mcu in my case it was [mcu ebb36]
i2c_bus: i2c3_PB3_PB4 # I2C bus for ebb36 v1.2
#i2c_speed: 400000  # didn't need this, so left it commented out.
#i2c_bus: i2c0f # didn't need this, so left it commented out.
# For a standard Voron stealthburner X carriage mount there should be no need to change the defaults below.
x_offset: 0
y_offset: 21.42
z_offset: 1.0
```

Hope this helps, anyone having issues.
Cheers!

bigtreetech, 2024-09-16:
Every single point here is already in the manual and config files though:

[image]

[image]

[image]

closed 2024-09-16 by bigtreetech

QuinFromCanada, 2026-02-01:
> then

When when this post was made klipper did not have native eddy current scanning so things weren't as cut and dry.  Infact if memory has it right you had to install the BTT fork of klipper to get things going.

Cheers.