bigtreetech/Eddy #73: Nozzle height too high

LeandroSQ, 2024-09-24:
Hello,
I've been trying to calibrate my Eddy, using it as endstop and **without the Z-Offset** config.

When the nozzle is touching the bed, Eddy is **2mm** from the bed surface
I ran `LDC_CALIBRATE_DRIVE_CURRENT CHIP=btt_eddy` with Eddy **20mm** from surface.
And running `PROBE_EDDY_CURRENT_CALIBRATE_AUTO CHIP=btt_eddy` (Manual does not work too) results in the nozzle being to high, unable to print.

I also did the temperature calibration, as I have the USB variant.

On the paper calibration, tried several times and all of them seem to result in the same issue.

aldoale, 2024-09-26:
same problem but nozzle too low in my case and I can't calibrate manually.

bigtreetech, 2024-09-26:
Completely clear out anything probe related from the automatically generated section. This includes all thermal offsets, z offsets, mesh settings, drive current, etc... start fresh.

Run the calibration, then run the thermal calibration and then see what happens. Should be good. Also, why did you choose not to use the z-offset out of interest?

LeandroSQ, 2024-09-26:
As the guide mentioned Z offset was not needed due to the nature of Eddy, I decided not to go with the an offset config, seemed a lot simpler as well.

Thank you for the instructions, will try later and let you know how it goes.

aldoale, 2024-10-20:
I tried to reinstall everything but the first layer is always wrong, did you solve it?

purpleriboflavin, 2025-01-03:
Same here, no matter how many times I calibrate, temperature or not, first layer is always wrong. Either too high or too low, need to recalibrate all the time to even get it to print.

LeandroSQ, 2025-02-19:
What fixed for me was creating a mount for my printer, the previous one was adjustable and it was too close to the bed. Mounting Eddy at around 3.5~mm when the nozzle is touching the bed is now working perfectly.

closed 2025-02-19 by LeandroSQ