bigtreetech/Eddy #77: Error during setup: probe_eddy_current sensor not in valid range

JostBrand, 2024-10-02:
I have the eddy USB on a Sovol SV06, and was able to place it excatly where the inductive probe was located so I copied over the X offset and Y offset, Mesh_min and mesh_max and Home_xy_position. 

This is my console output for the setup process. In step 28 I get the error `probe_eddy_current sensor not in valid range` when running `BED_MESH_CALIBRATE METHOD=rapid_scan`.
```
 00:39:53  $ G90
        G1 Z20.00 F600
00:39:58  $ **LDC_CALIBRATE_DRIVE_CURRENT CHIP=btt_eddy**
00:39:58  // probe_eddy_current btt_eddy: reg_drive_current: 15
// The SAVE_CONFIG command will update the printer config file
// with the above and restart the printer.
00:40:03  $ SAVE_CONFIG
00:40:04  // Klipper state: Disconnect
00:40:26  $ PROBE_EDDY_CURRENT_CALIBRATE_AUTO CHIP=btt_eddy
00:40:33  // Starting manual Z probe. Use TESTZ to adjust position.
// Finish with ACCEPT or ABORT command.
00:40:33  // Z position: ?????? --> 257.000 <-- ??????
00:40:37  $ TESTZ Z=-1
00:40:37  // Z position: ?????? --> 256.000 <-- 257.000
-- remove the steps to not pollute the ticket. 
00:41:25  $ ACCEPT
00:42:40  // temperature_probe btt_eddy: Z Calibration Temperature set to 31.97. The SAVE_CONFIG command will update the printer config file and restart the printer.
00:42:40  // probe_eddy_current: stddev=51.285 in 2486 queries
// The SAVE_CONFIG command will update the printer config file
// and restart the printer.
00:43:26  $ SAVE_CONFIG
00:43:27  // Klipper state: Disconnect
00:43:44  $ G28
00:44:02  // probe at 84.500,135.000 is z=2.478062
00:44:02  // Result is z=2.478062
00:44:16  $ BED_MESH_CALIBRATE METHOD=rapid_scan
00:44:16  // Beginning rapid surface scan at height 2.00...
00:44:16  // Sample hit distance +/- 1.6000mm, time window +/- ms 8.0000
00:44:30  !! probe_eddy_current sensor not in valid range
```

My [klipper.log](https://gist.github.com/JostBrand/e2f6f66f94a1b76b03d734dcf711b818)

Can someone please point me in the right direction to solve this?

rafik24, 2024-10-03:
I have the same issue, I tried fiddling with min/max x/y in vain.

Went over the calibration process again but still the same error message.

Some help would be appreciated.

SinglerGold2, 2024-10-06:
same issue here.

SinglerGold2, 2024-10-06:
I had the same issue but I think I have been able to solve the problem for myself:

The X-gantry was out of alignment, which meant that one side was much higher than the otherwith almost 2mm higher on the right than on the left. (Dual-Z)

I then recalibrated the two Z-screws and the two rails on which X rests and realigned X and now it works

cicada7772, 2024-10-06:
I have the same issue, I am able to make a mesh without the rapid scan and see that my bed has 0.55mm variance, which should be good enough for the probe to work.

JostBrand, 2024-10-07:
Solved - as it is often the case: Read the friendly manual was the solution.  The mesh_min and mesh_max are calculated differently.

Could be a bit more stressed in the documentation.

closed 2024-10-07 by JostBrand

goshi0, 2024-10-17:
sorry @JostBrand  can you elaborate?, I cant get rid of the problem ... Thanks

JostBrand, 2024-10-27:
Sure, just read these instructions really carefully:

```
# For the mesh dimensions below, the coordinates need to be reachable by the center of the probe. To calculate coordinates that will work, use the formula below:
# mesh x min = position_min_x + greater_of (15mm or x_offset) <--- in this term, only consider the x offset if it is positive, ignore if negative.
# mesh y min = position_min_y + greater_of (15mm or y_offset) <--- in this term, only consider the y offset if it is positive, ignore if negative.
# mesh x max = position_max_x - greater_of (15mm or |x_offset|) <--- in this term, only consider the x offset if it is negative, ignore if positive.
# mesh y max = position_max_y - greater_of (15mm or |y_offset|) <--- in this term, only consider the y offset if it is negative, ignore if positive.
# Example: Consider that you have a 300 x 300 bed with the max x and y positions being 300 and the min being 0. Your probe offsets are -20 for X and 10 for Y
# For mesh x min we ignore the x offset term because it is negative. Therefore mesh x min = 15
# For mesh y min we do not ignore the y offset term because it is positive but it is less than 15 so we use 15. Therefore mesh y min = 15
# For mesh x max we do not ignore the x offset term because it is negative. It is also greater than 15. Therefore mesh x max = 280
# For mesh y max we ignore the y offset term because it is positive but it is less than 15 so we use 15. Therefore mesh y max = 285
# The final result would be mesh_min: 15, 15 mesh_max: 280, 285
mesh_min: 10, 10  # modify these according to the above guide. If the probe cannot reach then you will get a klipper error when trying to scan a bed mesh.
mesh_max: 220, 220 # modify these according to the above guide. If the probe cannot reach then you will get a klipper error when trying to scan a bed mesh.
```