bigtreetech/Eddy #79: Failed calibration - frequency not increasing each step

SinglerGold2, 2024-10-05:
When I run `PROBE_EDDY_CURRENT_CALIBRATE_AUTO CHIP=btt_eddy` , I get the error Failed calibration - frequency not increasing each step . 
I  followed  the setup guide but I can't find a solution.

These are the results for the following commands:

`LDC_CALIBRATE_DRIVE_CURRENT CHIP=btt_eddy`
and then 
`PROBE_EDDY_CURRENT_CALIBRATE_AUTO CHIP=btt_eddy`

[image]

[image]

SinglerGold2, 2024-10-06:
It worked after a reboot of the system... and stopped working again after a firmware reset after I changed some stuff in the printer.cfg which were not eddy related

BTTJUSTIN, 2024-10-09:
Thanks for reaching out. Our support team does not operate using the GitHub platform but we would be happy to help you via email based ticketing system. Please visit https://biqu.equipment/pages/technical-support 
and submit a ticket and we will be in touch with you soon.

SinglerGold2, 2024-10-10:
> Thanks for reaching out. Our support team does not operate using the GitHub platform but we would be happy to help you via email based ticketing system. Please visit https://biqu.equipment/pages/technical-support and submit a ticket and we will be in touch with you soon.

trying/waiting to get that  confirmation email from your system for hours now... nothing. And yes, I checked my junk box.

closed 2024-11-22 by BTTJUSTIN

kingjamez, 2025-03-05:
For future peeps finding this via google like I did:
My solution to the error "Frequency not increasing each step" was to ensure that the EDDY was 2-3mm below the zero of the nozzle. In my case it was probably 5mm above the nozzle.

I took a full size SD card and moved the eddy down until the SD card fit between the EDDY and the bed when the nozzle was at zero.
After that the calbiration worked immediately.

MJD19994, 2025-03-18:
I never resolved my issues, I gave up and bought a Cartographer 3D probe which does touch calibration so it has higher success rates and is such an easier product to use vs BTTs Eddy.