bigtreetech/Eddy #80: Eddy with IDEX configuration not yet working

HenkVisser, 2024-10-08:
I spent many hours to get Eddy Coil working on two different 3D printers with IDEX configuration, to no avail. It doesnt let me calibrate the eddy with the PROBE_EDDY_CURRENT_CALIBRATE CHIP=btt_eddy command. I kept getting errors like: 'must home axis first', 'move out of range' or 'extruder temperature below minimum temperature'.

At last I did comment out everything with the dual carriage settings, and then it worked without problems.
I have 2 other 3D-printers working with the Eddy USB and Eddy Coil, therefor I know how everything should work.

So I suppose this is a bug, and I would love to get this to work in the IDEX mode!

[klippy (6).log](https://github.com/user-attachments/files/17297878/klippy.6.log)

bigtreetech, 2024-10-08:
That's an awfully South African name. Anyhow...

Could you upload the configs on the machine where it does not work?

HenkVisser, 2024-10-08:
Thanks!
I already did post the klippy log in my message (klippy (6).log). 
Or do you need other things as well?

The file has the calibrationdata of today in it, but I did uncomment the parts with dual carriage settings.

HenkVisser, 2024-10-13:
[Tenlog D3 config-20241013-115256.zip](https://github.com/user-attachments/files/17355128/Tenlog.D3.config-20241013-115256.zip)

Hereby the complete configfiles with the dualcarriage configs uncommented

HenkVisser, 2024-12-01:
> I am having the same problem with Eddy USB and my IDEX. [klippy.log](https://github.com/user-attachments/files/17965921/klippy.log)

I feel sorry for you!
I still haven´t had any feedback from bigtreetech and didn´t get it to work in the meantime.
I hope the will help us, otherwise we´d be better of buying a cartographer3D..

CNC959, 2024-12-01:
I actually did get it working, your post gave me the idea. I commented out 'everything' related to the dual carriage settings, including the fans, extruder, heater. I was then able to calibrate it. Once that was done I added everything back and everything worked as expected. The only thing is, if I need to recalibrate I will have to remove everything again. 
I expect that BTT will eventually address this as my resolution is only a work around.

HenkVisser, 2024-12-01:
Thank you for your explanation!
Yes I did the same, but then it wouldn't do the bed mesh before every
print. I only could load an existing stored bedmesh.
But you are right; It is doable, but I want perfect prints all the time..

I am curious if anyone has gotten it to work with an IDEX-setup.

I saw in your logfile, that you have a Cartesian printer as well. Could it
be that the eddy has to be somehow specifically linked to the T0 profile?
But then: with the klickyprobe and BLtouch I didn't have this problem.

CNC959, 2024-12-01:
My setup does the bed mesh via the BED_MESH_CALIBRATE METHOD=rapid_scan command, but if I try using just BED_MESH_CALIBRATE it behaves as a normal probe would. And thank you as well! I'm sure we aren't the only two people on the planet experiencing this problem!

decker9000, 2024-12-11:
I'm having the issue too. I'll try commenting out the dual carriage stuff. I might move it into a separate cfg file so I can just comment out the include to recalibirate quickly

CNC959, 2024-12-11:
Yeah, I just moved all the related to another config file and comment it out when I need to calibrate. It works for now.

decker9000, 2024-12-11:
Any chance you could share your config files that worked? Yours is more like my vanilla tl-d3 setup

CNC959, 2024-12-11:
Sure, I'll share to help out a brother. I removed everything to do with dual_carriage from the printer.cfg and moved it to IDEX.cfg, then just made an include for it. The only caveat is I had to remove the saved config for extruder1 at the bottom of the printer.cfg. I kept getting an error until I removed it, but I don't remember what the error was. Anyway, here is my IDEX.cfg. All this is highly experimental so be cautious. Good luck!

[IDEX.txt](https://github.com/user-attachments/files/18098004/IDEX.txt)

decker9000, 2024-12-12:
Thanks, got it working. I didn't need to comment out macros for my machine.