bigtreetech/Eddy #82: Invalid STEP and/or TARGET parameters resulted in too few expected samples: 1

aquari86, 2024-10-18:
When I use the command "TEMPERATURE_PROBE_CALIBRATE PROBE=btt_eddy TARGET=56 STEP=4", I get the error "Invalid STEP and/or TARGET parameters resulted in too few expected samples: 1". What could that be?

aquari86, 2024-10-19:
leider nein, das kommt bei mir auch an der gleichen stelle.

aquari86, 2024-10-19:
no but it just works again without me doing anything

buikpijn, 2024-10-21:
Same issue here.

dfbpurcell, 2024-10-21:
> Same issue here. Just turn it off, let it cool completely and perform again.

Yes I figured out that it happens when the calibration fails half way. Now the problem is its still failing half way. Eddy is way too temperamental, different errors at every single step. I've given up and I'm switching to cartographer + crtouch (for garolite)

BTTJUSTIN, 2024-10-25:
Thanks for reaching out. Our support team does not operate using the GitHub platform but we would be happy to help you via email based ticketing system. Please visit https://biqu.equipment/pages/technical-support 
and submit a ticket and we will be in touch with you soon.

closed 2024-11-22 by BTTJUSTIN

tethlah, 2025-02-12:
For anyone who comes across this and is looking for an answer (since for some reason BIQU doesn't want to make answers public for the good of everyone), You're going to get this error if the Eddy temperature is too high.  The number you are getting is the number of samples it collected based on your target temperature and step.  So if your Eddy is at 47 degrees, and you put in 56 with a step of 4, then you will have only 2 samples (56-47=9  9/2=8 rounded down which is 2 steps at 4 degrees).

It's not based on the bed or hotend, it's based on the temperature of the Eddy.  Like a previous poster said, you can turn it off and let it cool down.  If you want to speed this up, raise your Z to give it plenty of airflow, remove any covers or housings you have that's insulating it, and let it cool down.  It needs a minimum of 3 samples to be successful.  (so with a step of 4, the eddy needs to be 12 degrees below the target temperature).