bigtreetech/Eddy #83: probe_eddy_current sensor not in valid range

dravida77, 2024-10-18:
I have issues getting my Btt Eddy (USB) to work.
Ive made a fresh install of DietPi and Klipper, Moonraker, Fluidd, Klipperscreen, Crowsnest (via kiauh) to make sure my Virtual Enviroment runs within python3.
i updated the Firmware of the 2040.
The Sensor is mounted exactly 2mm above the Nozzle Tip.
I followed the Manual (https://github.com/krautech/btt-eddy-guide) very carefully without any luck.

The automated calibration kinda works but after a completed Bed Scan it drops the following error and wont create a Mesh.
"probe_eddy_current sensor not in valid range"

So i tried to go for the manual route but ive noticed that im unable to perform a "PROBE_EDDY_CURRENT_CALIBRATE CHIP=btt_eddy" wiothout homing Z,
so i tried to work arround the issue.
I homed every axis, moved everything to the needed position, turned the printer off and on again, 
and tried to run the commands with fixed locations like this:

SET_KINEMATIC_POSITION z=20
SET_KINEMATIC_POSITION x=165
SET_KINEMATIC_POSITION y=105
LDC_CALIBRATE_DRIVE_CURRENT CHIP=btt_eddy
SAVE_CONFIG

after this i moved the nozzle 5mm above the bed, turned the machine off and on again.

BED_MESH_CLEAR
SET_KINEMATIC_POSITION z=5
SET_KINEMATIC_POSITION x=165
SET_KINEMATIC_POSITION y=105
PROBE_EDDY_CURRENT_CALIBRATE CHIP=btt_eddy
SAVE_CONFIG
G28
BED_MESH_CALIBRATE METHOD=rapid_scan

The Machine will go for a complete Bed Scan, 
after the scanning is done i receive the same error again:
"probe_eddy_current sensor not in valid range"

You can find my eddy.cfg, printer.cfg and klippy.log here:
https://pastebin.com/x327WZ6j
https://pastebin.com/NYqt24CH
https://pastebin.com/295pRrvQ

Ive tried setting the drive current to 16ma instead of 15ma, ran everything several times again, and tried literally everything i was able to find online (which is not much) without any luck and gave up after 10h of trouble shooting.

Schmudus, 2024-10-18:
Hi,
I had similar issues. My Problem was that the bed was not leveled (Quad gantry level), so the distance was little too much on one corner.
Also you can try to play with the reg_drive_current. I set mine to 16, but be carefuily. If you do so, you need to do the complete calibration again!!!
Also you can play with the distance. I have a Eddy_Z_Offset of 1.5 and use in [bed_mesh] a horizontal_move_z: 2
Always take care that you don't cvrash your nozzle inside the bed. I destroyed one plate during adjustment and playing :-(

Schmudus, 2024-10-18:
Here's my eddy config parts. I put them in my printer.cfg

[probe_eddy_current btt_eddy]
sensor_type: ldc1612
i2c_mcu: eddy
i2c_bus: i2c0f
x_offset: -17
y_offset: 10
z_offset: 1.5
speed: 5.0
lift_speed: 20

[temperature_sensor btt_eddy_mcu]
sensor_type: temperature_mcu
sensor_mcu: eddy
min_temp: 10
max_temp: 100

[temperature_probe btt_eddy]
sensor_type: Generic 3950
sensor_pin: eddy:gpio26
horizontal_move_z: 2

[bed_mesh]
speed: 200
horizontal_move_z: 2
mesh_min: 10,10 
mesh_max: 333,340
probe_count: 15,15 #12,12 #9,9
algorithm: bicubic
bicubic_tension: 0.2 #0.3
move_check_distance: 5
split_delta_z: 0.02
mesh_pps:2,2
adaptive_margin: 5
fade_start: 0.6 #0
fade_end: 10
fade_target: 0

#*# <---------------------- SAVE_CONFIG ---------------------->
#*# DO NOT EDIT THIS BLOCK OR BELOW. The contents are auto-generated.
#*#
#*# [probe_eddy_current btt_eddy]
#*# reg_drive_current = 16

goshi0, 2024-10-20:
I have this problem too ,  it was working fine with the probe Z homing , but whe I tried to change the z offset to endstop that problem began to happen. Things I tried

Change the  reg_drive from 16 to 18 ; the given one was 17. In 16 It can't do gantry squaring 17 and 18 is fine.

Any one know what means the error? I supose that it means that when it begins to scan it's at z= X and when it ends it's not exactly at this  X if this is the case may  we can tweak this number . 

So anyone from bigtreetech can at least explain what these error means ? 

Thank you everyone

LittleReckliss, 2024-10-25:
I'm getting the same error. The area I'm trying to probe only has a deviation of 0.2mm. I know this because the Eddy can make me a mesh but I get this error when I try to do a rapid scan. I tried increasing the drive from 16 to 18, nothing. I tried slowing down the speed of the scan, didn't work. I'm at a loss. 

BTT, what does this error even mean?

closed 2024-11-22 by BTTJUSTIN