bigtreetech/Eddy #84: Eddy USB not handling chamber temperature

Zinput, 2024-10-18:
I have successfully set up an Eddy USB, but it has been very unreliable. I've now realized that once the temperature of the Eddy comes up to 60C, it is unable to home the printer. When I let it cool back down to 45C, it works again.

jbeima, 2024-11-04:
I have the EXACT same challenge...

The two times I have seen this happen were both following the temp calibration step.

Given that, I just installed it yesterday, so the machine has not gone into printing rotation again. I wonder how this thing will handle heat soaks... Hmmm Only time will tell.

jbeima, 2024-11-04:
I wonder how many people that are seeing the "Error during homing z: Eddy current sensor error" are really experiencing this...

Zinput, 2024-11-07:
I ended up using a workaround by modifying my homing and start gcode to bring the probe away from the bed when it is heating up. It then scans immediately after the bed is heated. You can start the heat soak after this process and it seems to work fine now.

serbiansky, 2024-12-26:
I have exactly the same issue, trying to do Temperature Compensation Calibration with 60c and when it reach over 60 it just froze whole printer and I am not able to connect any more over WiFI. After restart I can access but not to home z... I received TAP with my Voron and I decided to use Eddy but looks like I will switch to TAP...

Zinput, 2024-12-26:
It's probably best to use a different probe for printing ABS. I'm sure the bed mesh isn't all that accurate if we don't wait for a heat soak before doing it. This also seems like a software defined limitation, and I hope they implement a switch to allow higher temperatures.