bigtreetech/Eddy #85: Error => avrdude: ser_open(): can't open device "2e8a:0003"

Pmoshbr, 2024-10-21:
When compiling firmware, in step 11 "make flash FLASH_DEVICE=2e8a:0003" I receive this error from avrdude:

> lsusb
> Bus 001 Device 005: ID 1a86:7523 QinHeng Electronics HL-340 USB-Serial adapter
> Bus 001 Device 006: ID 2e8a:0003
> Bus 001 Device 003: ID 0424:ec00 Standard Microsystems Corp. SMSC9512/9514 Fast Ethernet Adapter
> Bus 001 Device 002: ID 0424:9514 Standard Microsystems Corp. SMC9514 Hub
> Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
> pi@fluiddpi:~/klipper $ make flash FLASH_DEVICE=2e8a:0003
>   Flashing out/klipper.elf.hex to 2e8a:0003 via avrdude
> avrdude: ser_open(): can't open device "2e8a:0003": No such file or directory
> 
> avrdude done.  Thank you.
> 
> make: *** [src/avr/Makefile:34: flash] Error 1
> pi@fluiddpi:~/klipper $ lsusb -t
> /:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=dwc_otg/1p, 480M
>     |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/5p, 480M
>         |__ Port 1: Dev 3, If 0, Class=Vendor Specific Class, Driver=smsc95xx, 480M
>         |__ Port 2: Dev 6, If 0, Class=Mass Storage, Driver=usb-storage, 12M
>         |__ Port 2: Dev 6, If 1, Class=Vendor Specific Class, Driver=, 12M
>         |__ Port 3: Dev 5, If 0, Class=Vendor Specific Class, Driver=ch341, 12M

I'm running Kippler at RPi 3b with Eddy USB v1.0
- Kippler v.0.12.0-317-gb89d5523
- Moonraker v.0.9.3-0-g71f9e677
- Python 3.8
- Raspbian GNU/Linux 10 (buster) 4.0

Could please somebody help me?

BTTJUSTIN, 2024-10-25:
Thanks for reaching out. Our support team does not operate using the GitHub platform but we would be happy to help you via email based ticketing system. Please visit https://biqu.equipment/pages/technical-support 
and submit a ticket and we will be in touch with you soon.

thechucklingatom, 2025-01-07:
I ran into this same issue, and was able to resolve it by verifying I had my make menuconfig selection correctly
[image]